COVESA/vsomeip #432: Issue with using Examples on Windows 10 PC

RBEI-ArunKumar, 2023-04-12:
I compiled the SomeIP libraries and examples on WIndows 10 using Visual Studio 2019 compiler. When I run the examples e.g. vsomeip_ctrl.exe, I always get this error shown below. Any idea on how\what to look for to avoid this error:

```
2023-04-12 14:46:38.894062 [error] vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]
2023-04-12 14:46:38.894590 [info] Application(vsomeip_ctrl, ffff) is initialized (11, 100).
2023-04-12 14:46:38.894950 [info] Starting vsomeip application "vsomeip_ctrl" (ffff) using 2 threads
2023-04-12 14:46:38.895327 [error] vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]
```

fcmonteiro, 2023-11-14:
Issue is being investigated internally.

duartenfonseca, 2024-09-06:
hi @RBEI-ArunKumar  there was an update to the windows build, could you use a new version and retry?

duartenfonseca, 2024-09-26:
closed due to no response

closed 2024-09-26 by duartenfonseca