COVESA/vsomeip #434: v3.3.0 build fails with MSVC

ip138, 2023-04-14:
Hi,

with the recent 3.3.0 release I cannot build on Windows using MSVC anymore.

**Environment**
Windows 21H2
Visual Studio 2022 / MSVC 14.35
Boost 1.81
`cmake options -DENABLE_SIGNAL_HANDLING=1  -DBOOST_ROOT=D:/boost_1_76_0 ..`

**Error list**
Severity	Code	Description	Project	File	Line	Suppression State
Error	C3861	'io_': identifier not found	vsomeip3	D:\vsomeip\implementation\endpoints\src\server_endpoint_impl.cpp	660	
Error	C3861	'io_': identifier not found	vsomeip3	D:\vsomeip\implementation\endpoints\src\server_endpoint_impl.cpp	685	
Error	C2065	'io_': undeclared identifier	vsomeip3	D:\vsomeip\implementation\endpoints\src\server_endpoint_impl.cpp	660	
Error	C2065	'io_': undeclared identifier	vsomeip3	D:\vsomeip\implementation\endpoints\src\server_endpoint_impl.cpp	685	
Error	C2679	binary '<<': no operator found which takes a right-hand operand of type 'std::string' (or there is no acceptable conversion)	vsomeip3	D:\vsomeip\implementation\endpoints\src\server_endpoint_impl.cpp	738	
Error	C2065	'udp_server_endpoint_impl_receive_op': undeclared identifier	vsomeip3	D:\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp	215	
Error	C2146	syntax error: missing ';' before identifier 'its_operation'	vsomeip3	D:\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp	215	
Error	C2065	'its_operation': undeclared identifier	vsomeip3	D:\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp	215	
Error	C2065	'its_operation': undeclared identifier	vsomeip3	D:\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp	234	
Error	C2040	'its_option': 'boost::asio::socket_base::receive_buffer_size' differs in levels of indirection from 'const char *'	vsomeip3	D:\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp	886	
Error	C3861	'localtime_r': identifier not found	vsomeip3	D:\vsomeip\implementation\logger\src\message.cpp	122	
Warning	C4297	'vsomeip_v3::logger::message::~message': function assumed not to throw an exception but does	vsomeip3	D:\vsomeip\implementation\logger\src\message.cpp	194	
Error	LNK1104	cannot open file 'Debug\vsomeip3.lib'	vsomeip3-e2e	D:\vsomeip\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\vsomeip3.lib'	vsomeip3-sd	D:\vsomeip\build\LINK	1	
Error	LNK1104	cannot open file 'Debug\vsomeip3.lib'	vsomeip3-cfg	D:\vsomeip\build\LINK	1

goncaloalmeida, 2023-04-14:
we are checking this and we will try to push a fix soon.
thanks @ip138

goncaloalmeida, 2023-06-05:
Solved in the last releases.
Please retest.

closed 2023-06-05 by goncaloalmeida