COVESA/vsomeip #435: non-portable format specifiers used for fscanf in routing_manager_impl.cpp

insi-eb, 2023-04-14:
routing_manager_impl.cpp uses non-portable format specifiers `%lu`, which lead to compiler warnings/errors (with -Werror) on e.g. 32bit ARM platforms.

The portable way is to use `SCNu64` from `cinttypes`:
                                                                                                                                                                  
```
diff --git a/implementation/routing/src/routing_manager_impl.cpp b/implementation/routing/src/routing_manager_impl.cpp                                                                                                                                                    
index 4ebca55..048bdf7 100644                                                                                                                                                                                                                                             
--- a/implementation/routing/src/routing_manager_impl.cpp                                                                                                                                                                                                                 
+++ b/implementation/routing/src/routing_manager_impl.cpp                                                                                                                                                                                                                 
@@ -5,6 +5,7 @@                                                                                                                                                                                                                                                           
                                                                                                                                                                                                                                                                          
 #include <climits>                                                                                                                                                                                                                                                       
 #include <iomanip>                                                                                                                                                                                                                                                       
+#include <cinttypes>                                                                                                                                                                                                                                                     
 #include <memory>                                                                                                                                                                                                                                                        
 #include <sstream>                                                                                                                                                                                                                                                       
 #include <forward_list>                                                                                                                                                                                                                                                  
@@ -4351,7 +4352,7 @@ void routing_manager_impl::memory_log_timer_cbk(                                                                                                                                                                                                    
     std::uint64_t its_data(0);                                                                                                                                                                                                                                           
     std::uint64_t its_dirtypages(0);                                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                          
-    if (EOF == std::fscanf(its_file, "%lu %lu %lu %lu %lu %lu %lu", &its_size,                                                                                                                                                                                           
+    if (EOF == std::fscanf(its_file, "%" SCNu64 " %" SCNu64 " %" SCNu64 " %" SCNu64 " %" SCNu64 " %" SCNu64 " %" SCNu64, &its_size,                                                                                                                                      
                     &its_rsssize, &its_sharedpages, &its_text, &its_lib,                                                                                                                                                                                                 
                     &its_data, &its_dirtypages)) {                                                                                                                                                                                                                       
         VSOMEIP_ERROR<< "memory_log_timer_cbk: error reading:"                                                                                                                                                                                                           
```

ip138, 2023-04-14:
see PR #429

goncaloalmeida, 2023-04-14:
PR merged :)

closed 2023-04-14 by goncaloalmeida