COVESA/vsomeip #436: A separate server service instance shall be used per network interface if a service needs to be offered on multiple network interfaces.

wonjae-song, 2023-04-18:
according to some/ip SD specification, (https://www.autosar.org/fileadmin/standards/R22-11/FO/AUTOSAR_PRS_SOMEIPServiceDiscoveryProtocol.pdf)
- [PRS_SOMEIPSD_00238] A separate server service instance shall be used per network interface if a service needs to be offered on multiple network interfaces
- [PRS_SOMEIPSD_00239] A separate client service instance shall be used per network interface if a service needs to be configured to be accessed using multiple different network interfaces

could I meet above specifications using vsomeip?

I understood that we need to use seperate routing app if we need to offer a service on multiple network interface (ex> eth0 and eth1).
as I know, we can seperate routing app using seperate vsomeip configuration file ( configure unicast, network, routing for each routing app)
but is it possible that one application use multiple routing? 
It seems that application communicate with routing app using /etc/"network"-0 UDS. but a configuration file can hold only one "network".

yangqiao95, 2023-05-13:
do  you  success  arm-Linux   to   arm-tc397-autosar?

goncaloalmeida, 2023-05-25:
@wonjae-song In vsomeip userguide you can search for "VSOMEIP_CONFIGURATION_<application>" for a brief explanation.

closed 2023-05-25 by goncaloalmeida