COVESA/vsomeip #437: On the issue of using udp for large data volume communication

xiyandong, 2023-04-23:
1. Starting from which version does it support the function of someip tp? If it has this function, can it send a quantity greater than 1416 bytes.

2. Some versions have bugs under big_payload_tests provides an example of udp sending big data packets, but when using the corresponding example for big data packets, an error will be reported.
[error] cei::check_message_size: Dropping to big message (614063 Bytes). Maximum allowed message size is: 1416 Bytes

goncaloalmeida, 2023-05-19:
@xiyandong we are giving support only for version 3.3.0, could you please confirm if you see this type of error on this version?

goncaloalmeida, 2023-05-25:
For large data volume, tcp should be used, if you really need to use UDP you can look at the vsomeip userguide and search for "someip-tp". You can also have a look at the test for this at "vsomeip/test/network_tests/someip_tp_tests" for a better understanding.

closed 2023-05-25 by goncaloalmeida