COVESA/vsomeip #440: compilation problem with boost::icl::interval

cml-ordnas, 2023-04-27:
Hi,

I am cross compiling vsomeip 3.3.0 with yocto (langdale, with boost v1.80.0) for an aarch64 platform and getting these build errors:

`sources/vsomeip/implementation/security/src/policy.cpp:74:18: error: no match for 'operator+=' (operand types are 'boost::icl::interval_map<unsigned int, boost::icl::interval_set<unsigned int> >' and 'std::pair<unsigned int, boost::icl::interval_set<unsigned int> >')
|    74 |     credentials_ += std::make_pair(its_uid_interval, its_gid_set);`

after some time of investigating it seems the aggregate on overlap principle on interval_maps is the problem here. After replacing it by traditional insert semantics the build is successfully done. Has anyone discovered such a problem too? 

thanks

goncaloalmeida, 2023-05-04:
@cml-ordnas no. in our internal CIs and local builds we don't have this error.

smooge, 2023-05-17:
I am seeing a possible similar problem with its_gid_set but with Fedora Linux 38 and boost-v1.78. The complete error I get is
```
[ 61%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o
/usr/bin/g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DVSOMEIP_VERSION=\"3.3.0\" -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_RECURSIVE_MUTEX_INIT_FUNC -Dvsomeip3_EXPORTS -I/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/interface -isystem /usr/include/dlt -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -DVSOMEIP_BOOST_VERSION=107800 -DUSE_DLT  -pie -Wl,-z,relro,-z,now -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIC -pie -Wl,-z,relro,-z,now -g  -std=c++14  -Wl,-export-dynamic -Wl,--version-script=/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/exportmap.gcc -DVSOMEIP_ENABLE_COMPAT -O2 -g -DNDEBUG -fPIC -MD -MT CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o -MF CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o.d -o CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o -c /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/tracing/src/channel_impl.cpp
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /usr/include/boost/icl/interval_map.hpp:13,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/../include/policy.hpp:15,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/policy.cpp:10:
In function 'boost::icl::segmental::join_under<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::value_type const&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::iterator)boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::iterator',
    inlined from 'boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::add_over(boost::icl::discrete_interval<unsigned int, std::less> const&, std::_Rb_tree_const_iterator<boost::icl::discrete_interval<unsigned int, std::less> >)' at /usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::_add(boost::icl::discrete_interval<unsigned int, std::less> const&)' at /usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::add(boost::icl::discrete_interval<unsigned int, std::less> const&)' at /usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from 'boost::icl::add<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::segment_type const&)boost::enable_if<boost::icl::is_interval_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >::type&' at /usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from 'boost::icl::add<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_i [... cut ...] ecurity_credentials_command const&)':
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/routing_manager_client.cpp:2812:44: note: at offset 40 into destination object 'its_gid_set' of size 48
 2812 |         boost::icl::interval_set<uint32_t> its_gid_set;
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

If I should open this as a separate ticket I can do so, but I was seeing that it was a similar part of boost

smooge, 2023-05-18:
The problem turns into a compile time problem on Fedora due to `vsomeip-3.3.0/CMakeLists.txt` around line 75

```
set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
```

I missed it at first because I saw the first `-Wall -Wextra -Wformat -Wformat-security -Wconversion ` and did not see the rest at the end of the line `-Wformat -Wformat-security -Wpedantic -Werror` The code will compile with that and one other warning

goncaloalmeida, 2023-05-19:
@smooge did you test with boost 1.74? Internally we don't use fedora, only ubuntu.

smooge, 2023-05-19:
So the code does not generate that warning if I use the compilers shipped with CentOS Stream 9 which comes with
```
gcc-c++-11.3.1-4.4.el9.x86_64
libstdc++-11.3.1-4.4.el9.x86_64
boost-devel-1.75.0-8.el9.x86_64
```

However using the next generation gcc compiler 12, it does generate the warning in that section

```
 rpm -q gcc-c++ libstdc++ boost-devel
gcc-c++-12.3.1-1.fc37.x86_64
libstdc++-12.3.1-1.fc37.x86_64
boost-devel-1.78.0-9.fc37.x86_64

```

and it will crash out with 
```
    inlined from 'vsomeip_v3::routing_manager_client::on_update_security_credentials(vsomeip_v3::protocol::update_security_credentials_command const&)' at /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/routing_manager_client.cpp:2816:27:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))
      |     ^~~~~~~~~~
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/routing_manager_client.cpp: In member function 'vsomeip_v3::routing_manager_client::on_update_security_credentials(vsomeip_v3::protocol::update_security_credentials_command const&)':
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/routing_manager_client.cpp:2812:44: note: at offset 40 into destination object 'its_gid_set' of size 48
 2812 |         boost::icl::interval_set<uint32_t> its_gid_set;
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

The problem is the 'hidden' -Werror in the CMakeLists.txt. Cleaning that out will allow for the code to compile. I would expect the kernel in Ubuntu 23.04 to give a similar breakage, but I see the libboost there is 1.74.

cml-ordnas, 2023-05-22:
Hello,
Yes, I can confirm that removing the -Werror flag in CMakeList.txt also solves my compilation problem.
Thanks to @smooge.

closed 2023-09-01 by fcmonteiro