COVESA/vsomeip #443: request-to-response between 2 someip arm-machines   problem

yangqiao95, 2023-05-09:
requsest.json
{
    "unicast" : "192.168.1.4",
    "netmask" : "255.255.255.0",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "true"
    },
    "applications" :
    [

        {
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "routing" : "client-sample1",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }

[image]

request.sh
#!/bin/sh
route add -host 224.224.224.245 dev  eth0
sleep 10
cd /userdata/someip-arm/source/vsomeip-3.1.20.3/examples-yq/build
export LD_LIBRARY_PATH=/userdata/someip-arm/install/boost_1_74_0/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/userdata/someip-arm/install/vsomeip-3.1.20.3/lib:$LD_LIBRARY_PATH
VSOMEIP_CONFIGURATION=../vsomeip-request-yq.json VSOMEIP_APPLICATION_NAME=client-sample ./request-sample

Response.json
{
    "unicast" : "192.168.1.3",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "service-sample",
            "id" : "0x1277"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509",
            "multicast" :
            {
                "address" : "224.225.226.233",
                "port" : "32344"
            },
            "events" :
            [
                {
                    "event" : "0x8777",
                    "is_field" : "true",
                    "update-cycle" : 2000
                },
                {
                    "event" : "0x8778",
                    "is_field" : "true",
                    "update-cycle" : 0
                },
                {
                    "event" : "0x8779",
                    "is_field" : "true"
                }
            ],
            "eventgroups" :
            [
                {
                    "eventgroup" : "0x4455",
                    "events" : [ "0x8777", "0x8778" ]
                },
                {
                    "eventgroup" : "0x4465",
                    "events" : [ "0x8778", "0x8779" ],
                    "is_multicast" : "true"
                },
                {
                    "eventgroup" : "0x4555",
                    "events" : [ "0x8777", "0x8779" ]
                }
            ]
        }
    ],
    "routing" : "service-sample1",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

[image]

response.sh
#!/bin/sh
route add -host 224.224.224.245 dev  eth0
sleep 10
cd /userdata/someip-arm/source/vsomeip-3.1.20.3/examples-yq/build
export LD_LIBRARY_PATH=/userdata/someip-arm/install/boost_1_74_0/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/userdata/someip-arm/install/vsomeip-3.1.20.3/lib:$LD_LIBRARY_PATH
VSOMEIP_CONFIGURATION=../vsomeip-response-yq.json VSOMEIP_APPLICATION_NAME=service-sample ./response-sample

I pass  the demo in ubuntu,   why   not   pass   in arm-linux?

fcmonteiro, 2023-05-10:
Hi @yangqiao95 
Does this issue occurs with the latest vsomeip 3.3?
We have dropped the support to 3.1
Thanks

cc: @goncaloalmeida

fcmonteiro, 2023-05-10:
yes, we use arm to arm machines in linux
In principle, you shouldn't need to change anything in the config files

yangqiao95, 2023-05-11:
> 是的，我们在 linux 中使用 arm 来武装机器 原则上，您不需要更改配置文件中的任何内容

which  is   boost of  version for  vsomeIP  3.3  in  arm-linux？   I  try   to   boost 1.74 and 1.82，but  it  fail to  complie  example of vsomip  3.3  .  please  help  me 

> 是的，我们在 linux 中使用 arm 来武装机器 原则上，您不需要更改配置文件中的任何内容

which  is   boost of  version for  vsomeIP  3.3  in  arm-linux？   I  try   to   boost 1.74 and 1.82，but  it  fail to  complie  example of vsomip  3.3  .  please  help  me

fcmonteiro, 2023-05-12:
Boost 1.74 should work fine.
What issue do you have when compiling in ubuntu?

yangqiao95, 2023-05-12:
> Boost 1.74 should work fine. What issue do you have when compiling in ubuntu?

I   deal  with the  question of  boost  1.74.   But      I    fail  to  pass   version 3.3  in  two-arm  linux .   why?  
please  help  me to  solute  this question.

request-sample.sh
#!/bin/sh
  route add -n 224.224.224.245 dev  eth0
cd /userdata/someip-arm/source/vsomeip-3.3.0/examples-yq/build
export LD_LIBRARY_PATH=/userdata/someip-arm/install/boost_1_74_0/lib:$LD_LIBRARY_PATH
export LD_LIBRARY_PATH=/userdata/someip-arm/install/vsomeip-3.3.0/lib:$LD_LIBRARY_PATH 
VSOMEIP_CONFIGURATION=../vsomeip-request-yq.json VSOMEIP_APPLICATION_NAME=client-sample ./request-sample

root@a1000:/userdata/someip-arm/source/vsomeip-3.3.0/examples-yq/request-response-sample# ./request-sample.sh 
2022-05-13 05:15:08.196951 [info] Using configuration file: "../vsomeip-request-yq.json".
2022-05-13 05:15:08.198849 [info] Parsed vsomeip configuration in 1ms
2022-05-13 05:15:08.199876 [info] Configuration module loaded.
2022-05-13 05:15:08.200402 [info] Initializing vsomeip application "client-sample".
2022-05-13 05:15:08.200880 [info] Instantiating routing manager [Proxy].
2022-05-13 05:15:08.201741 [info] Client [1344] is connecting to [0] at /tmp/vsomeip-0
2022-05-13 05:15:08.202490 [info] Application(client-sample, 1344) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2022-05-13 05:15:08.204231 [info] Starting vsomeip application "client-sample" (1344) using 2 threads I/O nice 255
2022-05-13 05:15:08.206457 [info] main dispatch thread id from application: 1344 (client-sample) is: ffffa45ab1c0 TID: 2300
2022-05-13 05:15:08.207022 [info] shutdown thread id from application: 1344 (client-sample) is: ffffa3daa1c0 TID: 2301
2022-05-13 05:15:08.209467 [info] io thread id from application: 1344 (client-sample) is: ffffa57d8010 TID: 2298
2022-05-13 05:15:08.209366 [info] io thread id from application: 1344 (client-sample) is: ffffa35a91c0 TID: 2302
2022-05-13 05:15:18.212055 [warning] Client 0x1344 request client timeout! Trying again...
2022-05-13 05:15:18.213984 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2022-05-13 05:15:18.215436 [info] Application/Client 1344: Reconnecting to routing manager.
2022-05-13 05:15:28.219192 [warning] Client 0x1344 request client timeout! Trying again...
2022-05-13 05:15:28.221112 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2022-05-13 05:15:28.222352 [info] Application/Client 1344: Reconnecting to routing manager.
2022-05-13 05:15:38.226008 [warning] Client 0x1344 request client timeout! Trying again...
2022-05-13 05:15:38.228448 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2022-05-13 05:15:38.229631 [info] Application/Client 1344: Reconnecting to routing manager.
2022-05-13 05:15:48.233135 [warning] Client 0x1344 request client timeout! Trying again...
2022-05-13 05:15:48.235023 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2022-05-13 05:15:48.235970 [info] Application/Client 1344: Reconnecting to routing manager.
2022-05-13 05:15:59.239716 [warning] Client 0x1344 request client timeout! Trying again...
2022-05-13 05:15:59.241698 [info] receive_cbk local_uds_client_endpoint: connection_reset/EOF/bad_descriptor
2022-05-13 05:15:59.242821 [info] Application/Client 1344: Reconnecting to routing manager.

vsomeip-request-yq.json
{
    "unicast" : "192.168.1.3",
    "netmask" : "255.255.255.0",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "true"
    },
    "applications" : 
    [

        {    
            "name" : "client-sample",
            "id" : "0x1344"
        }
    ],
    "routing" : "client-sample1",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
[... 115 lines cut ...]
                    "is_multicast" : "true"                                        
                },
                {
                    "eventgroup" : "0x4555",
                    "events" : [ "0x8777", "0x8779" ] 
                }
            ]
        }
    ],
    "routing" : "service-sample1",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

goncaloalmeida, 2023-05-19:
@yangqiao95 the support for 3.1 closed last month, now we only support 3.3.0.
And in both version we tried locally all seems build ok.
The boost version that I used was the 1.74

yangqiao95, 2023-05-21:
> @yangqiao95上个月关闭了对 3.1 的支持，现在我们只支持 3.3.0。 在我们本地尝试的两个版本中，一切似乎都可以正常构建。 我使用的升压版本是 1.74
> 
> 关于具体例子的执行情况，我们会在内部进行检查。

thank  you  。  I   have  soluted  this  question。
but   I  have  anther  the  question。   Do  you  try   arm（soc-linux）  to   arm（mcu-such  as  infineon-aurix-TC397）in some/ip  ?

closed 2023-06-07 by fcmonteiro