COVESA/vsomeip #446: Initial field notification does not seem to work for TCP

evesson, 2023-05-11:
I created a simple server and client application. Server provides a field and client subscribes to the field notification. When running both applications on the same host it all work as expected: when the client application is created it gets the initial value of the field. 

When running the server remote, using TCP, the client never gets that initial field value. Instead I get the following log message:
"notify_one_unlocked: Notifying 0708.314.8008 to client 102 failed. Event payload not set!"

Note that the client gets all updates from the server for that field, even via TCP, when the server pushes updates. It's just that initial notification that is missing.

Using Wireshark I could observe the initial field value being sent by the server whenever the client gets connected.

fcmonteiro, 2023-05-15:
@evesson can you provide us some logs?

evesson, 2023-05-16:
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.183098 [info] Using configuration folder: /run/vsomeip/conf.
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.254543 [info] Parsed vsomeip configuration in 76ms
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.258320 [info] Configuration module loaded.
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.261651 [info] Initializing vsomeip application SomeipProvider.
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.264015 [info] Instantiating routing manager [Proxy].
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.280383 [info] Client [ffff] is connecting to [0] at /run/vsomeip/vsomeip-0
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.296089 [info] Application(SomeipProvider, ffff) is initialized (11, 100).
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.447645 [info] Starting vsomeip application SomeipProvider (ffff) using 2 threads I/O nice 255
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.608576 [info] shutdown thread id from application: ffff (SomeipProvider) is: f4afd3c0 TID: 628
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.617508 [info] main dispatch thread id from application: ffff (SomeipProvider) is: f52fe3c0 TID: 627
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.635376 [info] io thread id from application: ffff (SomeipProvider) is: f5aff3c0 TID: 626
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.696469 [info] io thread id from application: ffff (SomeipProvider) is: f42fc3c0 TID: 629
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.971897 [info] create_local_server: Listening @ /run/vsomeip/vsomeip-110
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.982058 [info] Client 110 (SomeipProvider) successfully connected to routing ~> registering..
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.983635 [info] Registering to routing manager @ vsomeip-0
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.527777 [info] Application/Client 110 (SomeipProvider) is registered.
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.704579 [info] REGISTER EVENT(0110): [0708.0314:eventtype=2:is_provided=false:reliable=1]
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.277525 [info] Client [100] is connecting to [110] at /run/vsomeip/vsomeip-110
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.101949 [info] Application/Client 0110 is registering.
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.640950 [info] REGISTERED_ACK(0110)
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.698893 [info] REGISTER EVENT(0110): [0708.0314:eventtype=0:is_provided=false:reliable=1]
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.833061 [info] endpoint_manager_impl::create_remote_client: 192.168.7.2:882 reliable: 1 using local port: 0
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.414882 [info] SUBSCRIBE(0110): [0708.0314.000a:ffff:1]
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.709925 [info] REQUEST(0110): [0708.0314:255.4294967295]
<30>May 16 11:47:07 routingmanagerd[246]: 2023-05-16 11:47:07.450008 [info] **notify_one_unlocked: Notifying 0708.314.8008 to client 110 failed. Event payload not set!**
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.322330 [info] ON_AVAILABLE(0110): [0708.0314:1.0]
<30>May 16 11:47:07 client[623]: 2023-05-16 11:47:07.429729 [info] SUBSCRIBE ACK(0100): [0708.0314.000a.ffff]

goncaloalmeida, 2023-05-25:
Please provide a pcap to analyse this better.

evesson, 2023-05-26:
I can't reproduce that behavior any longer. Seems to work fine now, tried a whole bunch of scenarios, so maybe I made the wrong observation in the first place.

I always get that "**notify_one_unlocked: Notifying 0708.314.8008 to client 110 failed. Event payload not set!**" though.

I guess you could close this bug and then I will file another one if I observe it again.

fcmonteiro, 2023-05-26:
Can you share with us your server/client examples for us to have a look?
Thanks

closed 2023-06-12 by fcmonteiro