COVESA/vsomeip #448: Unable to execute vsomeip notify-subscriber example from the different systems

udit23231, 2023-05-12:
I am trying to run notify-sample and subscribe-sample application in two different instance in the same network, Here client is not able to recognize the notify service

It works fine when I use the same instance, Is there anything that I need to specify in the configs.

In my configs i have specified the local ip in unicast and in the subscriber-client mentioned the server IPs, rest all the things i keep the defaults

[image]

udit23231, 2023-05-12:
Hi @fcmonteiro , tried with the latest too but get same issue.

yangqiao95, 2023-05-15:
what  is different  systems  ？   ubuntn  and   android?    do  you  try  linux(arm) -to - TC397(ECU)?

udit23231, 2023-05-15:
Both having same Linux ubuntu OS

yangqiao95, 2023-05-15:
> 两者都具有相同的 Linux ubuntu 操作系统

linux  to   ubuntu   ?

udit23231, 2023-05-15:
ubuntu to ubuntu communication same version

udit23231, 2023-05-16:
I already added those routes in both the instances but same issue

yangqiao95, 2023-05-16:
config  files  refer  to   https://github.com/COVESA/vsomeip/issues443    
ubuntu  to  ubuntu   successfully   pass

udit23231, 2023-05-16:
Don't have access to above URL gives me 404

yangqiao95, 2023-05-16:
> 无法访问上面的 URL 给我 404

https://github.com/COVESA/vsomeip/issues/443

fcmonteiro, 2023-05-22:
@udit23231, are you able to see the Offers/Finds using wireshark?
I've just tested it locally and with 2 different machines and it's working fine.

udit23231, 2023-05-23:
Yes its working fine now locally.  In cloud its failing maybe because multicasting is not supporting there.

closed 2023-05-23 by udit23231