COVESA/vsomeip #449: Compilation problems with vsomeip3 on Fedora Linux after F36

smooge, 2023-05-18:
I am seeing a possible similar problem with its_gid_set but with Fedora Linux 38 and boost-v1.78. The complete error I get is
```
[ 61%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o
/usr/bin/g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DVSOMEIP_VERSION=\"3.3.0\" -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_RECURSIVE_MUTEX_INIT_FUNC -Dvsomeip3_EXPORTS -I/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/interface -isystem /usr/include/dlt -O2 -flto=auto -ffat-lto-objects -fexceptions -g -grecord-gcc-switches -pipe -Wall -Wp,-U_FORTIFY_SOURCE,-D_FORTIFY_SOURCE=3 -Wp,-D_GLIBCXX_ASSERTIONS -specs=/usr/lib/rpm/redhat/redhat-hardened-cc1 -fstack-protector-strong -specs=/usr/lib/rpm/redhat/redhat-annobin-cc1  -m64  -mtune=generic -fasynchronous-unwind-tables -fstack-clash-protection -fcf-protection -fno-omit-frame-pointer -mno-omit-leaf-frame-pointer -DVSOMEIP_BOOST_VERSION=107800 -DUSE_DLT  -pie -Wl,-z,relro,-z,now -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIC -pie -Wl,-z,relro,-z,now -g  -std=c++14  -Wl,-export-dynamic -Wl,--version-script=/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/exportmap.gcc -DVSOMEIP_ENABLE_COMPAT -O2 -g -DNDEBUG -fPIC -MD -MT CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o -MF CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o.d -o CMakeFiles/vsomeip3.dir/implementation/tracing/src/channel_impl.cpp.o -c /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/tracing/src/channel_impl.cpp
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /usr/include/boost/icl/interval_map.hpp:13,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/../include/policy.hpp:15,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/policy.cpp:10:
In function 'boost::icl::segmental::join_under<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::value_type const&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::iterator)boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::iterator',
    inlined from 'boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::add_over(boost::icl::discrete_interval<unsigned int, std::less> const&, std::_Rb_tree_const_iterator<boost::icl::discrete_interval<unsigned int, std::less> >)' at /usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::_add(boost::icl::discrete_interval<unsigned int, std::less> const&)' at /usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::add(boost::icl::discrete_interval<unsigned int, std::less> const&)' at /usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from 'boost::icl::add<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::segment_type const&)boost::enable_if<boost::icl::is_interval_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >::type&' at /usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from 'boost::icl::add<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >(boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>&, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::element_type const&)boost::enable_if<boost::icl::is_interval_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >, boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator> >::type&' at /usr/include/boost/icl/concept/interval_set.hpp:79:20,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::add(unsigned int const&)' at /usr/include/boost/icl/interval_base_set.hpp:252:24,
    inlined from 'boost::icl::interval_base_set<boost::icl::interval_set<unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>, unsigned int, std::less, boost::icl::discrete_interval<unsigned int, std::less>, std::allocator>::insert(unsigned int const&)' at /usr/include/boost/icl/interval_base_set.hpp:289:19,
   inlined from 'vsomeip_v3::policy::deserialize(unsigned char const*&, unsigned int&)' at /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/policy.cpp:71:23:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))
      |     ^~~~~~~~~~
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/policy.cpp: In member function 'vsomeip_v3::policy::deserialize(unsigned char const*&, unsigned int&)':
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/security/src/policy.cpp:70:37: note: at offset 40 into destination object 'its_gid_set' of size 48
   70 |     boost::icl::interval_set<gid_t> its_gid_set;
      |                                     ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/vsomeip3.dir/build.make:1224: CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o] Error 1
gmake[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/../include/../../configuration/include/configuration.hpp:16,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/../include/routing_manager_base.hpp:28,
                 from /home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/../include/routing_manager_client.hpp:19,
[... cut ...] date_security_credentials_command const&)':
/home/ssmoogen/rpmbuild/BUILD/vsomeip-3.3.0/implementation/routing/src/routing_manager_client.cpp:2812:44: note: at offset 40 into destination object 'its_gid_set' of size 48
 2812 |         boost::icl::interval_set<uint32_t> its_gid_set;
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors

```

_Originally posted by @smooge in https://github.com/COVESA/vsomeip/issues/440#issuecomment-1551937209_

smooge, 2023-05-18:
I am able to compile the code without errors with gcc-11 and gcc-12. It is the newer gcc-13 which is in Fedora 38 which brings up the warning. 

Working EL9 set

```
$ rpm -q gcc-c++ boost libstdc++
gcc-c++-11.3.1-4.4.el9.x86_64
boost-1.75.0-8.el9.x86_64
libstdc++-11.3.1-4.4.el9.x86_64
```

Non-Working F38 set

```
$ rpm -q gcc-c++ boost libstdc++
gcc-c++-13.1.1-2.fc38.x86_64
boost-1.78.0-11.fc38.x86_64
libstdc++-13.1.1-2.fc38.x86_64
```

The problem stems from the settings in CMakeLists.txt

```
 76     set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -    fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
```

I missed it first because there is a -Wformat-security earlier in the line. I was able to get past the problem with gcc-13 (aka it compiles and runs but the code probably needs to be fixed) with consolidating the warnings together and removing the -Werror

```
76     set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -Wpedantic -fexceptions -fstrict-aliasing -fstack-protect    or-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -fPIE")
 
```

goncaloalmeida, 2023-05-19:
@smooge if the problem is solved, please close the ticket.

closed 2023-05-19 by smooge