COVESA/vsomeip #450: Cannot publish same payload repeatedly.

MoraeFox, 2023-05-19:
Hello community

I managed to let two different hosts communicate each other using notify-sample.cpp and subscribe-sample.cpp.
But I noticed that the payload should always changed to publish like below. 
Otherwise it says [info] Cannot set payload for event [1111.1111.1111]. It isn't provided

`            while (is_offered_ && running_) {

                if(DATA_1){
                    its_data[0] = { 0xDF };
                    its_data[1] = { 0xAD};
                }
                else{
                    its_data[0] = { 0x4A };
                    its_data[1] = { 0xCA};
                }

                {
                    std::lock_guard<std::mutex> its_lock(payload_mutex_);
                    payload_->set_data(its_data, its_size);

                    std::cout << "Setting event (Length=" << std::dec << its_size << ")." << std::endl;
                    app_->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload_);
                }
                DATA_1 = !DATA_1;

                std::this_thread::sleep_for(std::chrono::milliseconds(cycle_));
            }`
            
Can I ask why this is happening and if there is Autosar requirements for this?

Thanks in advance!

MoraeFox, 2023-05-19:
As far as 
https://www.autosar.org/fileadmin/standards/R1-3/FO/AUTOSAR_PRS_SOMEIPProtocol.pdf
concern (part 4.2.4.1), publishing on change is just one of strategies.

Still don't know how many strategies are there is vsomeip and how to, but I'll keep digging and if there is no comment, I'll close this issue in couple of days

goncaloalmeida, 2023-05-19:
hello @MoraeFox we will investigate also this problem.
Thank you

MoraeFox, 2023-05-22:
Thanks.
As the function prepare_update_payload_unlocked( ); has design aspect of detecting changes of payloads (on 3.3.0), I see no problem with this and closing this issue.

fcmonteiro, 2023-05-22:
Hi @MoraeFox ,
You can pass the force flag to the notify forcing it to always send the same payload.
`app_->notify(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_EVENT_ID, payload_, true);`

Regarding that log message, it shouldn't appear.
We have some fixes for the set_payload function that should be available in our next release.

MoraeFox, 2023-05-23:
Hi @fcmonteiro : )
Force option in the end of the notify( ); works fine and send same payload repeatedly.
Thanks!