COVESA/vsomeip #459: V3.3.8 build fails with MSVC

carCV, 2023-06-06:
Env:
Win10
CMAKE：3.26.3
VS2022：MSVC143
Boost：1.79

cmake command:
cmake -G "NMake Makefiles" -DENABLE_SIGNAL_HANDLING=1 -DDIAGNOSIS_ADDRESS=0x10 -DBOOST_ROOT="D:/software/Boost/179" -DCMAKE_INSTALL_PREFIX="D:/software/SOMEIP/vsomeip388" ../

nmake error: 
D:\software\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp(215): error C2065: “udp_server_endpoint_impl_receive_op”: undeclared identifier
D:\software\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp(215): error C2146: syntax error: Lack of ";" (before the identifier "its operation")
D:\software\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp(215): error C2065: “its_operation”: undeclared identifier
D:\software\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp(234): error C2065: “its_operation”: undeclared identifier
NMAKE : fatal error U1077: “D:\software\CMake\cmake-3.26.3\bin\cmake.exe -E cmake_cl_compile_depends --dep-file=CMakeFiles\vsomeip3.dir\implementation\endpoints\src\udp_server_endpoint_impl.cpp.obj.d --working-dir=D:\software\vsomeip\build --filter-prefix=" Attention: include file:  " -- D:\software\MICROS~2\2022\COMMUN~1\VC\Tools\MSVC\1436~1.325\bin\Hostx64\x86\cl.exe @C:\Users\JIAMIN~1.LIN\AppData\Local\Temp\nmA7ED.tmp”: return code “0x2”
Stop.
NMAKE : fatal error U1077: “"D:\software\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.36.32532\bin\HostX64\x86\nmake.exe" -s -f CMakeFiles\vsomeip3.dir\build.make /nologo -SL                 CMakeFiles\vsomeip3.dir\build”: return code “0x2”
Stop.
NMAKE : fatal error U1077: “"D:\software\Microsoft Visual Studio\2022\Community\VC\Tools\MSVC\14.36.32532\bin\HostX64\x86\nmake.exe" -s -f CMakeFiles\Makefile2 /nologo -S                  all”: return code “0x2”
Stop.

goncaloalmeida, 2023-06-06:
@carCV in our Windows env. the build is **working fine**.
We are using Boost 1.65:
 -   BOOST_INCLUDEDIR=---\boost_1_65_0
 -   BOOST_LIBRARYDIR=---\boost_1_65_0\lib64-msvc-14.1
and without any "NMake Makefiles".

We will try to have a look if possible reproduce the exact env. that you have.

carCV, 2023-06-06:
it seem that `implementation/endpoints/include/udp_server_endpoint_impl_receive_op.hpp` header file Not applicable to windows platform

> @carCV in our Windows env. the build is **working fine**. We are using Boost 1.65:
> 
> * BOOST_INCLUDEDIR=---\boost_1_65_0
> * BOOST_LIBRARYDIR=---\boost_1_65_0\lib64-msvc-14.1
>   and without any "NMake Makefiles".
> 
> We will try to have a look if possible reproduce the exact env. that you have.

it seem that `implementation/endpoints.include/udp_server_endpoint_impl_receive_op.hpp` header file Not applicable to windows platform

carCV, 2023-06-06:
it seem that `implementation/endpoints.include/udp_server_endpoint_impl_receive_op.hpp` header file Not applicable to windows platform

goncaloalmeida, 2023-06-06:
@carCV could yu please try boost 1.65.0 in your env.?

carCV, 2023-06-07:
D:\software\vsomeip\build>cmake -G "NMake Makefiles" -DENABLE_SIGNAL_HANDLING=1 -DDIAGNOSIS_ADDRESS=0x10 -DBOOST_ROOT="D:/software/Boost/165" -DCMAKE_INSTALL_PREFIX="D:/software/SOMEIP/vsomeip388" ..
-- The C compiler identification is MSVC 19.36.32532.0
-- The CXX compiler identification is MSVC 19.36.32532.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/software/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x86/cl.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: D:/software/Microsoft Visual Studio/2022/Community/VC/Tools/MSVC/14.36.32532/bin/Hostx64/x86/cl.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found
-- Found Threads: TRUE
CMake Error at D:/software/CMake/cmake-3.26.3/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:230 (message):
  Could NOT find Boost (missing: system thread filesystem) (found suitable
  version "1.65.1", minimum required is "1.55")
Call Stack (most recent call first):
  D:/software/CMake/cmake-3.26.3/share/cmake-3.26/Modules/FindPackageHandleStandardArgs.cmake:600 (_FPHSA_FAILURE_MESSAGE)
  D:/software/CMake/cmake-3.26.3/share/cmake-3.26/Modules/FindBoost.cmake:2377 (find_package_handle_standard_args)
  CMakeLists.txt:161 (find_package)

goncaloalmeida, 2023-06-07:
you need to install all the necessary packages:
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - not found

miaozzZ, 2023-06-08:
I have the same issue on win10 vs2022 boost1.81

1>udp_server_endpoint_impl.cpp
1>D:\code\someip\vsomeip\implementation\endpoints\src\udp_server_endpoint_impl.cpp(215,9): error C2065: “udp_server_endpoint_impl_receive_op”: undeclared identifier

because the header file does not support windows when boost version >= 1.66
vsomeip\implementation\endpoints\include\udp_server_endpoint_impl_receive_op.hpp

```
#ifndef VSOMEIP_V3_UDP_SERVER_ENDPOINT_IMPL_RECEIVE_OP_HPP_
#define VSOMEIP_V3_UDP_SERVER_ENDPOINT_IMPL_RECEIVE_OP_HPP_

#if VSOMEIP_BOOST_VERSION >= 106600
#if defined(__linux__) || defined(ANDROID)

......

struct udp_server_endpoint_impl_receive_op {
    ......
```
Is that a plan to fix this ?

sebastian-bergt, 2023-06-16:
Stuck with the same issue.
MSVC 17
Boost 1.71.0

sebastianbergt, 2023-07-20:
[remove_linux_specific_header.zip](https://github.com/COVESA/vsomeip/files/12110329/remove_linux_specific_header.zip)
After some trial and error, I got it to compile for windows with this patch. This is however removing the additional error handling and potential retry logic the `udp_server_endpoint_impl_receive_op` introduced. I have not had a chance to test it with multicast in production yet, but will keep you updated.

nhelou, 2023-07-31:
Is there an official fix planed? It's an unfortunate situation not to be able to use the latest build for such a long time. We have other dependencies to boost 1.76, so being forced to go back to 1.65 is not really an option.

goncaloalmeida, 2023-09-07:
@nhelou not yet. We working on this, but we need to adapt our internal jobs to use 1.76 instead of 1.65 (the official release that we use).

goncaloalmeida, 2023-11-29:
Issue fixed in the last release (3.4.10)