COVESA/vsomeip #466: Not able to run Hello_World_Service and Hello_World_Client. Showing error related Configuration module could not be loaded.

manuunam8888, 2023-06-14:
Hi Everyone,

I have followed this steps as below and facing error like "**1 Configuration module could not be loaded!**" . Please help me resolve this issue.
Commands ran before this:
   **_mkdir build
   cd build/
   cmake ..
   make
   sudo make install
   cd ../examples/hello_world/
   mkdir build
   cd build/
   cmake ..
   make_** 

Error found at:
~/Manohar/vsomeip-master/examples/hello_world/build$ _VSOMEIP_CONFIGURATION=../helloworld-local.json \
> VSOMEIP_APPLICATION_NAME=hello_world_service \
> ./hello_world_service_
**1 Configuration module could not be loaded!**

reymor, 2023-06-14:
Hi! As fcmonteiro point out, this is related with the path of your libraries. When you compile the libraries you need to install them, this is done with 

`make install`

Your dynamic linker need to be able to find these symbols. You could check that your vsomeip libraries are located by ld using 

`
ldconfig - v | grep vsomeip
`

The output should be something like this: 

[image]

If the path where the libraries are installed is not added to the dynamic linker, the solution could be added this to 

`
/etc/ld.so.conf.d/ <--- here you create a file with .conf extension which is the path where your libs are installed
`

after you did that, you rebuild the cache of library filename, by running;

`
ldconfig
`

---- The other solution could be just append the path where the libraries are installed to the  ---- 

`
export LD_LIBRARY_PATH=/path/to/vsomeip/libraries
`

 before you run your program. Or by a script as fcmonteiro  said.

manuunam8888, 2023-06-15:
Thanks it worked

closed 2023-06-15 by manuunam8888

goncaloalmeida, 2023-06-23:
https://github.com/COVESA/vsomeip/issues/431#issuecomment-1584785650

TienVuHong, 2024-06-07:
> Hi! As fcmonteiro point out, this is related with the path of your libraries. When you compile the libraries you need to install them, this is done with
>
> `make install`
>
> Your dynamic linker need to [...]

That help,
Thanks