COVESA/vsomeip #468: Cannot run helloWorld "1 Configuration module could not be loaded!"

ArslanMunaim, 2023-06-16:
Hello, I am following wiki I setup boost 1_82_0, Then clonned vsome/ip, followed the steps and build it: 

$ mkdir build
$ cd build
$ cmake -DCMAKE_INSTALL_PREFIX:PATH=/usr/local/ ..
$ cmake -DENABLE_SIGNAL_HANDLING=1 
$ make
$ make install

$ cd ../helloworld
$ cmake CMakeLists.txt 
$ make
$ ./hello_world_client
$ sudo ldconfig
$ sudo ldconfig -p | grep libvsomeip

Now I am trying to build helloWorld example but it is not running every time i got this message "**1 Configuration module could not be loaded**!" 

I tried bunch of things, for example running sudo ldconfig giving path to library but still nothing, can anybody help me in resolving that issue?

Thanks

fcmonteiro, 2023-06-16:
Please see https://github.com/COVESA/vsomeip/issues/466

ArslanMunaim, 2023-06-16:
yeah, i tried that but still no leads 

> Please see #466

reymor, 2023-06-16:
mmmm, it is really weird. Could please show the outputs/more information of the commands? There is not black magic behind this. You have a dynamic linker and this dynamic linker is looking for the libraries in paths that your system/you provide.

Where did you install the boost libraries? 

For a rapid testing, try what fcmonteiro suggest. If it does not work, execute this command and please show the outputs: 

`ldconfig -p | grep vsomeip`

if this outputs shows information, try ldd over the libvsomeip3-cfg.so and check that every library has a correct link, for instace:

`
ldd /path/to/library/libvsomeip3-cfg.so
`
In this case, I had linked the boots 1.74 but I did not have this on my path or ld.so directory, so it shows me: 

[image]

This will also arise the same problem that you are talking. So, in this case what I need to do is add the path where my boost libraries are installed.

This shall resolve any problem that you have.

ArslanMunaim, 2023-06-16:
I installed the Boost library at home dir, I clonned vsome/ip in boost folder, build it there and try to build helloworld from examples.

while running ldconfig -p | grep vsomeip i get this results
	libvsomeip3.so.3 (libc6,x86-64) => /usr/local/lib/libvsomeip3.so.3
	libvsomeip3.so (libc6,x86-64) => /usr/local/lib/libvsomeip3.so
	libvsomeip3-sd.so.3 (libc6,x86-64) => /usr/local/lib/libvsomeip3-sd.so.3
	libvsomeip3-sd.so (libc6,x86-64) => /usr/local/lib/libvsomeip3-sd.so
	libvsomeip3-e2e.so.3 (libc6,x86-64) => /usr/local/lib/libvsomeip3-e2e.so.3
	libvsomeip3-e2e.so (libc6,x86-64) => /usr/local/lib/libvsomeip3-e2e.so
	libvsomeip3-cfg.so.3 (libc6,x86-64) => /usr/local/lib/libvsomeip3-cfg.so.3
	libvsomeip3-cfg.so (libc6,x86-64) => /usr/local/lib/libvsomeip3-cfg.so

reymor, 2023-06-16:
What about the output of ldd over cfg module ?

ArslanMunaim, 2023-06-16:
i am getting this output:

linux-vdso.so.1 (0x00007ffc9f397000)
	libvsomeip3.so.3 => /local/home/munaim/boost_1_82_0/vsomeip/build/libvsomeip3.so.3 (0x00007eff12d4a000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007eff12d0b000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007eff12ce8000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007eff12b06000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007eff12aeb000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007eff128f9000)
	libboost_thread.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007eff128cb000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007eff128c5000)
	/lib64/ld-linux-x86-64.so.2 (0x00007eff13066000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007eff12776000)

reymor, 2023-06-16:
You said that you used 1_82_0, but if you see your outputs it is pointing to at system level of a boost library 1_71_0. Check which is the library that you used for the compilation

If you confirm that it is 1_82_0. Just make the configuration of the path according to that.

ArslanMunaim, 2023-06-16:
okay i will have a look at that 
Thanks

ArslanMunaim, 2023-06-16:
Hello again, 

Yeah, i tried that but still no luck 

linux-vdso.so.1 (0x00007ffcb3dfc000)
	libvsomeip3.so.3 => /local/home/munaim/vsomeip/build/libvsomeip3.so.3 (0x00007f397d628000)
	libboost_filesystem.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.71.0 (0x00007f397d5e9000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f397d5c6000)
	libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f397d3e4000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f397d3c9000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f397d1d7000)
	libboost_thread.so.1.71.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.71.0 (0x00007f397d1a9000)
	libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f397d1a3000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f397d940000)

still the same problem

reymor, 2023-06-16:
What exactly did you do?

ArslanMunaim, 2023-06-16:
So, i tried installing boost 1.71.0, tried to look into library paths, build vsomeip project again, but still same error

reymor, 2023-06-16:
For real, this is strange. What comes to mind is that you have done something wrong in the process. I recommend starting from a clean installation, and paying attention to every detail. 

1) Install the boost that you want to use (e.g 1.71)
2) Clone and follow installations steps

btw, which distro are you using (ubuntu)?

ArslanMunaim, 2023-06-16:
Hi, Thanks I will try again,  yes, I am using Ubuntu.

closed 2023-06-21 by ArslanMunaim

Yoho-test, 2024-10-29:
> Hi, Thanks I will try again, yes, I am using Ubuntu.

Hi, have you had a solution to solve this issue?