COVESA/vsomeip #472: Are there any special reason why there are duplicated definition in application.hpp?

reymor, 2023-06-17:
Hi, 

I hope you are ok. As I stated in the title, are there any special reason why there are duplicated definition in [application.hpp](https://github.com/COVESA/vsomeip/blob/master/interface/vsomeip/application.hpp)? 

For instance: register_availability_handler()

https://github.com/COVESA/vsomeip/blob/9d8aa22243243c617c97e79de26bbd36d9592429/interface/vsomeip/application.hpp#L542-L567

https://github.com/COVESA/vsomeip/blob/9d8aa22243243c617c97e79de26bbd36d9592429/interface/vsomeip/application.hpp#L1031-L1056

Maybe there are technical details which I am not able to understand. If there is not special reason, I could prepare a PR. 

Thanks in advance.

lutzbichler, 2023-06-23:
The additional definitions are done to add extended versions of an API method while preserving ABI compatibility, which we need to do as long as we do not change the major version.

reymor, 2023-06-23:
Thanks for your answer @lutzbichler. Sorry for this, but I would like to understand more about this. Is there any material that you would recommend me? 

In this case, both methods had the same _signature_ , so API is the same, no?. From the beginning I suspect that it was something related with ABI compatibility, but I do not know in which ways it is affected. 

Thanks in advance

goncaloalmeida, 2023-10-27:
@reymor what type of material? C++ guidelines/books?

closed 2023-10-27 by goncaloalmeida