COVESA/vsomeip #474: Discovering services which are not known in advance

reymor, 2023-06-19:
**Version**: 3.3.8

Sorry for the spam but the different questions are not related. I will say that  I have used vsomeip in certain cases in which this was not needed. However, I was trying to test this: 

- A node in which there are services offered by the Service Discovery,  I connect a new node which does not know which services are offered in this node, I assumed that this could be management by an availability_handler. However, this handler had: 

`
typedef std::function< void (service_t, instance_t, bool) > availability_handler_t;
`

The avaibility handler always is called when a new node offered a service. However, if there was not called in advance in application like subscribe or request_service, the bool y pass as false. So, how could I know if this new service is present or not in the node? 

I mean: 

1. Register the service in advance (with request_service): no problems, when the node is connected is pass as available (true), and if the node is disconnected then this handler is called and we got available (false). 
2. No registration of the service (no request, just availability handler). The handler is called in both cases with false. So, we cannot known which services are offered in the node and are able to subscribe. 

The use of case that I had in mid is: There is a node in which is connected this hypothetical program and this node does not known anything about the services offered in the node, once we discover the services, it shows the services and it could subscribe to these services. 

Kind of trace the different calls and I got that, the availability_state is pass as AS_OFFERED from routing_manager_impl.cpp#L2490. After different differents steps, it got until this point:

https://github.com/COVESA/vsomeip/blob/9d8aa22243243c617c97e79de26bbd36d9592429/implementation/runtime/src/application_impl.cpp#L908-L912

In which, we just pass true in cases where the availabity_states is equal to AS_AVAILABLE and if the other states are filter out. 

https://github.com/COVESA/vsomeip/blob/9d8aa22243243c617c97e79de26bbd36d9592429/interface/vsomeip/enumeration_types.hpp#L91-L96

I'm not really sure if there is other way of doing what I want to do. Also, I'm not an expert in vsomeip source code and behavior of the protocol. So, any help will be appreciate. 

The code that I'm using is something like this (simplified): 

```cpp
app_->register_availability_handler(
  IN_SERVICE_MSG3_ID,
  IN_INSTANCE_MSG3_ID,
  std::bind(&someip_service::on_availability, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available)
{  
  std::cout << "Service ["
                  << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
                  << "] is "
                  << (_is_available ? "available." : "NOT available.")
                  << std::endl;        
}
```
When I init the app, I got a output like: 

`
Service [0200.1] is NOT available.
[info] REMOTE SUBSCRIBE(0000) --> a connected a node
Service [0200.1] is NOT available.
`

Thanks in advance

duartenfonseca, 2025-04-08:
hi @reymor does this still happens on 3.5.5 ?

reymor, 2025-04-08:
Hello @duartenfonseca 

I will test this

closed 2025-11-19 by fcmonteiro