COVESA/vsomeip #480: 1 Configuration module could not be loaded !

devashishmishra90, 2023-06-23:
[image]

goncaloalmeida, 2023-06-23:
Please check https://github.com/COVESA/vsomeip/issues/468

closed 2023-06-23 by goncaloalmeida

devashishmishra90, 2023-06-23:
**i cloned vsomeip into home directory then,**
mkdir build
cd build
cmake ..
make
make install

**after that i go to** 
vsomeip/examples/helloworld
mkdir build
cd build
cmake ..
make

& **RUN THIS**
VSOMEIP_CONFIGURATION=../helloworld-local.json \
VSOMEIP_APPLICATION_NAME=hello_world_service \
./hello_world_service

still getting the problem after looking into #466

goncaloalmeida, 2023-06-23:
https://github.com/COVESA/vsomeip/issues/431#issuecomment-1584785650

devashishmishra90, 2023-06-23:
Hey mate, I also did whatever was mentioned in #431 but still the issue is there

hmluqman, 2024-07-06:
In my case, LD_LIBRARY_PATH was missing in my Linux host system. It was first time I was compiling C++.