COVESA/vsomeip #481: 1 Configuration module could not be loaded!

devashishmishra90, 2023-06-24:
I don't know whats happening.
but for last 2 days i tried everything to fix this.

Still no any positive sign.

tried every method on this

devashishmishra90, 2023-06-24:
this is output of sudo make install 
devashish@devashish-VirtualBox:~/SOMEIP_NEW/vsomeip/build$ sudo make install

Consolidate compiler generated dependencies of target vsomeip3

[ 70%] Built target vsomeip3

[ 73%] Built target vsomeip3-cfg

[ 89%] Built target vsomeip3-sd

[ 98%] Built target vsomeip3-e2e

[100%] Built target routingmanagerd

Install the project...

-- Install configuration: "RelWithDebInfo"

-- Up-to-date: /include/vsomeip/../compat/vsomeip/application.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/constants.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/defines.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/enumeration_types.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/error.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/export.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/function_types.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/handler.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/internal/deserializable.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/internal/serializable.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/message.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/message_base.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/payload.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/plugin.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/plugins/application_plugin.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/plugins/pre_configuration_plugin.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/primitive_types.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/runtime.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/trace.hpp

-- Up-to-date: /include/vsomeip/../compat/vsomeip/vsomeip.hpp

-- Up-to-date: /include/vsomeip/application.hpp

-- Up-to-date: /include/vsomeip/constants.hpp

-- Up-to-date: /include/vsomeip/defines.hpp

-- Up-to-date: /include/vsomeip/deprecated.hpp

-- Up-to-date: /include/vsomeip/enumeration_types.hpp

-- Up-to-date: /include/vsomeip/error.hpp

-- Up-to-date: /include/vsomeip/export.hpp

-- Up-to-date: /include/vsomeip/function_types.hpp

-- Up-to-date: /include/vsomeip/handler.hpp

-- Up-to-date: /include/vsomeip/internal/deserializable.hpp

-- Up-to-date: /include/vsomeip/internal/logger.hpp

-- Up-to-date: /include/vsomeip/internal/policy_manager.hpp

-- Up-to-date: /include/vsomeip/internal/serializable.hpp

-- Up-to-date: /include/vsomeip/message.hpp

-- Up-to-date: /include/vsomeip/message_base.hpp

-- Up-to-date: /include/vsomeip/payload.hpp

-- Up-to-date: /include/vsomeip/plugin.hpp

-- Up-to-date: /include/vsomeip/plugins/application_plugin.hpp

-- Up-to-date: /include/vsomeip/plugins/pre_configuration_plugin.hpp

-- Up-to-date: /include/vsomeip/primitive_types.hpp

-- Up-to-date: /include/vsomeip/runtime.hpp

-- Up-to-date: /include/vsomeip/structured_types.hpp

-- Up-to-date: /include/vsomeip/trace.hpp

-- Up-to-date: /include/vsomeip/vsomeip.hpp

-- Up-to-date: /include/vsomeip/vsomeip_sec.h

-- Installing: /lib/libvsomeip3.so.3.3.8

-- Up-to-date: /lib/libvsomeip3.so.3

-- Up-to-date: /lib/libvsomeip3.so

-- Installing: /lib/libvsomeip3-e2e.so.3.3.8

-- Up-to-date: /lib/libvsomeip3-e2e.so.3

-- Set runtime path of "/lib/libvsomeip3-e2e.so.3.3.8" to ""

-- Up-to-date: /lib/libvsomeip3-e2e.so

-- Installing: /lib/libvsomeip3-cfg.so.3.3.8

-- Up-to-date: /lib/libvsomeip3-cfg.so.3

-- Set runtime path of "/lib/libvsomeip3-cfg.so.3.3.8" to ""

-- Up-to-date: /lib/libvsomeip3-cfg.so

-- Installing: /lib/libvsomeip3-sd.so.3.3.8

-- Up-to-date: /lib/libvsomeip3-sd.so.3

-- Set runtime path of "/lib/libvsomeip3-sd.so.3.3.8" to ""

-- Up-to-date: /lib/libvsomeip3-sd.so

-- Up-to-date: /etc/vsomeip/vsomeip.json

-- Up-to-date: /etc/vsomeip/vsomeip-local.json

-- Up-to-date: /etc/vsomeip/vsomeip-tcp-client.json

-- Up-to-date: /etc/vsomeip/vsomeip-tcp-service.json

-- Up-to-date: /etc/vsomeip/vsomeip-udp-client.json

-- Up-to-date: /etc/vsomeip/vsomeip-udp-service.json

-- Up-to-date: /lib/cmake/vsomeip3/vsomeip3Config.cmake

-- Up-to-date: /lib/cmake/vsomeip3/vsomeip3ConfigVersion.cmake

-- Up-to-date: /lib/cmake/vsomeip3/vsomeip3Targets.cmake

-- Up-to-date: /lib/cmake/vsomeip3/vsomeip3Targets-relwithdebinfo.cmake

-- Up-to-date: /lib/pkgconfig/vsomeip3.pc

chrsch, 2023-06-25:
@devashishmishra90 You need to build using cmake and variable ENABLE_MULTIPLE_ROUTING_MANAGERS.

Use the var for building vsomeip itself and also for examles/hello_world. 

`mkdir build && cd build && cmake -DENABLE_MULTIPLE_ROUTING_MANAGERS=1 .. && make`

devashishmishra90, 2023-06-26:
I tried this but not working this anymore.

t14z, 2023-06-26:
你好！你的邮件已收到

devashishmishra90, 2023-06-26:
@mohitraopathru 
[image]

BrunoLSilvaCTW, 2023-06-26:
Hi @devashishmishra90. Are you running `sudo ldconfig` to update the links to the shared libraries?

mohitraopathru, 2023-06-26:
https://github.com/COVESA/vsomeip/issues/393#issue-1476518234

devashishmishra90, 2023-06-26:
[image]
@BrunoLSilvaCTW

mohitraopathru, 2023-06-26:
If LD library path is set already then also run that LD_LIBRARY_PATH command and after that run 
Sudo ldconfig

chrsch, 2023-06-26:
> TRY THIS COMMAND FIRST YOU NEED TO SET LIB PATH export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/PATH_TO_LIB PATH_TO_LIB :- in ubuntu it's usually/use/local/lib

This also made i work for me without the ENABLE_MULTIPLE_ROUTING_MANAGERS flag.

fcmonteiro, 2023-06-26:
@devashishmishra90 call the hello_world_service, with something like this,

VSOMEIP_CONFIGURATION=./config/vsomeip-local.json VSOMEIP_APPLICATION_NAME=service-sample LD_LIBRARY_PATH=/home/ubuntu/Projects/install_folder/lib:build ./notify-sample

reymor, 2023-06-26:
It is not enough having vsomeip-cfg in ldconfig path. Also, you have to add to the boost library path. 

@devashishmishra90 If you follow the steps in https://github.com/COVESA/vsomeip/issues/468 you shall get a working library. 

Could please give me the output from ldd over de cfg library ? 

`
ldd /path/to/library/libvsomeip3-cfg.so
`

rvautard, 2023-07-05:
Same issue on my side:
ldd /usr/local/lib/libvsomeip3-cfg.so
        linux-vdso.so.1 (0x00007ffdbcffa000)
        libvsomeip3.so.3 => /usr/local/lib/libvsomeip3.so.3 (0x00007f9871800000)
        libboost_filesystem.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_filesystem.so.1.74.0 (0x00007f9871b21000)
        libstdc++.so.6 => /lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f9871400000)
        libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f9871b01000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f9871000000)
        libboost_thread.so.1.74.0 => /lib/x86_64-linux-gnu/libboost_thread.so.1.74.0 (0x00007f9871add000)
        libsystemd.so.0 => /lib/x86_64-linux-gnu/libsystemd.so.0 (0x00007f9871739000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f9871bc4000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f9871652000)
        liblzma.so.5 => /lib/x86_64-linux-gnu/liblzma.so.5 (0x00007f9871ab2000)
        libzstd.so.1 => /lib/x86_64-linux-gnu/libzstd.so.1 (0x00007f9871331000)
        liblz4.so.1 => /lib/x86_64-linux-gnu/liblz4.so.1 (0x00007f9871632000)
        libcap.so.2 => /lib/x86_64-linux-gnu/libcap.so.2 (0x00007f9871aa5000)
        libgcrypt.so.20 => /lib/x86_64-linux-gnu/libgcrypt.so.20 (0x00007f9870ec2000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007f987130b000)

xuantambk, 2023-08-05:
after built success..
sudo make install will install all lib in /usr/local/lib..
from that it will create symbol link  for *.so.
Check your .usr/local/lib after sudo make install
after that clean helloworl --> rebuild it with new lib in /usr/local/lib 
and run it again...

closed 2023-09-01 by fcmonteiro

WSYGJR, 2024-02-29:
I have the same question!!! I tried the steps, but no one can work. my vsomeip-cfg and boost are in the same path. they can link cause my ldconfig all have path. but it was error

t14z, 2024-02-29:
你好！你的邮件已收到