COVESA/vsomeip #482: wireshark Packet capturing

devashishmishra90, 2023-06-27:


fcmonteiro, 2023-06-27:
Not possible running both service and client on the same machine.
However, you can run one of then in a docker or vm and capture the traffic in that interface.
Thanks

devashishmishra90, 2023-06-27:
what if i have 2 VMs & installed vsomeip on both machine

fcmonteiro, 2023-06-27:
Should also work, if the network part is working properly between the two VMs.

devashishmishra90, 2023-06-27:
Just assume i'm running service.cpp on VM A, & client.cpp on VM B.
here how i will set the IP of service & client. coz both have different IP

fcmonteiro, 2023-06-27:
You should have a config file for each app, and edit the field unicast with the IP of the VM.

devashishmishra90, 2023-06-27:
Hello,
on VM A[192.168.56.101] , i build the project with cmake -DUNICAST_ADDRESS=192.168.56.101 .. then run make, make install
& then go to example & configure helloworld-local.json & put 192.168.56.101 in unicast section 
then i run hello_world_service.

on VM B[192.56.102], i did the same thing, & run hello_world_client

but i can't see  any packet on VM A.

what i'm doing wrong ?

goncaloalmeida, 2023-10-02:
@devashishmishra90 you still have this issue?
cc @fcmonteiro

closed 2023-10-23 by goncaloalmeida