COVESA/vsomeip #483: listen socket is not closed

wenking, 2023-06-27:
The transport layer protocol of the service is TCP, if client process exit and if some response message in queue in routing process and not send,then stop server process. you will find the listen socket is not closed

fcmonteiro, 2023-06-27:
Can you add more details to the issue, and how we can reproduce it.
Thanks

wenking, 2023-06-28:
1:the service transport layer protocol   is tcp.
2:client loop send request message
3:service send response message
4:stop client process
5:stop service process
6:check routing processs socket, you will find the listen socket is still active and routing SD always send offer service message of the service

goncaloalmeida, 2023-09-07:
@wenking we are working on a possible fix for this problem.
We will keep the ticket updated