COVESA/vsomeip #486: Communication between 2 RPIs

devashishmishra90, 2023-06-29:
Hello Everyone.
I have 2 Raspberry Pis & installed vsomeip on both of them.
I have mentioned all the steps below, **but I can't communicate between them.**
& **didn't find packets on wireshark** that mentioned SOMEIP messages are being communicated.
Steps-
both RPI are connected to wifi router.
both have IP address- 192.168.29.110 RPI A, 192.168.29.202 RPI B
**BUILD PROCESS [on both RPI i followed ]**
1. cd vsomeip && mkdir build && cd build
2.  for RPI A side, cmake -DUNICAST_ADDRESS=192.168.29.110 
3.  for RPI B side, cmake -DUNICAST_ADDRESS=192.168.29.202
4.  now on RPI A side, after running cmake .. && make in examples/helloworld/build. configure hello-world-local.json & put unicast address to 192.168.29.110 & netmask to 255.255.255.0
5. now on RPI B side, after running cmake .. && make in examples/helloworld/build. configure hello-world-local.json & put unicast address to 192.168.29.202 & netmask to 255.255.255.0
6. when i run hello_world_service on RPI A side & hello_world_client on RPI B side.

then I don't see any communication in between these two, neithe in command window or wireshark.

what i'm doing wrong please let me know.
simple setup it is.

devashishmishra90, 2023-06-29:
@ArslanMunaim
Do you have any view on this

ArslanMunaim, 2023-06-30:
Hi, can you please share your config .json files??

devashishmishra90, 2023-07-03:
this is terminal output
[image]

devashishmishra90, 2023-07-03:
this json file 
[image]

ArslanMunaim, 2023-07-03:
Hi, as you mentioned you have to communicate between different RPi's you have to write config file for each RPi, You have to mention your unicast address and net mask, also updating application name and routing tags. I would suggest you to consult https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes Communication between two devices part.

devashishmishra90, 2023-07-06:
when i'm running this I'm getting this one, & after that nothing on wireshark, only one time packet on wireshark.
[image]
Doubts related to routing

dprogm, 2023-07-10:
@devashishmishra90 Any update on this? We are facing exactly the same issue and have no clue what can be wrong in our setup or configuration. We are trying to connect from a linux host to an embedded device. :sunglasses:

ogkang1201, 2023-10-10:
did you add the route on your device? im facing same problem with you too

jsb9945, 2023-11-14:
Did you solve it? I have the same problem.

closed 2024-11-04 by duartenfonseca