COVESA/vsomeip #487: make error

mominux, 2023-06-30:
Hello ,
I'm trying to make Vsomeip from the source [3.3.8](https://github.com/COVESA/vsomeip/releases/tag/3.3.8) 

first i use this command : 
```
$ cmake -DENABLE_SIGNAL_HANDLING=1 ..

-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Setting build type to 'RelWithDebInfo' as none was specified.
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/local/include (found suitable version "1.55.0", minimum required is "1.55") found components: system thread filesystem chrono date_time atomic 
-- Using boost version: 105500
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'automotive-dlt >= 2.11'
--   No package 'automotive-dlt' found
-- Checking for module 'libsystemd'
--   Found libsystemd, version 245
-- Default configuration folder: /etc/vsomeip
-- Default configuration file: /etc/vsomeip.json
Predefined base path: /tmp
Predefined unicast address: 127.0.0.1
Predefined diagnosis address: 0x01
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.17") found components: doxygen dot 
dot found
asciidoc found
source-highlight found
CMake Warning at CMakeLists.txt:628 (find_package):
  By not providing "Findbenchmark.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "benchmark", but CMake did not find one.

  Could not find a package configuration file provided by "benchmark" with
  any of the following names:

    benchmarkConfig.cmake
    benchmark-config.cmake

  Add the installation prefix of "benchmark" to CMAKE_PREFIX_PATH or set
  "benchmark_DIR" to a directory containing one of the above files.  If
  "benchmark" provides a separate development package or SDK, be sure it has
  been installed.

-- GTEST_ROOT is not defined. For building the tests the variable
             GTEST_ROOT has to be defined. Tests can not be built.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/build
```
after make i got below errors : 
```
$ make -j12

Scanning dependencies of target vsomeip3
[  1%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_definition.cpp.o
[  3%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/credentials.cpp.o
[  3%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.o
[  3%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_impl.cpp.o
[  4%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_manager_base.cpp.o
[  5%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/endpoint_manager_impl.cpp.o
[  6%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/local_tcp_server_endpoint_impl.cpp.o
[  6%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/local_tcp_client_endpoint_impl.cpp.o
[  7%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/local_uds_client_endpoint_impl.cpp.o
[  8%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/netlink_connector.cpp.o
[  8%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/local_uds_server_endpoint_impl.cpp.o
[  9%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/server_endpoint_impl.cpp.o
[ 10%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/tcp_client_endpoint_impl.cpp.o
[ 10%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/tcp_server_endpoint_impl.cpp.o
/home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/server_endpoint_impl.cpp:841:57: error: ‘stream_protocol’ is not a member of ‘boost::asio::local’; did you mean ‘stream_protocol_ext’?
  841 | template class server_endpoint_impl<boost::asio::local::stream_protocol>;
      |                                                         ^~~~~~~~~~~~~~~
      |                                                         stream_protocol_ext
/home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/server_endpoint_impl.cpp:841:57: error: ‘stream_protocol’ is not a member of ‘boost::asio::local’; did you mean ‘stream_protocol_ext’?
  841 | template class server_endpoint_impl<boost::asio::local::stream_protocol>;
      |                                                         ^~~~~~~~~~~~~~~
      |                                                         stream_protocol_ext
/home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/server_endpoint_impl.cpp:841:72: error: template argument 1 is invalid
  841 | template class server_endpoint_impl<boost::asio::local::stream_protocol>;
      |                                                                        ^
[ 11%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/tp.cpp.o
[ 12%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/tp_message.cpp.o
make[2]: *** [CMakeFiles/vsomeip3.dir/build.make:206: CMakeFiles/vsomeip3.dir/implementation/endpoints/src/server_endpoint_impl.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:209: CMakeFiles/vsomeip3.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

reymor, 2023-07-01:
Hello @mominux

You are using boost 1.55.0. This version is not supported in current version of vsomeip. You could try two things:

1) Upgrade your boost library (e.g 1.74, 1.65.0) 
2) Switch to branch maintain/3.1 which compile with boost 1.55.0

mominux, 2023-07-03:
> Hello @mominux
> 
> You are using boost 1.55.0. This version is not supported in current version of vsomeip (3.3.8). You could try two things:
> 
> 1. Upgrade your boost library (e.g 1.65.0, 1.74.0)
> 2. Switch to branch maintain/3.1 which compile with boost 1.55.0 or use [3.1.20.3](https://github.com/COVESA/vsomeip/releases/tag/3.1.20.3)

Hi @reymor .thank you very much. However, this does not work for me. Could you please list the additional dependencies versions for the current VsomeIP (3.3.8) version? 
Because, as you mentioned, I updated my boost to the most recent version, but now I'm getting more and more issues.

reymor, 2023-07-03:
I wil say that there is no more dependencies than:

cmake >= 3.13 
boost*
C++14 (gcc >= 4.8) **

*In branch maintain/3.1 it is clear that the supported boost are 1.55, 1.66, 1.70, 1.74, 1.76 and you shall use any of them. You could check this yourself in cmakelist and in implemetantion/helper. In addition, I will say that I have compiled without problem with 1.55, 1.66, 1.70, 1.74, and 1.74 in ubuntu env. In windows with 1.70, 1.74 and 1.76. I have not tested with any other version in windows. 

However, currently, the new versions (from 3.1 onwards). They try to increase compatibility with new versions of boost which is good. There is no helper functions any more. In theory, it means that we could use new and old versions of boost. In practice, this is not in that way. I had problems with make it works in windows enviroments. In Linux env one of them is realated with the one that you faced (boost version 1.55). 

** Also, I tried with gcc 13 and I got an error that I did not investigate, but I will say that with never versions of gcc it will not compile.

Could you please show me the error that you had? Which compiler are you using?

mominux, 2023-07-03:
The C compiler identification is GNU 9.4.0, The CXX compiler identification is GNU 9.4.0, Boost-1.82.0, automotive-dlt version 2.18.9, libsystemd version 245
i also switched to branch 3.3.8 and 3.1 and tried them . 

and the  errors :
,,,
usr/local/include/boost/asio/prefer.hpp:168:19: error: template argument 1 is invalid
  168 |     decay<Property>::type::is_preferable
      |                   ^
/usr/local/include/boost/asio/prefer.hpp:168:22: error: ‘<expression error>::type’ has not been declared
  168 |     decay<Property>::type::is_preferable
      |                      ^~~~
/usr/local/include/boost/asio/prefer.hpp:169:3: error: expected ‘(’ before ‘>’ token
  169 |   >::type,
      |   ^
      |   (
/usr/local/include/boost/asio/prefer.hpp:169:3: error: template argument 4 is invalid
/usr/local/include/boost/asio/prefer.hpp:169:3: error: template argument 5 is invalid
/usr/local/include/boost/asio/prefer.hpp:190:5: error: wrong number of template arguments (1, should be at least 2)
  190 |     >::value
      |     ^
In file included from /usr/local/include/boost/asio/execution/allocator.hpp:23,
                 from /usr/local/include/boost/asio/execution.hpp:18,
                 from /usr/local/include/boost/asio/io_context.hpp:27,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/buffer.hpp:18,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/client_endpoint_impl.hpp:23,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:20:
/usr/local/include/boost/asio/is_applicable_property.hpp:47:8: note: provided for ‘template<class T, class Property, class> struct boost::asio::is_applicable_property’
   47 | struct is_applicable_property :
      |        ^~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/local/include/boost/asio/execution/blocking.hpp:25,
                 from /usr/local/include/boost/asio/execution/any_executor.hpp:32,
                 from /usr/local/include/boost/asio/execution.hpp:19,
                 from /usr/local/include/boost/asio/io_context.hpp:27,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/buffer.hpp:18,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/client_endpoint_impl.hpp:23,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:20:
/usr/local/include/boost/asio/prefer.hpp:191:3: error: template argument 1 is invalid
  191 |   >::type,
      |   ^
/usr/local/include/boost/asio/prefer.hpp:193:5: error: ‘decay’ was not declared in this scope; did you mean ‘std::decay’?
  193 |     decay<Property>::type::is_preferable
      |     ^~~~~
      |     std::decay
In file included from /usr/include/c++/9/ratio:38,
                 from /usr/include/c++/9/chrono:38,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:6:
/usr/include/c++/9/type_traits:2004:11: note: ‘std::decay’ declared here
 2004 |     class decay
      |           ^~~~~
In file included from /usr/local/include/boost/asio/execution/blocking.hpp:25,
                 from /usr/local/include/boost/asio/execution/any_executor.hpp:32,
                 from /usr/local/include/boost/asio/execution.hpp:19,
                 from /usr/local/include/boost/asio/io_context.hpp:27,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/buffer.hpp:18,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/../include/client_endpoint_impl.hpp:23,
                 from /home/abdoli/workspce/ip-netwwork/some_ip/vsomeip/implementation/endpoints/src/client_endpoint_impl.cpp:20:
/usr/local/include/boost/asio/prefer.hpp:193:19: error: template argument 1 is invalid
  193 |     decay<Property>::type::is_preferable
      |                   ^
/usr/local/include/boost/asio/prefer.hpp:193:22: error: ‘<expression error>::type’ has not been declared
  193 |     decay<Property>::type::is_preferable
      |                      ^~~~
/usr/local/include/boost/asio/prefer.hpp:194:3: error: expected ‘(’ before ‘>’ token
  194 |   >::type,
      |   ^
      |   (
/usr/local/include/boost/asio/prefer.hpp:194:3: error: template argument 4 is invalid
/usr/local/include/boost/asio/prefer.hpp:194:3: error: template argument 5 is invalid
/usr/local/include/boost/asio/prefer.hpp:212:5: error: wrong number of template arguments (1, should be at least 2)
  212 |     >::value
[... 606 lines cut ...]
/usr/local/include/boost/asio/require.hpp:240:20: error: expected ‘;’ at end of member declaration
  240 |   typedef typename decay<
      |                    ^~~~~
      |                         ;
/usr/local/include/boost/asio/require.hpp:240:25: error: expected unqualified-id before ‘<’ token
  240 |   typedef typename decay<
      |                         ^
/usr/local/include/boost/asio/require.hpp:250:27: error: expected nested-name-specifier before ‘BOOST_ASIO_ELLIPSIS’
  250 |     typename P1, typename BOOST_ASIO_ELLIPSIS PN>
      |                           ^~~~~~~~~~~~~~~~~~~
/usr/local/include/boost/asio/require.hpp:250:47: error: expected ‘>’ before ‘PN’
[... cut ...]
      |                                          P1
/usr/local/include/boost/asio/require.hpp:261:10: error: template argument 3 is invalid
  261 |   >::type>
      |          ^
/usr/local/include/boost/asio/require.hpp:312:3: error: ‘BOOST_ASIO_NODISCARD’ does not name a type; did you mean ‘BOOST_ASIO_MOVE_ARG’?
  312 |   BOOST_ASIO_NODISCARD BOOST_ASIO_CONSTEXPR typename enable_if<
      |   ^~~~~~~~~~~~~~~~~~~~
'''
i just pasted some of the errors but they are most related to the "boost::asio::" .

reymor, 2023-07-03:
I would recommend to test with boost 1.74

closed 2023-07-07 by goncaloalmeida