COVESA/vsomeip #489: failed to link import function (env, _ZN10vsomeip_v37runtime3getEv)

Anilkumar1818, 2023-07-04:
Cmakelists.txt

cmake_minimum_required(VERSION 3.0)
project(wasm-app)
#find_package(vsomeip3)
set (WAMR_ROOT_DIR ${CMAKE_CURRENT_SOURCE_DIR}/../../..)

set (WAMR_BUILD_PLATFORM "linux")

# Reset default linker flags
# set (CMAKE_SHARED_LIBRARY_LINK_C_FLAGS "")
# set (CMAKE_SHARED_LIBRARY_LINK_CXX_FLAGS "")

# set (CMAKE_CXX_STANDARD 14)

if (APPLE)
    set (HAVE_FLAG_SEARCH_PATHS_FIRST 0)
    set (CMAKE_C_LINK_FLAGS "")
    set (CMAKE_CXX_LINK_FLAGS "")
endif ()
  include_directories(/home/anil/Downloads/vsomeip-3.3.0/build/out/include)
#   link_directories(/home/anil/Downloads/vsomeip-3.3.0/build/out/lib)
set (CMAKE_SYSTEM_PROCESSOR wasm32)

if (NOT DEFINED WASI_SDK_DIR)
    set (WASI_SDK_DIR               "/opt/wasi-sdk")
endif ()

#set (CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -Wall -Wextra -Wformat -Wformat-security -Wno-unused")

set (CMAKE_CXX_COMPILER_TARGET        "wasm32-wasi")
#set (CMAKE_CXX_COMPILER               "${WASI_SDK_DIR}/bin/clang++")
set (CMAKE_CXX_COMPILER "/opt/wasi-sdk/bin/clang++")

set (CMAKE_CXX_FLAGS                  "${CMAKE_CXX_FLAGS} -O3 -Wl,--allow-undefined")

add_executable(anil main_v.cpp)
target_link_libraries(anil)

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
main_v.cpp

#include <iomanip>
#include <iostream>

#include <vsomeip/vsomeip.hpp>
#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is " << (_is_available ? "available." : "NOT available.")  << std::endl;
    std::cout<<"hello";
}

int main() {
    std::cout << "vsomeip appp \n";
    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}

I am using the provided CMakeLists.txt and .cpp file to run the vsomeip program in the WAMR runtime. During the compilation process, the files are being compiled successfully. However, I am encountering runtime errors. The specific errors are as follows:

[14:30:49:118 - 7FA1BC4E5CC0]: warning: failed to link import function (env, _ZN10vsomeip_v37runtime3getEv)

goncaloalmeida, 2023-10-23:
@Anilkumar1818 you still have this problem?
Please use the last release and delete the build folder.
Thanks

Anilkumar1818, 2023-10-23:
Thank you for your reply,
Let me try once!!!

closed 2023-11-10 by goncaloalmeida