COVESA/vsomeip #494: Client is taking 10 sec interval to execute the other functionality once Server went down permanently

manuunam8888, 2023-07-20:
We implemented 2 application based on hello world server and client which is provided in examples.
App1 will be acting as server in one IP and similarly App2 will be the client which will be subscribed to APP1 from different IP
We are bringing down App1(server) permanently as part of our testing 
Now App2 which is already Registered to App1 is going for retry of ~10sec and then coming to stopping the application APP2

We modified "**VSOMEIP_SD_DEFAULT_CYCLIC_OFFER_DELAY**" and "**cyclic_offer_delay_**" to some less time 10ms still it is taking 10sec
We tried modified macros in "**internal.hpp.in**" to reducing it to 10ms  and same **result**

**Please suggest any solutions how we can reduce this retry of ~10 sec or how we can avoid this waiting time ?

Any solution how we can shutdown App2 immediately instead of waiting for 10 sec.

Any other solution where we can run it in different thread so that other functions wont wait until this service discovery comes down?**

manuunam8888, 2023-07-24:
@reymor/ @DiogoPedrozza  Any idea about this issue?

reymor, 2023-07-25:
Hello @manuunam8888 

I do not understand what your issue with **cycle_offer_delay** is. Also, I am not familiar with this attribute. However, as far as I know this is related with sending OfferService in main phase. 

If I understand what you want to do, you could do it with `availability_handler_t` in which when you shutdown your "Server" this will trigger a change in the state in you availability_handler and you could shut down your "client/app".

manuunam8888, 2023-08-01:
Thanks @reymor , Your input helped us to resolve the issue for our application.

closed 2023-08-01 by manuunam8888