COVESA/vsomeip #499: [error] Subscriber's IP isn't in the same subnet as host's IP

MoonGyu-Jeong, 2023-08-08:
Hi, I want to communicate windows and ubuntu via attribute. 
I set 
`route add 224.244.224.245 mask 255.255.255.255 10.159.214.39 -p` in windows and 
`sudo route add -nv 224.244.224.245` in Ubuntu.

But I got below errors. How to solve this problem? I attach my logs and json configuration.

**[Windows Service]**
```
[error] Subscriber's IP isn't in the same subnet as host's IP: 10.157.145.161
[error] vsomeip_v3::sd::service_discovery_impl::process_eventgroupentry: Invalid port or IP address in first IPv4 endpoint option specified! 10.157.145.161 session: 0010
```

**[Ubuntu Client]**
```
[info] vsomeip_v3::event::notify_one_unlocked: Notifying a001 to client 1313 failed. Event payload not set!
```

**[Windows Servcie json]**
```json
{
    "unicast" : "10.159.214.39",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "windows-simulator-service",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30490"
        }
    ],
    "routing" : "windows-simulator-service",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp"
    }
}
```

**[Ubuntu Client json]**
```json
{
    "unicast" : "10.157.145.161",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "linux-simulator-client",
            "id" : "0x1313"
        } 
    ],
    "routing" : "linux-simulator-client",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp"
    }
}
```

**[fdepl]**
```fdepl
SomeIpServiceID = 5763
	attribute LocationUTCTime{
		SomeIpGetterID = 8193
		
		SomeIpNotifierID = 40961
		SomeIpAttributeReliable = false
		SomeIpNotifierEventGroups = { 4097 }
			
			
	}
	attribute TimeSource{
		SomeIpGetterID = 8194
		
		SomeIpNotifierID = 40962
		SomeIpAttributeReliable = false
		SomeIpNotifierEventGroups = { 4097 }
			
			
	}
```

reymor, 2023-08-08:
Hello again @MoonGyu-Jeong

Your devices shall be in the same subnet. For instance:

Device A IP: 10.159.214.39

Assuming 255.255.255.0 subnet mask

Device B IP shall be 10.159.214.X

You could change the mask in your json for matching a different subnet. The attribute is `netmask`

MoonGyu-Jeong, 2023-08-08:
Thank you again @reymor :)
Thanks to you, I solve this problem easily.
Hear is my json config file that I added `netmask` attribute.

Because I cannot change IP address, I change the mask.
[Windows IP address] 10.159.214.39 + 255.0.0.0 -> 10.X.X.X
[Ubuntu IP address] 10.157.145.161 + 255.0.0.0 -> 10.X.X.X

```json
{
    "unicast" : "10.159.214.39",
    "netmask" : "255.0.0.0"
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "windows-simulator-service",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30490"
        }
    ],
    "routing" : "windows-simulator-service",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.244.224.245",
        "port" : "30490",
        "protocol" : "udp"
    }
}
```

closed 2023-08-08 by MoonGyu-Jeong