COVESA/vsomeip #500: Communication between two devices without service discovery

govarthananram96, 2023-08-08:
I'm trying to establish communication between two devices (device1 10.10.1.91/32 and device2 10.10.1.92/32). 

i've added static routing using below command
ip route add 10.10.1.92 via 10.10.1.92

Application are running in both devices, but device2 is not able find the service provided by device2. can someone support me on how to establish communication between two device with static routing and no service discovery ?

doschropa, 2023-08-09:
As far as I know, you have to enter the Service ID and Instance ID along with the IP address and port in the configuration JSON of the other device.

But even then I don't think events will work.

duartenfonseca, 2025-06-18:
@govarthananram96 what version did this happen? can you reproduce it still with the most recent version?