COVESA/vsomeip #501: Documentation: service-example.cpp with outdated parameterization

nachtfuchs, 2023-08-08:
https://github.com/COVESA/vsomeip/wiki/vsomeip-in-10-minutes#request--response

the service-example.cpp has an outdated command "app->send(its_response, true);". Solution is:  "app->send(its_response);"

goncaloalmeida, 2023-09-01:
@nachtfuchs please create a PR with the change if you want :)

nachtfuchs, 2023-09-02:
Hi Goncalo,

I tried to, but could not figure out how to make the pull request for the
wiki. I researched and thought that pull-requests to wikis are not
possible. Which made sense in my understanding of the wiki.
If I am wrong, please let me know. Then I will try to find a way to
celebrate the pull-request.

Best

nachtfuchs

goncaloalmeida, 2023-09-07:
@nachtfuchs right, we will do it. thanks!

goncaloalmeida, 2023-09-08:
Edition done.

closed 2023-09-08 by goncaloalmeida