COVESA/vsomeip #502: how open documentation/vsomeip.eap

GitHubAmos, 2023-08-10:
1. What's content vsomeip.eap?
2. How use it's?

reymor, 2023-08-10:
vsomeip.eap is Enterprise Architect project. So, in other words, it is the architecture of the library. You shall use Enerprise Architect for openning it. 

The view is something like this:

[image]

[image]

The only thing that I know about this file is that it is outdated. So, it is based on what is called vsomeip2 instead of the current implementation that is called vsomeip3

GitHubAmos, 2023-08-10:
thanks for your answer

closed 2023-08-21 by goncaloalmeida