COVESA/vsomeip #504: Building all targets on Windows

LenaJok, 2023-08-14:
Hi, I am struggling to build all targets using mingw32-make. I am using boost 1.74 with the latest version of vsomeip 3.3.8.

C:\Users\<username>\Documents\GitHub\vsomeip\build>mingw32-make
[  0%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.obj
In file included from C:\Users\<username>\Documents\GitHub\vsomeip\implementation\endpoints\src\client_endpoint_impl.cpp:18:
C:/Users/<username>/Documents/GitHub/vsomeip/interface/vsomeip/internal/logger.hpp:21:49: warning: 'dllimport' attribute ignored [-Wattributes]
   21 | enum class VSOMEIP_IMPORT_EXPORT level_e : std::uint8_t {
      |                                                 ^~~~~~~
In file included from C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/client_endpoint_impl.hpp:23,
                 from C:\Users\<username>\Documents\GitHub\vsomeip\implementation\endpoints\src\client_endpoint_impl.cpp:20:
C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/buffer.hpp: In member function 'void vsomeip_v3::train::reset()':
C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/buffer.hpp:26:37: error: no match for 'operator=' (operand types are 'std::chrono::nanoseconds' {aka 'std::chrono::duration<long long int, std::ratio<1, 1000000000> >'} and 'int')
   26 |     #define DEFAULT_NANOSECONDS_MAX 1000000000
      |                                     ^~~~~~~~~~
C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/buffer.hpp:56:34: note: in expansion of macro 'DEFAULT_NANOSECONDS_MAX'
   56 |         minimal_debounce_time_ = DEFAULT_NANOSECONDS_MAX;
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
In file included from C:/msys64/mingw64/include/c++/13.1.0/chrono:41,
                 from C:\Users\<username>\Documents\GitHub\vsomeip\implementation\endpoints\src\client_endpoint_impl.cpp:6:
C:/msys64/mingw64/include/c++/13.1.0/bits/chrono.h:596:19: note: candidate: 'constexpr std::chrono::duration<_Rep, _Period>& std::chrono::duration<_Rep, _Period>::operator=(const std::chrono::duration<_Rep, _Period>&) [with _Rep = long long int; _Period = std::ratio<1, 1000000000>]'
  596 |         duration& operator=(const duration&) = default;
      |                   ^~~~~~~~
C:/msys64/mingw64/include/c++/13.1.0/bits/chrono.h:596:29: note:   no known conversion for argument 1 from 'int' to 'const std::chrono::duration<long long int, std::ratio<1, 1000000000> >&'
  596 |         duration& operator=(const duration&) = default;
      |                             ^~~~~~~~~~~~~~~
C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/buffer.hpp:26:37: error: no match for 'operator=' (operand types are 'std::chrono::nanoseconds' {aka 'std::chrono::duration<long long int, std::ratio<1, 1000000000> >'} and 'int')
   26 |     #define DEFAULT_NANOSECONDS_MAX 1000000000
      |                                     ^~~~~~~~~~
C:/Users/<username>/Documents/GitHub/vsomeip/implementation/endpoints/include/buffer.hpp:57:39: note: in expansion of macro 'DEFAULT_NANOSECONDS_MAX'
   57 |         minimal_max_retention_time_ = DEFAULT_NANOSECONDS_MAX;
      |                                       ^~~~~~~~~~~~~~~~~~~~~~~
C:/msys64/mingw64/include/c++/13.1.0/bits/chrono.h:596:19: note: candidate: 'constexpr std::chrono::duration<_Rep, _Period>& std::chrono::duration<_Rep, _Period>::operator=(const std::chrono::duration<_Rep, _Period>&) [with _Rep = long long int; _Period = std::ratio<1, 1000000000>]'
  596 |         duration& operator=(const duration&) = default;
      |                   ^~~~~~~~
C:/msys64/mingw64/include/c++/13.1.0/bits/chrono.h:596:29: note:   no known conversion for argument 1 from 'int' to 'const std::chrono::duration<long long int, std::ratio<1, 1000000000> >&'
  596 |         duration& operator=(const duration&) = default;
      |                             ^~~~~~~~~~~~~~~
mingw32-make[2]: *** [CMakeFiles\vsomeip3.dir\build.make:76: CMakeFiles/vsomeip3.dir/implementation/endpoints/src/client_endpoint_impl.cpp.obj] Error 1
mingw32-make[1]: *** [CMakeFiles\Makefile2:174: CMakeFiles/vsomeip3.dir/all] Error 2
mingw32-make: *** [Makefile:135: all] Error 2

reymor, 2023-08-15:
Hello @LenaJok 

Could try to check this issue https://github.com/COVESA/vsomeip/issues/459 ?

LenaJok, 2023-08-15:
> Hello @LenaJok
> 
> Could try to check this issue #459 ?

Thanks, it worked! I somehow missed this issue.

closed 2023-08-15 by LenaJok