COVESA/vsomeip #505: Need help running hello_world example on Windows -> successful build

LenaJok, 2023-08-16:
Hi,

After successful build of all examples on Windows, when I try running hello world examples I get the following error when running service and client executables (error is the same on both). The default configuration is not changed except the unicast address. Can somebody help me with this?

Thank you!

set VSOMEIP_CONFIGURATION=C:\Users\<username>\Documents\GitHub\vsomeip\out\build\x64-Debug\examples\helloworld-local.json && set VSOMEIP_APPLICATION_NAME=hello_world_client && hello_world_client.exe
2023-08-16 14:36:28.812299 [info] Using configuration file: "C:\Users\<username>\Documents\GitHub\vsomeip\out\build\x64-Debug\examples\helloworld-local.json ".
2023-08-16 14:36:28.812423 [info] Parsed vsomeip configuration in 1ms
2023-08-16 14:36:28.812526 [info] Configuration module loaded.
2023-08-16 14:36:28.812602 [info] Initializing vsomeip (3.3.8) application "hello_world_client ".
2023-08-16 14:36:28.812677 [info] Instantiating routing manager [Proxy].
2023-08-16 14:36:28.813442 [error] **vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]**
2023-08-16 14:36:28.813492 [info] Application(hello_world_client , ffff) is initialized (11, 100).
2023-08-16 14:36:28.813591 [info] Starting vsomeip application "hello_world_client " (ffff) using 2 threads
2023-08-16 14:36:28.813703 [error] vsomeip_v3::endpoint_manager_base::create_local_unlocked: Cannot get guest address of client [0000]
2023-08-16 14:36:28.813820 [info] io thread id from application: ffff (hello_world_client ) is: 86dc
2023-08-16 14:36:28.813858 [info] shutdown thread id from application: ffff (hello_world_client ) is: 22bc
2023-08-16 14:36:28.813867 [info] main dispatch thread id from application: ffff (hello_world_client ) is: 6af8
2023-08-16 14:36:28.813875 [info] io thread id from application: ffff (hello_world_client ) is: 80e0

LenaJok, 2023-08-16:
I can see that others have similar issues: https://github.com/COVESA/vsomeip/issues/432

fcmonteiro, 2023-08-17:
Hi @LenaJok ,
Our latest build is failing on Windows when using recent Boost versions. We are currently looking at this build issue.
Can you try to build vSomeIP and the examples using boost 1.65, maybe the issue is related.
Thanks

fcmonteiro, 2023-11-14:
Closing as double of #432 
Issue is being investigated internally.

closed 2023-11-14 by fcmonteiro