COVESA/vsomeip #506: vsomeip for IPv6, reliable TCP connection of event, stub said "Didn't find connection: fe80::e918:4044:18f0:fcdd:44841"

biggoldenbee, 2023-08-18:
Current on a project, we have to use IPv6. And sets unicast and multicast as below:
device one
{
	"unicast":	"fe80::a00:27ff:feec:4584%enp0s3",
...
	"service-discovery":	{
		"enable":	"true",
		"multicast":	"ff0e::1234%enp0s3",
}
device two
{
	"unicast":	"fe80::e918:4044:18f0:fcdd%eth0",
...
	"service-discovery":	{
		"enable":	"true",
		"multicast":	"ff0e::1234%eth0",
}
After that, runs stub on device one and runs proxy on device two, and when proxy tries to subscribe reliable event then the stub reports:
2023-07-18 11:16:38.041450 [info] Didn't find TCP connection: Subscription rejected for: fe80::e918:4044:18f0:fcdd:37867
...
According to above log check the function function "bool tcp_server_endpoint_impl::is_established(const std::shared_ptr<endpoint_definition>& _endpoint)" of "implementation/endpoints/src/tcp_server_endpoint_impl.cpp" , then found in connections_ the endpoint is "[fe80::e918:4044:18f0:fcdd%enp0s3]:37867" but the "endpoint(_endpoint->get_address(), _endpoint->get_port())" is "fe80::e918:4044:18f0:fcdd:37867", so finally the function "is_established" reports to reject connection.

Hence I tried to modify the source code of is_established as below:
bool tcp_server_endpoint_impl::is_established(const std::shared_ptr<endpoint_definition>& _endpoint) {
    bool is_connected = false;
    endpoint_type endpoint(_endpoint->get_address(), _endpoint->get_port());
    {
        std::lock_guard<std::mutex> its_lock(connections_mutex_);
        auto connection_iterator = connections_.find(endpoint);

        if (connection_iterator != connections_.end()) {
            is_connected = true;
        } else {
            VSOMEIP_INFO << "Didn't find TCP connection: Subscription "
                    << "rejected for: " << endpoint.address().to_string() << ":"
                    << std::dec << static_cast<std::uint16_t>(endpoint.port());
            for (auto i = connections_.begin(); i!= connections_.end(); i++) {
                if ( (std::string::npos!=i->first.address().to_string().find(endpoint.address().to_string()))
                     && (i->first.port()==endpoint.port()) ) {
                    is_connected = true;
                    VSOMEIP_INFO << "Tries to find out: ipv6 subscription "
                        << "accept for: " << i->first;
                    break;
                }
            }
        }
    }
    return is_connected;
}

And tried to modify send_queued as below:
void tcp_server_endpoint_impl::send_queued(const queue_iterator_type _queue_iterator) {
    connection::ptr its_connection;
    {
        std::lock_guard<std::mutex> its_lock(connections_mutex_);
        auto connection_iterator = connections_.find(_queue_iterator->first);
        if (connection_iterator != connections_.end()) {
            its_connection = connection_iterator->second;
        } else {
            VSOMEIP_INFO << "Didn't find connection: "
                    << _queue_iterator->first.address().to_string() << ":" << std::dec
                    << static_cast<std::uint16_t>(_queue_iterator->first.port())
                    << " dropping outstanding messages (" << std::dec
                    << _queue_iterator->second.second.size() << ").";
            bool isfound=false;
            for (auto i = connections_.begin(); i!= connections_.end(); i++) {
                if ( (std::string::npos!=i->first.address().to_string().find(_queue_iterator->first.address().to_string()))
                     && (i->first.port()==_queue_iterator->first.port()) ) {
                    isfound = true;
                    its_connection = i->second;
                    VSOMEIP_INFO << "Tries to find out: ipv6 subscription "
                        << "accept for: " << i->first;
                    break;
                }
            }
            if (false==isfound) queues_.erase(_queue_iterator->first);
        }
    }
    if (its_connection) {
        its_connection->send_queued(_queue_iterator);
    }
}

Of course, proxy subscribes reliable event successfully and gets the notification from stub too, pls see the snpshot of wireshark of about test result.
I tested on ver 3.1.20.1/3.1.20.2/3.1.20.3 and 3.3.0/3.3.6/3.3.8, these versions have the same issue, and now I haven't solved attribute value changed issue on IPv6, please see the log as below:
2023-07-18 11:33:35.759600 [info] notify_one_unlocked: Notifying 83e8 to client 100 failed. Event payload not set!
Checking availability!
2023-07-18 11:33:35.760379 [info] SOME/IP routing ready.
2023-07-18 11:33:45.767655 [info] vSomeIP 3.1.20.1 | (default)
2023-07-18 11:33:55.781058 [info] vSomeIP 3.1.20.1 | (default)
2023-07-18 11:34:02.918002 [info] Reboot detected: IP=fe80::a00:27ff:feec:4584%eth0
2023-07-18 11:34:02.921803 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 212992
Available...
v2xproxy is running under receive mode
callStatus is 0
callStatus is 0
2023-07-18 11:34:05.277472 [info] vSomeIP 3.1.20.1 | (default)
2023-07-18 11:34:15.537365 [info] vSomeIP 3.1.20.1 | (default)

I don't know how to solve "Event payload not set!" of subscribing attribute value changed event of IPv6 now.

fcmonteiro, 2023-08-18:
Hi @biggoldenbee,
We have dropped support to vSomeIP 3.1.x.
Can you retest with the latest 3.3.8?
Thanks

biggoldenbee, 2023-08-19:
Yes, I took one day retest on 3.3.8, but I found another issue: the proxy couldn't get SD message. So I traced "../implementation/endpoints/src/udp_server_endpoint_impl.cpp" and "../implementation/routing/src/routing_manager_impl.cpp", finally in the function "void udp_server_endpoint_impl::on_multicast_received(...)" I got the reason-that _destination is "0.0.0.0". Actually on my test ubuntu the _destination should be "fe80::a00:27ff:fe59:d2ba" a valid ipv6 address. This time I don't know how to modify source code of 3.3.8. And detailed log info as below:
...
2023-08-19 17:13:23.001131 [info] Route "ff0e::1234/128 if: lo gw: n/a" state changed: up
2023-08-19 17:13:23.001334 [error] service_discovery_impl::start() { 
2023-08-19 17:13:23.001494 [error] 0
2023-08-19 17:13:23.001996 [info] udp_server_endpoint_impl: SO_RCVBUFFORCE successful.
2023-08-19 17:13:23.002177 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-08-19 17:13:23.002344 [error] void udp_server_endpoint_impl::receive_unicast() {
2023-08-19 17:13:23.002531 [debug] Joining to multicast group ff0e::1234%enp0s3 from fe80::a00:27ff:fec7:4630%enp0s3
2023-08-19 17:13:23.003142 [error] service_discovery_impl::start() { 2
2023-08-19 17:13:23.003362 [info] SOME/IP routing ready.
2023-08-19 17:13:23.003306 [error] udp_server_endpoint_impl::set_multicast_option
2023-08-19 17:13:23.003762 [error] udp_server_endpoint_impl::set_multicast_option _is_join
2023-08-19 17:13:23.003794 [error] if (!multicast_socket_) {
2023-08-19 17:13:23.003902 [error] if (!multicast_socket_) { is_v4_ :0
2023-08-19 17:13:23.003959 [error] if (!multicast_local) { is_v4_ :0
2023-08-19 17:13:23.004010 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-08-19 17:13:23.004074 [error] void udp_server_endpoint_impl::receive_multicast(uint8_t _multicast_id) { 
2023-08-19 17:13:23.004101 [error] void udp_server_endpoint_impl::receive_multicast(uint8_t _multicast_id) { 0
2023-08-19 17:13:23.004141 [error] is_v6ff0e::1234 fe80::a00:27ff:fec7:4630%enp0s3 2
2023-08-19 17:13:24.766533 [error] void udp_server_endpoint_impl::on_multicast_received(92 
2023-08-19 17:13:24.767187 [error] _destination0.0.0.0 ff0e::1234
2023-08-19 17:13:24.767253 [error] multicast_remote_::local_fe80::a00:27ff:fec7:4630%enp0s3
2023-08-19 17:13:24.767704 [error] on_multicast_received 0
2023-08-19 17:13:24.767754 [error] Ignored SD message from unknown address.::
...

On stub side, the log info as below:
[CAPI][INFO] Loading configuration file '/home/someip/usr/local.x86/bin/commonapi.ini'
[CAPI][INFO] Using default binding 'someip'
[CAPI][INFO] Using default shared library folder '/home/someip/usr/local.x86/lib'
[CAPI][INFO] Registering function for creating "commonapi.v2x.service:v1_0" stub adapter.
[CAPI][INFO] Registering stub for "local:commonapi.v2x.service:v1_0:commonapi.v2x.service"
2023-08-19 17:34:53.650981 [info] Using configuration file: "/home/someip/workspace/3rd.x86/test/v2x/vsomeip-service.ipv6.json".
2023-08-19 17:34:53.651288 [info] Parsed vsomeip configuration in 0ms
2023-08-19 17:34:53.651336 [info] Configuration module loaded.
2023-08-19 17:34:53.651562 [info] Initializing vsomeip (3.3.8) application "service-sample".
2023-08-19 17:34:53.651657 [info] Instantiating routing manager [Host].
2023-08-19 17:34:53.652044 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2023-08-19 17:34:53.652448 [info] Service Discovery enabled. Trying to load module.
2023-08-19 17:34:53.653466 [info] Service Discovery module loaded.
2023-08-19 17:34:53.653898 [info] Application(service-sample, 4000) is initialized (11, 100).
2023-08-19 17:34:53.654810 [info] offer_event: Event [1200.2200.83e8] uses configured cycle time 0ms
2023-08-19 17:34:53.654979 [info] REGISTER EVENT(4000): [1200.2200.83e8:is_provider=true]
2023-08-19 17:34:53.655126 [info] offer_event: Event [1200.2200.83e9] uses configured cycle time 0ms
2023-08-19 17:34:53.655059 [info] Starting vsomeip application "service-sample" (4000) using 2 threads I/O nice 255
2023-08-19 17:34:53.655176 [info] REGISTER EVENT(4000): [1200.2200.83e9:is_provider=true]
2023-08-19 17:34:53.655654 [info] OFFER(4000): [1200.2200:1.0] (true)
2023-08-19 17:34:53.655603 [info] Client [4000] routes unicast:fe80::a00:27ff:fe59:d2ba%enp0s3, prefix:24
2023-08-19 17:34:53.656409 [info] create_local_server: Listening @ /tmp/vsomeip-4000
Successfully Registered Service!
Waiting for calls... (Abort with CTRL+C)
2023-08-19 17:34:53.656986 [info] Watchdog is disabled!
2023-08-19 17:34:53.656757 [info] main dispatch thread id from application: 4000 (service-sample) is: 7f63fd44f700 TID: 5930
2023-08-19 17:34:53.659266 [info] io thread id from application: 4000 (service-sample) is: 7f63fdc50700 TID: 5929
2023-08-19 17:34:53.659620 [info] vSomeIP 3.3.8 | (default)
2023-08-19 17:34:53.659767 [info] Network interface "lo" state changed: up
2023-08-19 17:34:53.660139 [info] Route "ff0e::1234/128 if: lo gw: n/a" state changed: up
2023-08-19 17:34:53.660319 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-08-19 17:34:53.656664 [info] shutdown thread id from application: 4000 (service-sample) is: 7f63fcc4e700 TID: 5931
2023-08-19 17:34:53.659043 [info] io thread id from application: 4000 (service-sample) is: 7f63ef7fe700 TID: 5933
2023-08-19 17:34:53.660406 [debug] Joining to multicast group ff0e::1234%enp0s3 from fe80::a00:27ff:fe59:d2ba%enp0s3
[... cut ...]
Waiting for calls... (Abort with CTRL+C)
Waiting for calls... (Abort with CTRL+C)
Waiting for calls... (Abort with CTRL+C)
2023-08-19 17:35:03.135603 [info] vSomeIP 3.3.8 | (default)
Waiting for calls... (Abort with CTRL+C)
Waiting for calls... (Abort with CTRL+C)
^C2023-08-19 17:35:05.984159 [info] Stopping vsomeip application "service-sample" (4000).
Waiting for calls... (Abort with CTRL+C)

And on wireshark can see that stub broadcasts SD message, pls see snapshot.
[image]

I think this is fatal issue.

biggoldenbee, 2023-08-19:
void udp_server_endpoint_impl::on_multicast_received(
        boost::system::error_code const &_error,
        std::size_t _bytes,
        uint8_t _multicast_id,
        const boost::asio::ip::address &_destination) {

VSOMEIP_ERROR << "multicast_remote_:" << multicast_remote_ ;
VSOMEIP_ERROR << "_destination:" << _destination ;
VSOMEIP_ERROR << "local_:" << local_ ;

2023-08-19 19:03:10.329068 [error] multicast_remote_:[::]:30490
2023-08-19 19:03:10.329955 [error] _destination:0.0.0.0
2023-08-19 19:03:10.330018 [error] local_:[fe80::a00:27ff:fec7:4630%enp0s3]:30490
2023-08-19 19:03:12.513919 [error] multicast_remote_:[::]:30490
2023-08-19 19:03:12.515030 [error] _destination:0.0.0.0
2023-08-19 19:03:12.515085 [error] local_:[fe80::a00:27ff:fec7:4630%enp0s3]:30490

biggoldenbee, 2023-08-20:
Retested on V3.3.8, in connections the endpoint is still "[fe80::a00:27ff:fec7:4630%enp0s3]:42799" so it can't find "fe80::a00:27ff:fec7:4630:42799" out. Traced log info as below:

2023-08-20 07:37:29.731453 [info] Reboot detected: IP=fe80::a00:27ff:fec7:4630%enp0s3
2023-08-20 07:37:29.733638 [info] Didn't find TCP connection: Subscription rejected for: fe80::a00:27ff:fec7:4630:42799
2023-08-20 07:37:29.733921 [error] [fe80::a00:27ff:fec7:4630%enp0s3]:42799
2023-08-20 07:37:29.734606 [error] TCP connection to target1: [fe80::a00:27ff:fec7:4630:42799] not established for subscription to: [1200.2200.4555] 

SO I have to patch above code then it runs very well.

goncaloalmeida, 2023-09-07:
@biggoldenbee thanks for the information.
We will try to reproduce it locally, and check your patch.

biggoldenbee, 2023-10-07:
> @biggoldenbee thanks for the information. We will try to reproduce it locally, and check your patch.

[HelloWorld.tar.gz](https://github.com/COVESA/vsomeip/files/12836052/HelloWorld.tar.gz)
[patch.zip](https://github.com/COVESA/vsomeip/files/12836054/patch.zip)
I attached my HelloWorld sample and the files what I did a patch for temp solution for ipv6, please have a look. And I used boost 1.65. And when you tar xvf HelloWorld.tar.gz, under HelloWorld folder there is vsomeip-service.ipv6.json file, this is my ipv6 config file of Helloworld, you may reference it. And I added "ff0e::1234" to be as multicast.

nachtfuchs, 2023-10-10:
Please note that your **destination:0.0.0.0** is because of a bug in vsomeip 3.3.8. There is a pull-request #462 that solves the service discovery issue for IPv6.

goncaloalmeida, 2023-10-23:
PR https://github.com/COVESA/vsomeip/pull/462 merged :)
Closing this ticket, if it occurs again please reopen the ticket.

cc @fcmonteiro @DiogoPedrozza

closed 2023-10-23 by goncaloalmeida

maramharsha99, 2023-10-31:
can i build communication between client and service without using common
API i just want to use vsomeip
but i want to build communication between different machines client on one
machine and service in another machine