COVESA/vsomeip #507: use some-tp the memory increase very fast

ForrestGQZ, 2023-08-18:
There is one service and contain 20 events,every event send 28KB data,the frequency is 10HZ.
As the fllowing log the "queue" and VmRSS increase very fast.

log:
2022-09-27 03:28:03.100187 [info] memory usage: VmSize 324204 kB, VmRSS 31076 kB, shared pages 5328 kB, text 16 kB, data 67468 kB | monotonic time: 3508.518230571 cpu time: 1.851213573
2022-09-27 03:28:03.104076 [info] status local client endpoints: 1
2022-09-27 03:28:03.104861 [info] status lce: /tmp/vsomeip-15b1 queue: 0 data: 0
2022-09-27 03:28:03.105662 [info] status start remote client endpoints:
2022-09-27 03:28:03.106507 [info] status uce: 192.168.9.23:55132 -> 192.168.9.10:63126 queue: 0 data: 0
2022-09-27 03:28:03.107279 [info] status end remote client endpoints: 1
2022-09-27 03:28:03.107963 [info] status start server endpoints:
2022-09-27 03:28:03.108692 [info] status lse: /tmp/vsomeip-0 connections: 1 queues: 0
2022-09-27 03:28:03.109458 [info] status lse: client:  recv_buffer: 28721
2022-09-27 03:28:03.110167 [info] status use: 30490 number queues: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2022-09-27 03:28:03.110901 [info] status use: client: 192.168.9.10:30490 queue: 0 data: 0
2022-09-27 03:28:03.111648 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2022-09-27 03:28:03.112385 [info] status use: 63127 number queues: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2022-09-27 03:28:03.113532 [info] status use: client: 192.168.9.10:38001 queue: 16762 data: 23230092
2022-09-27 03:28:03.114311 [info] status end server endpoints:3
2022-09-27 03:28:08.107012 [info] memory usage: VmSize 335032 kB, VmRSS 52824 kB, shared pages 5328 kB, text 16 kB, data 89084 kB | monotonic time: 3513.518910976 cpu time: 3.461873070
2022-09-27 03:28:08.115263 [info] status local client endpoints: 1
2022-09-27 03:28:08.126154 [info] status lce: /tmp/vsomeip-15b1 queue: 1 data: 1065
2022-09-27 03:28:08.127102 [info] status start remote client endpoints:
2022-09-27 03:28:08.128002 [info] status uce: 192.168.9.23:55132 -> 192.168.9.10:63126 queue: 0 data: 0
2022-09-27 03:28:08.128787 [info] status end remote client endpoints: 1
2022-09-27 03:28:08.129536 [info] status start server endpoints:
2022-09-27 03:28:08.130300 [info] status lse: /tmp/vsomeip-0 connections: 1 queues: 0
2022-09-27 03:28:08.131041 [info] status lse: client:  recv_buffer: 28721
2022-09-27 03:28:08.131741 [info] status use: 30490 number queues: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2022-09-27 03:28:08.132466 [info] status use: client: 192.168.9.10:30490 queue: 0 data: 0
2022-09-27 03:28:08.133215 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2022-09-27 03:28:08.133950 [info] status use: 63127 number queues: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2022-09-27 03:28:08.134664 [info] status use: client: 192.168.9.10:38001 queue: 31784 data: 44049336
2022-09-27 03:28:08.135434 [info] status end server endpoints:3
2022-09-27 03:28:13.080881 [info] vSomeIP 3.1.20.3 | (default)
2022-09-27 03:28:13.110037 [info] memory usage: VmSize 346500 kB, VmRSS 74832 kB, shared pages 5328 kB, text 16 kB, data 110956 kB | monotonic time: 3518.519213519 cpu time: 5.86615070

configuration:
    "service" : "0x292E",
            "instance" : "0x1",
            "unreliable" : "63127",
            "someip-tp" : 
            {
                "service-to-client":
                [
                    "0x8001",
                    "0x8002",
                    "0x8003",
                    "0x8004",
                    "0x8005",
                    "0x8006",
                    "0x8007",
                    "0x8008",
                    "0x8009",
                    "0x800A",
                    "0x800B",
                    "0x800C",
                    "0x800D",
                    "0x800E",
                    "0x800F",
                    "0x8010",
                    "0x8011",
                    "0x8012",
                    "0x8013",
                    "0x8014"
                ]
            },

fcmonteiro, 2023-08-18:
Hi @ForrestGQZ ,
Can you check if this also occurs in the latest vSomeIP 3.3.8?
we have dropped support to 3.1.x
Thanks

ForrestGQZ, 2023-08-19:
Hi @fcmonteiro
The same issue also happen in the latest vSomeIP 3.3.8
But I changed the transmission frequency to 50Hz. Memory does not increase.

ForrestGQZ, 2023-08-21:
I made a mistake, the transmission frequency is 100HZ.And change to 20HZ after running about 7 minutes,memory increase.

ForrestGQZ, 2023-08-23:
Base on v3.3.8, Sender memory increase, receiver work fine. 20 events,  data size is 9000,  transmission frequency is 100HZ.

2023-08-23 20:54:50.681685 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 20:54:50.681774 [info] status use: client: 172.17.0.3:47828 queue: 7 data: 9152
2023-08-23 20:54:50.681882 [info] status end server endpoints:3
2023-08-23 20:54:52.661078 [info] memory usage: VmSize 387916 kB, VmRSS 10156 kB, shared pages 7088 kB, text 16 kB, data 52748 kB | monotonic time: 1247635.446487434 cpu time: 17.289843434
2023-08-23 20:54:52.682536 [info] status local client endpoints: 1
2023-08-23 20:54:52.682785 [info] status lce: /tmp/vsomeip-eee9 queue: 0 data: 0
2023-08-23 20:54:52.682843 [info] status start remote client endpoints:
2023-08-23 20:54:52.682880 [info] status end remote client endpoints: 0
2023-08-23 20:54:52.682973 [info] status start server endpoints:
2023-08-23 20:54:52.683063 [info] status lse: /tmp/vsomeip-0 connections: 1 targets: 0
2023-08-23 20:54:52.683115 [info] status lse: client:  recv_buffer: 9051
2023-08-23 20:54:52.683200 [info] status use: 30490 number targets: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2023-08-23 20:54:52.683258 [info] status use: client: 172.17.0.3:30490 queue: 0 data: 0
2023-08-23 20:54:52.683389 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2023-08-23 20:54:52.683430 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 20:54:52.683462 [info] status use: client: 172.17.0.3:47828 queue: 0 data: 0
2023-08-23 20:54:52.683514 [info] status end server endpoints:3
2023-08-23 20:54:54.663475 [info] memory usage: VmSize 387936 kB, VmRSS 10232 kB, shared pages 7088 kB, text 16 kB, data 52768 kB | monotonic time: 1247637.446726982 cpu time: 17.607189299
2023-08-23 20:54:54.684021 [info] status local client endpoints: 1
2023-08-23 20:54:54.684276 [info] status lce: /tmp/vsomeip-eee9 queue: 0 data: 0
2023-08-23 20:54:54.684336 [info] status start remote client endpoints:
2023-08-23 20:54:54.684415 [info] status end remote client endpoints: 0
2023-08-23 20:54:54.684453 [info] status start server endpoints:
2023-08-23 20:54:54.684511 [info] status lse: /tmp/vsomeip-0 connections: 1 targets: 0
2023-08-23 20:54:54.684594 [info] status lse: client:  recv_buffer: 9051
2023-08-23 20:54:54.684645 [info] status use: 30490 number targets: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2023-08-23 20:54:54.684829 [info] status use: client: 172.17.0.3:30490 queue: 0 data: 0
2023-08-23 20:54:54.684878 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2023-08-23 20:54:54.684910 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 20:54:54.684935 [info] status use: client: 172.17.0.3:47828 queue: 0 data: 0
2023-08-23 20:54:54.684962 [info] status end server endpoints:3
2023-08-23 20:54:56.504795 [info] vSomeIP 3.3.8 | (default)
2023-08-23 20:54:56.665044 [info] memory usage: VmSize 387976 kB, VmRSS 10292 kB, shared pages 7088 kB, text 16 kB, data 52828 kB | monotonic time: 1247639.446882151 cpu time: 17.953578167
2023-08-23 20:54:56.685602 [info] status local client endpoints: 1
2023-08-23 20:54:56.685964 [info] status lce: /tmp/vsomeip-eee9 queue: 0 data: 0
2023-08-23 20:54:56.686179 [info] status start remote client endpoints:
2023-08-23 20:54:56.686274 [info] status end remote client endpoints: 0
2023-08-23 20:54:56.686313 [info] status start server endpoints:
2023-08-23 20:54:56.686397 [info] status lse: /tmp/vsomeip-0 connections: 1 targets: 0
2023-08-23 20:54:56.686509 [info] status lse: client:  recv_buffer: 9051
2023-08-23 20:54:56.686559 [info] status use: 30490 number targets: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2023-08-23 20:54:56.686627 [info] status use: client: 172.17.0.3:30490 queue: 0 data: 0
2023-08-23 20:54:56.686771 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2023-08-23 20:54:56.686819 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 20:54:56.687149 [info] status use: client: 172.17.0.3:47828 queue: 0 data: 0
2023-08-23 20:54:56.687249 [info] status end server endpoints:3
2023-08-23 20:54:58.666604 [info] memory usage: VmSize 388064 kB, VmRSS 10352 kB, shared pages 7088 kB, text 16 kB, data 52916 kB | monotonic time: 1247641.447036479 cpu time: 18.283299839
2023-08-23 20:54:58.687744 [info] status local client endpoints: 1
2023-08-23 20:54:58.687908 [info] status lce: /tmp/vsomeip-eee9 queue: 0 data: 0
2023-08-23 20:54:58.687984 [info] status start remote client endpoints:
2023-08-23 20:54:58.688024 [info] status end remote client endpoints: 0
2023-08-23 20:54:58.688350 [info] status start server endpoints:
2023-08-23 20:54:58.688670 [info] status lse: /tmp/vsomeip-0 connections: 1 targets: 0
2023-08-23 20:54:58.688850 [info] status lse: client:  recv_buffer: 9051
2023-08-23 20:54:58.688968 [info] status use: 30490 number targets: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2023-08-23 20:54:58.689080 [info] status use: client: 172.17.0.3:30490 queue: 0 data: 0
2023-08-23 20:54:58.689733 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2023-08-23 20:54:58.689972 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 20:54:58.690036 [info] status use: client: 172.17.0.3:47828 queue: 0 data: 0
[... cut ...]
2023-08-23 21:04:06.602347 [info] status use: 30490 number targets: 2 recv_buffer: 1416 multicast_recv_buffer: 1416
2023-08-23 21:04:06.602387 [info] status use: client: 172.17.0.3:30490 queue: 0 data: 0
2023-08-23 21:04:06.602443 [info] status use: client: 239.0.0.102:30490 queue: 0 data: 0
2023-08-23 21:04:06.602473 [info] status use: 63126 number targets: 1 recv_buffer: 1416 multicast_recv_buffer: 0
2023-08-23 21:04:06.602501 [info] status use: client: 172.17.0.3:47828 queue: 7 data: 9152
2023-08-23 21:04:06.602566 [info] status end server endpoints:3

goncaloalmeida, 2023-09-07:
@ForrestGQZ thanks for the information.
We will try to reproduce locally and analyse the problem

fcmonteiro, 2023-12-05:
Issue confirmed internally.

goncaloalmeida, 2024-05-20:
on the next release of vsomeip-lib we will include a fix for this issue.

duartenfonseca, 2024-08-13:
the fix to this problem is going to be integrated here: https://github.com/COVESA/vsomeip/pull/733/commits/940d72af2cb39b99c8e5ea6e820696c8fdca5ce6

closed 2024-08-13 by duartenfonseca