COVESA/vsomeip #517: vsomeip 3.1.20 not subscribe

guoyang1, 2023-09-04:
we use vsomeip 3.1.20, find sometimes when the client reboot and send offer message again, the server not subscribe

goncaloalmeida, 2023-09-04:
We did not support anymore the vsomeip 3.1.20, please use the last version.
https://github.com/COVESA/vsomeip/discussions/420

closed 2023-09-04 by goncaloalmeida

guoyang1, 2023-09-04:
> We did not support anymore the vsomeip 3.1.20, please use the last version. #420
can i require that the 3.1.20 has this bug or not ,because we add some com interface in vsomip to adapt our project,so it has much  worke to do to change the version of vsomeip

guoyang1, 2023-09-04:
print warning log : creat_event_entry :Didn't insert sybscription add subscription does not match reliability type :[a447:0001.001]] 1
cannot create subscription doesn not match reliability type[a447.0001.0001] 1