COVESA/vsomeip #522: Regular vsomeip Meeting

paulboyes, 2023-09-18:
Earlier this year, it was requested and suggested that we hold a regular meeting for vsomeip meeting.    A monthly meeting was scheduled for the 3rd Monday of each month at 7am PT / 1600 CET - see the the [Community Calendar](https://wiki.covesa.global/display/WIK4/COVESA+Common+Meeting+Schedule).  We have had a couple meetings but no maintainers have showed up.  There is definitely significant interest from OEMs.  My guess is the maintainers are not aware of the meeting time.  I'd like to setup a meeting for next Monday at 7am PT /10am ET / 16:00 CET to get things going.  Would tis work for the maintainers and contributors?  @goncaloalmeida @JoaoMarquues @fcmonteiro @DiogoPedrozza @BrunoLSilvaCTW @AndreArbo @anaritarodrigues 

Thank you!

paulboyes, 2023-09-19:
It's my understanding that there is a holiday in Germany on Oct 3 and many take the 2nd off.  Hence I have updated the meeting to this coming Monday, Sept 25.  Can the teams make it?

DiogoPedrozza, 2023-09-20:
Hi @paulboyes. Yes, we will try to join the meeting with some maintainers.

paulboyes, 2023-09-20:
Great.  Thanks Diogo.  Talk with you Monday.

closed 2023-09-21 by fcmonteiro