COVESA/vsomeip #523: Pcap and RoutingManager log file Subscribe transmission results are different

wjdqjqrnjs, 2023-09-20:
The results of the pcap file and RoutingManager Log file are different.

**[Pcap File]**

When checking pcap, the problem is sending **_the same subscribe msg twice in succession_**.

10.0.32.0	10.0.32.100	SOME/IP-SD	98	SOME/IP Service Discovery Protocol [Subscribe]

        --> Session id:1

10.0.32.0	10.0.32.100	SOME/IP-SD	98	SOME/IP Service Discovery Protocol [Subscribe]

        --> Session id:2

**[RoutingManager Log File]**

But, when I checked the test app and routingmanager log files, I confirmed that **_subscribe was sent once_**.
       
[info] REQUEST(0007): [2001.2001:1.4294967295]
[info] configuration_impl:find_specific_port #1: service: 2001 instance: 2001 reliable: 1 return specific port: 31001
[info] endpoint_manager_impl::create_remote_client: 10.0.32.100:30511 reliable: 1 using local port: 31001
[info] configuration_impl:find_specific_port #1: service: 2001 instance: 2001 reliable: 0 return specific port: 31002
[info] endpoint_manager_impl::create_remote_client: 10.0.32.100:30512 reliable: 0 using local port: 31002
[info] udp_client_endpoint_impl::connect: SO_RCVBUFFORCE successful!
[info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 1703936 (1703936) local port:31002 remote:10.0.32.100:30512
**_[info] SUBSCRIBE(0007): [2001.2001.0006:800b:1]_**
[warning] Didn't receive a multicast SD message for 2200ms.
[info] udp_server_endpoint_impl<multicast>: SO_RCVBUFFORCE: successful.
[info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
[info] UNSUBSCRIBE(0007): [2001.2001.0006.800b]
[warning] Didn't receive a multicast SD message for 2200ms.
[info] udp_server_endpoint_impl<multicast>: SO_RCVBUFFORCE: successful.
[info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
[warning] Didn't receive a multicast SD message for 2200ms.

So, need to find out which part is causing the issue.

fcmonteiro, 2023-09-21:
Hi @wjdqjqrnjs,
Can you attach the pcap and log files for us to have a look?
Thanks

wjdqjqrnjs, 2023-09-22:
Hello @fcmonteiro 
Thanks for checking my issue.

I attachced the pcap and routing manager log file

Referring to pcap 43~46 lines, subscribe msg is transmitted twice.

*Note*
The same message is output twice in pcap, so 43 and 44 line sub (session id 1) are same, 45 and 46 line sub (session id 2) are same.

Thanks.
[github Issue files.zip](https://github.com/COVESA/vsomeip/files/12697598/github.Issue.files.zip)

123130, 2023-10-30:
@wjdqjqrnjs Hi. I have run the example(hello_world, subscribe/notify and request/response) correctly, but I did not catch any packets via the command(sudo tcpdump -i lo -w someip.pcap).
Do you know where the problem is?
request:
[image]
response:
[image]
[image]

Rajkumar528cse, 2023-12-20:
Dear @wjdqjqrnjs,

Is the issue has been resolved? if yes, could you please share the solution to me?

Thank you,
Raj.

Rajkumar528cse, 2023-12-20:
Looks like this issue/bug has been introduced in v3.3.8 and there is no such issue in v3.1.20.
@123130 could you please provide your inputs on this issue?

Rajkumar528cse, 2024-01-09:
Dear @123130 and @fcmonteiro,
Could you please confirm on below queries?
Q1. Was this an issue in vsomeip v3.1.20 and it got fixed in v3.3.8 ?
Q2. [OR] This is an issue introduced in v3.3.8 ?

When we tried to find root cause, it is found that there is some issue in the function send_subscription(...) from service_discovery_impl.cpp, please check the added image.
[image]

Thank you.

duartenfonseca, 2025-04-08:
@wjdqjqrnjs @AGN-2022 does this still happen on 3.5.5?

closed 2025-06-18 by duartenfonseca