COVESA/vsomeip #524: Cannot find free client port after link reboot detection (v3.3.8)

wjdqjqrnjs, 2023-09-20:
In Vsomeip 3.3.8 version,

Still after reboot, the issue (Cannot find free client port after link reboot detection) occurs

#428 
I refer to #428 comments.

I confirmed that the issue has been resolved in the latest version.

I would like to inquire if there are any additional settings or contents needed when building in version 3.3.8.

2023-08-28 15:38:40.490094 [info] REGISTERED_ACK(0007)
2023-08-28 15:38:40.491959 [info] REGISTER EVENT(0007): [2001.2001.8001:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.492398 [info] REGISTER EVENT(0007): [2001.2001.8002:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.492644 [info] REGISTER EVENT(0007): [2001.2001.8003:eventtype=0:is_provided=false:reliable=1]
2023-08-28 15:38:40.492875 [info] REGISTER EVENT(0007): [2001.2001.8004:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.493083 [info] REGISTER EVENT(0007): [2001.2001.8005:eventtype=2:is_provided=false:reliable=2]
2023-08-28 15:38:40.493284 [info] REGISTER EVENT(0007): [2001.2001.8006:eventtype=2:is_provided=false:reliable=2]
2023-08-28 15:38:40.493524 [info] REGISTER EVENT(0007): [2001.2001.8007:eventtype=2:is_provided=false:reliable=2]
2023-08-28 15:38:40.493735 [info] REGISTER EVENT(0007): [2001.2001.8008:eventtype=2:is_provided=false:reliable=2]
2023-08-28 15:38:40.493956 [info] REGISTER EVENT(0007): [2001.2001.800b:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.494168 [info] REGISTER EVENT(0007): [2001.2001.8501:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.494391 [info] REGISTER EVENT(0007): [2001.2001.8502:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.494595 [info] REGISTER EVENT(0007): [2001.2001.8503:eventtype=0:is_provided=false:reliable=2]
2023-08-28 15:38:40.494809 [info] REGISTER EVENT(0007): [2001.2001.850a:eventtype=2:is_provided=false:reliable=2]
2023-08-28 15:38:40.577148 [info] REQUEST(0007): [2001.2001:1.4294967295]
**_2023-08-28 15:38:40.879041 [error] Cannot find free client port for communication to service: 8193 instance: 8193 remote_port: 30511 reliable: 1_**
2023-08-28 15:38:40.879453 [info] configuration_impl:find_specific_port #1: service: 2001 instance: 2001 reliable: 0 return specific port: 31002
2023-08-28 15:38:40.880009 [info] endpoint_manager_impl::create_remote_client: 10.0.32.100:30512 reliable: 0 using local port: 31002
2023-08-28 15:38:40.880723 [info] udp_client_endpoint_impl::connect: SO_RCVBUFFORCE successful!
2023-08-28 15:38:40.880843 [info] udp_client_endpoint_impl::connect: SO_RCVBUF is: 1703936 (1703936) local port:31002 remote:10.0.32.100:30512
2023-08-28 15:38:42.878892 [warning] Didn't receive a multicast SD message for 2200ms.
2023-08-28 15:38:42.881881 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUFFORCE: successful.
2023-08-28 15:38:42.882263 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 1703936 (1703936) local port:30490
2023-08-28 15:38:44.423147 [info] vSomeIP 3.3.8 | (default)

Thank you.

fcmonteiro, 2023-09-21:
Hi @wjdqjqrnjs,
We are aware of the issue still happening in v3.3.8
Internally we already have some minor changes, but we are still working on this.

How can we reproduce your issue? We want to check if it's occurs in our latest version and properly fix this.
Thanks

cc: @DiogoPedrozza @goncaloalmeida

wjdqjqrnjs, 2023-09-22:
Hello @fcmonteiro,
Thanks for checking my issue.

I share two scenario.

There are **two reboot-related scenario**.
But, the problem is occurring in the **first scenario**.

**[First Scenario]**
1) DUT Client Service sends multicast SOME/IP-SD message carrying a _**Find Service Entry**_.
2) TESTER verify received SOME/IP-SD message.
3) TESTER sends SOME/IP **_clientServiceSubscribeEventgroup request_** for specific eventgroup.
4) TESTER sends multicast SOME/IP-SD message carrying an **_Offer Service Entry_**.
5) DUT sends unicast SOME/IP-SD message carrying a **_Subscribe Eventgroup Entry_**.
6) TESTER verifies received SOME/IP-SD message.
_**7) TESTER sends unicast **_Subscribe Eventgroup Acknowledgement Entry_**.**_
8) TESTER sends multicast SOME/IP-SD message with **_Reboot Flag set to 1 and Session ID set to 0x0001 carrying an Offer Service Entry_**.
9) DUT sends unicast SOME/IP-SD message carrying a **_Subscribe Eventgroup Entry_**.

**[Second Scenario]**

1) DUT client service sends multicast SOME/IP-SD message carrying a **_Find Service Entry_**.
2) TESTER verifies received SOME/IP-SD message.
3) TESTER sends **_clientServiceSubscribeEventgroupSingle request_** for specific eventgroup.
4) TESTER sends multicast SOME/IP-SD message carrying an **_Offer Service Entry_**.
5) DUT sends unicast SOME/IP-SD **_Subscribe Eventgroup entry_**.
6) TESTER verifes received SOME/IP-SD message.
7) TESTER sends unicast SOME/IP-SD message with **_Reboot Flag set to 1 and Session ID set to 0x0001, carrying an Offer Service Entry_**.

8) DUT resets TCP Connection by sending SYN.
9) TESTER accepts TCP connection by sending SYN + ACK.
10) DUT sends unicast SOME/IP-SD Subscribe Eventgroup entry.
11) TESTER verifies received SOME/IP-SD message.

In the second scenario, a reboot is also in progress, but the issue does not occur after that scenario.

Only after first scenario, the issue occurs.

**[My opinion]**
Looking at 7th process in first scenario, there is a difference in where the subscribe-ack msg is transmitted.

I think the presence or absence of sub-ack transmission before reboot may cause this issue.

Thanks.

cc: @DiogoPedrozza @goncaloalmeida

fcmonteiro, 2023-10-02:
Hi @wjdqjqrnjs 
Thanks for the detailed explanation.
We are looking at this internally.

AlessandroGiusa, 2024-05-22:
Hi, is this issue still being actively worked on?