COVESA/vsomeip #525: Is overlays not supported on V3.3.8?

zhaoxin39913, 2023-09-20:
I didn't find any description about "overlays" in the user manual of V3.3.8, also the macro of ENABLE_CONFIGURATION_OVERLAYS can not be found in source code. Does that mean "overlays" is not supported any more ?

fcmonteiro, 2023-09-21:
Hi @zhaoxin39913,
Yes, the ENABLE_CONFIGURATION_OVERLAYS macro was dropped in vSomeIP 3.3.x
In replacement you can use `VSOMEIP_CONFIGURATION_<application>` environment variable, please check our [UserGuide](https://github.com/COVESA/vsomeip/blob/655bb34279bf9ded679451258b8eaf2c8a1a7ea9/documentation/vsomeipUserGuide#L266) on how to use it.
Thanks.

zhaoxin39913, 2023-09-22:
Thanks for the quick response.

So the dedicated application configuration via "VSOMEIP_CONFIGURATION_<application>" in 3.3.x is enabled by default, right ? And the format is the same as overlays configuration?

fcmonteiro, 2023-10-02:
Yes, as far as I know the usage of VSOMEIP_CONFIGURATION_x should be similar to the overlay.
But if you have problems using it, please let us know.
Thanks

closed 2023-10-02 by fcmonteiro