COVESA/vsomeip #527: Compilation error with gcc12 and boost 1.81.0

pberginkonsult, 2023-09-25:
When building vsomeip version 3.3.8 with Yocto 4.1 (mickledore) there is compilation errors in two files:

```
FAILED: CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o 
<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot-native/usr/bin/aarch64-wrs-linux/aarch64-wrs-linux-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DVSOMEIP_VERSION=\"3.3.8\" -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_RECURSIVE_MUTEX_INIT_FUNC -Dvsomeip3_EXPORTS -I<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/interface -mcpu=cortex-a57 -march=armv8-a+crc -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fmacro-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git=/usr/src/debug/vsomeip/3.3.8-r0  -fdebug-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git=/usr/src/debug/vsomeip/3.3.8-r0  -fmacro-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/build=/usr/src/debug/vsomeip/3.3.8-r0  -fdebug-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/build=/usr/src/debug/vsomeip/3.3.8-r0  -fdebug-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot=  -fmacro-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot=  -fdebug-prefix-map=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden -DVSOMEIP_ENABLE_SIGNAL_HANDLING -DVSOMEIP_BOOST_VERSION=108100 -DWITHOUT_SYSTEMD  -pie -Wl,-z,relro,-z,now -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE -g  -std=c++14  -Wl,-export-dynamic -Wl,--version-script=<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/exportmap.gcc -O2 -g -DNDEBUG -fPIC -MD -MT CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o -MF CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o.d -o CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o -c <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/policy.cpp
In file included from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/concept/interval_set.hpp:14,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/associative_interval_container.hpp:16,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_base_set.hpp:24,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_set.hpp:14,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_map.hpp:13,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/../include/policy.hpp:15,
                 from <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/policy.cpp:10:
In function 'typename Type::iterator boost::icl::segmental::join_under(Type&, const typename Type::value_type&, typename Type::iterator) [with Type = boost::icl::interval_set<unsigned int>]',
    inlined from 'boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::add_over(const interval_type&, iterator) [with DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from 'boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::_add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from 'typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::segment_type&) [with Type = interval_set<unsigned int>]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from 'typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::element_type&) [with Type = interval_set<unsigned int>]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/concept/interval_set.hpp:79:20,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_base_set.hpp:252:24,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::insert(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/interval_base_set.hpp:289:19,
    inlined from 'bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)' at <builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/policy.cpp:71:23:
<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))
      |     ^~~~~~~~~~
<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/policy.cpp: In member function 'bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)':
<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/git/implementation/security/src/policy.cpp:70:37: note: at offset 40 into destination object 'its_gid_set' of size 48
   70 |     boost::icl::interval_set<gid_t> its_gid_set;
      |                                     ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

```
FAILED: CMakeFiles/vsomeip3.dir/implementation/routing/src/routing_manager_client.cpp.o 
<builddir>/tmp-glibc/work/cortexa57-wrs-linux/vsomeip/3.3.8-r0/recipe-sysroot-native/usr/bin/aarch64-wrs-linux/aarch64-wrs-linux-g++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DVSOMEIP_INTERNAL_SUPPRESS_DEPRECATED -DVSOMEIP_VERSION=\"3.3.8\" -D_GTHREAD_USE_MUTEX_INIT_FUNC -D_GTHREAD_USE_ [... cut ...]
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
```

Seems that both of them are related to `boost/icl/interval_set.hpp:14'. I have not been able to analyze if the warning is a real one or a false positive and would request help from the community and someone more familiar with the code. To silence the warning `-Wno-stringop-overflow` can be used but wanted to have verdict if this is something to look in to first.

Compiler: `aarch64-wrs-linux-g++ (GCC) 12.3.0`

VariableVictor, 2023-11-13:
I installed the arch aur package. it works. the 3.3.0 version vsomeip can be compiled and installed.
so I read the PKGBUILD script.
`
build() {
    cd "${pkgname}-${pkgver}"
    mkdir -p build
    cd build
    cmake -E env CXXFLAGS="-Wno-error=stringop-overflow" \
    cmake .. -DCMAKE_INSTALL_PREFIX:PATH=/usr \
             -DCMAKE_BUILD_TYPE=Release
    make
}
`
you can see the error which you can not compile vsomeip is the warning of "stringop-overflow",so you can set -Wno-error option when you compile it.
In the end, I succeed to compile the lastest vsomeip of 3.4.9.1

pberginkonsult, 2023-11-13:
> you can see the error which you can not compile vsomeip is the warning of "stringop-overflow",so you can set -Wno-error option when you compile it. In the end, I succeed to compile the lastest vsomeip of 3.4.9.1

@VariableVictor thanks for your input!

This is my current workaround to compile this version. But I don't treat that as a solution. The error has some meaning and I hope someone with good knowledge around this code base can help out with a proper analysis if this is a false positive that can be ignored or rewrite the code to improve it and remove the error.

pberginkonsult, 2023-11-30:
> Issue fixed in the last release (3.4.10)

@goncaloalmeida can you pleases reference the commit the fixes the issue? I still see same issue with version 3.4.10

goncaloalmeida, 2023-11-30:
@pberginkonsult sorry. I closed this by mistake.
What we fixed was compiling in the versions that we provide support: windows and Linux builds with the last boost version (1.83)

anuragsinah, 2024-02-15:
I have tried it boost 1.84.0 and VSomeIP 3.4.10. 
Here we have to build the boost library with "CXXFLAGS" = "-fPIC" flag. And have to convert the error to the warning using "CXXFLAGS"= "-Wno-error=stringop-overflow" flag while building VSomeIP.
In this way I was able to build the VSomeIP for my freedesktopSDK 22.08 in flatpak.

Adolphsa, 2024-02-18:
@anuragsinah   could you show your boost1.84.0 compile script file?

anuragsinah, 2024-02-22:
I downloaded boost/1.84.0 from source code after this please init and clone all the submodule as well -
` git clone https://github.com/boostorg/boost.git`
`git checkout tags/boost-1.84.0`
`git submodule update --init --recursive`

After this please compress this boost folder (.zip, .tar etc)

Below is my snippet from manifest file to build boost. Please provide the correct location for boost compressed file.

`{
      "name": "boost",
      "buildsystem": "cmake",
      "cleanup": ["/bin"],
      "config-opts": [
        "-Werror=stringop-overflow="
      ],
      "build-options": {
        "env": {
          "CXXFLAGS": "-fPIC"
        }
      },
      "sources": [
        {
          "type": "archive",
          "path": "{{please provide the path to compressed file}}"
        }
      ]
    }`

tobylorenz, 2024-02-26:
If this is a required compiler flag, then let's add it to CMakeLists.txt:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 82ab4b3..6641452 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -73,7 +73,7 @@ endif()
 
     set(NO_DEPRECATED "")
     set(OPTIMIZE "")
-    set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
+    set(OS_CXX_FLAGS "${OS_CXX_FLAGS} -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -Wno-error=stringop-overflow -fexceptions -fstrict-aliasing -fstack-protector-strong -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Wpedantic -Werror -fPIE")
 
     # force all use of std::mutex and std::recursive_mutex to use runtime init
     # instead of static initialization so mutexes can be hooked to enable PI as needed
```

However, it would be better to evaluate and understand the issue. I fear that just adding a compiler flag hides an actual issue.

lutzbichler, 2024-03-13:
Replace both occurrences of

its_gid_set.insert(its_gid)

by

its_gid_set.insert(boost::icl::discrete_interval<uint32_t>(its_gid, its_gid))

As it seems, gcc12/gcc13 issue a lot of these errors which actually are false positives. However, the change is a noop and silences the error (at least for me).

brenkem, 2024-04-18:
I can confirm that this issue (see [1]) is still present with current master  (see [2]) and boost 1.74.0 but also 1.83.0 (combined with gcc (Debian 12.2.0-14) 12.2.0.

I took the workaround above and converted the error to a warning.

[2]:
```
commit cf497232adf84f55947f7a24e1b64e04b49f1f38 (HEAD -> master, origin/master, origin/HEAD)
Author: Matthew Russell <matthew.g.russell@gmail.com>
Date:   Wed Apr 10 06:23:27 2024 -0400
```

[1]:
````
…
[ 65%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/utility/src/wrappers.cpp.o
[ 66%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/utility/src/utility.cpp.o
[ 66%] Building CXX object CMakeFiles/vsomeip3.dir/implementation/utility/src/wrappers_qnx.cpp.o
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /usr/include/boost/icl/interval_map.hpp:13,
                 from /tmp/vsomeip/implementation/security/src/../include/policy.hpp:15,
                 from /tmp/vsomeip/implementation/security/src/policy.cpp:10:
In function 'typename Type::iterator boost::icl::segmental::join_under(Type&, const typename Type::value_type&, typename Type::iterator) [with Type = boost::icl::interval_set<unsigned int>]',
    inlined from 'boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_set<DomainT, Compare, Interval, Alloc>::add_over(const interval_type&, iterator) [with DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at /usr/include/boost/icl/interval_set.hpp:180:49,
    inlined from 'boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::iterator boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::_add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at /usr/include/boost/icl/interval_base_set.hpp:513:32,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const segment_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at /usr/include/boost/icl/interval_base_set.hpp:258:13,
    inlined from 'typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::segment_type&) [with Type = interval_set<unsigned int>]' at /usr/include/boost/icl/concept/interval_set.hpp:71:22,
    inlined from 'typename boost::enable_if<boost::icl::is_interval_set<Type>, Type>::type& boost::icl::add(Type&, const typename Type::element_type&) [with Type = interval_set<unsigned int>]' at /usr/include/boost/icl/concept/interval_set.hpp:79:20,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::add(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at /usr/include/boost/icl/interval_base_set.hpp:252:24,
    inlined from 'SubType& boost::icl::interval_base_set<SubType, DomainT, Compare, Interval, Alloc>::insert(const element_type&) [with SubType = boost::icl::interval_set<unsigned int>; DomainT = unsigned int; Compare = std::less; Interval = boost::icl::discrete_interval<unsigned int, std::less>; Alloc = std::allocator]' at /usr/include/boost/icl/interval_base_set.hpp:289:19,
    inlined from 'bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)' at /tmp/vsomeip/implementation/security/src/policy.cpp:71:23:
/usr/include/boost/icl/detail/interval_set_algo.hpp:358:5: error: writing 9 bytes into a region of size 8 [-Werror=stringop-overflow=]
  358 |     const_cast<value_type&>(key_value<Type>(first_))
      |     ^~~~~~~~~~
/tmp/vsomeip/implementation/security/src/policy.cpp: In member function 'bool vsomeip_v3::policy::deserialize(const vsomeip_v3::byte_t*&, uint32_t&)':
/tmp/vsomeip/implementation/security/src/policy.cpp:70:37: note: at offset 40 into destination object 'its_gid_set' of size 48
   70 |     boost::icl::interval_set<gid_t> its_gid_set;
      |                                     ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/vsomeip3.dir/build.make:1224: CMakeFiles/vsomeip3.dir/implementation/security/src/policy.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /usr/include/boost/icl/concept/interval_set.hpp:14,
                 from /usr/include/boost/icl/associative_interval_container.hpp:16,
                 from /usr/include/boost/icl/interval_base_set.hpp:24,
                 from /usr/include/boost/icl/interval_set.hpp:14,
                 from /tmp/vsomeip/implementation/routing/src/../include/../../configuration/include/configuration.hpp:16,
                 from /tmp/vsomeip/implementation/routing/src/../include/routing_manager_base.hpp:28,
[... cut ...]
/tmp/vsomeip/implementation/routing/src/routing_manager_client.cpp:2835:44: note: at offset 40 into destination object 'its_gid_set' of size 48
 2835 |         boost::icl::interval_set<uint32_t> its_gid_set;
      |                                            ^~~~~~~~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/vsomeip3.dir/build.make:1126: CMakeFiles/vsomeip3.dir/implementation/routing/src/routing_manager_client.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:191: CMakeFiles/vsomeip3.dir/all] Error 2
make: *** [Makefile:146: all] Error 2
````

duartenfonseca, 2025-04-08:
@brenkem @pberginkonsult https://github.com/COVESA/vsomeip/pull/879 <- does this PR solve the issues?

duartenfonseca, 2025-05-30:
Hey all. I was able to reproduce the problem, and using 3.5.5 you can check that this has been fixed. 
here is how i validated it:
- [boost.json](https://github.com/user-attachments/files/20523756/boost.json)
- [full-manifest.json](https://github.com/user-attachments/files/20523759/full-manifest.json)
- [vsomeip3.json](https://github.com/user-attachments/files/20523761/vsomeip3.json)

``` bash
sudo apt install flatpak flatpak-builder
flatpak install flathub org.freedesktop.Platform//22.08
flatpak install flathub org.freedesktop.Sdk//22.08
mdkir test_vsip_flatpak
cd test_vsip_flatpak
# move here the 3 jsons
mkdir build-dir
flatpak-builder --force-clean build-dir full-manifest.json
```

closed 2025-05-30 by duartenfonseca