COVESA/vsomeip #532: vsomeip between rasberry pi and pc

ogkang1201, 2023-10-10:
i tried helloWorld client on rasberry pi and helloWorld server on pc but it doesn't work. its kinda funny that the code that i used works perfectly on pc to pc communication but doesn't work on rasberry pi.

here's my code and file

```CMakeLists.txt
cmake_minimum_required(VERSION 2.8)

set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -pthread -std=c++0x")
include_directories(
   src-gen
   $ENV{RUNTIME_PATH}/capicxx-core-runtime/include
   $ENV{RUNTIME_PATH}/capicxx-someip-runtime/include
   $ENV{RUNTIME_PATH}/vsomeip/interface
)
link_directories(
   $ENV{RUNTIME_PATH}/capicxx-core-runtime/build
   $ENV{RUNTIME_PATH}/capicxx-someip-runtime/build
   $ENV{RUNTIME_PATH}/vsomeip/build
)
add_executable(HelloWorldClient
   src/HelloWorldClient.cpp
   src-gen/v1/commonapi/HelloWorldSomeIPProxy.cpp
   src-gen/v1/commonapi/HelloWorldSomeIPDeployment.cpp
)
target_link_libraries(HelloWorldClient CommonAPI CommonAPI-SomeIP vsomeip3)
add_executable(HelloWorldService
   src/HelloWorldService.cpp
   src/HelloWorldStubImpl.cpp
   src-gen/v1/commonapi/HelloWorldSomeIPStubAdapter.cpp
   src-gen/v1/commonapi/HelloWorldStubDefault
   src-gen/v1/commonapi/HelloWorldSomeIPDeployment
)
target_link_libraries(HelloWorldService CommonAPI CommonAPI-SomeIP vsomeip3)
``` 

My Server Jsonfile
```c++
{
    "unicast" : "192.168.0.28",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false"},
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "service",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
``` 
my client json file 
{
    "unicast" : "192.168.0.74",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "Hello",
            "id" : "0x1313"
        } 
    ],
    "routing" : "client",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

dprogm, 2023-10-10:
The first thing that catches my eye is the different application name. You are telling SomeIP via `VSOMEIP_APPLICATION_NAME=Service` the application name: here it is `Service` respectively `client`. This should also be reflected inside the JSON configuration file where you typed `Hello` instead. Furthermore:
* Make sure to use the correct IP addresses on both ends
* Disable a possible firewall
* I guess you cross compiled for the Raspberry Pi (ARM)

Please take also a look into this: https://github.com/COVESA/capicxx-someip-runtime/issues/26 I had similar issues but the configuration should be okay at least.

gunnarx, 2023-10-16:
@ogkang1201 I believe you can (should) close your own issue now if the problem is solved.

closed 2023-10-23 by goncaloalmeida