COVESA/vsomeip #537: Cannot connect to server: Didn't receive valid clientID! Won't register application.

dprogm, 2023-10-14:
I am currently facing the following issue: I am trying to run the `HelloWorld` example and getting the following error messages, depending on what I am starting first (it only applies on the side running the proxy routing manager part).

**Client Side Logging**

```bash
[CAPI][DEBUG] Adding proxy mapping: local:org.example.HelloWorld:v1_0:test --> libproxy_binding_shared_lib.so
[CAPI][DEBUG] Adding stub mapping: local:org.example.HelloWorld:v1_0:test --> libstub_binding_shared_lib.so
[CAPI][INFO] Loading configuration file 'example/config/commonapi.ini'
[CAPI][INFO] Using default binding 'someip'
[CAPI][INFO] Using default shared library folder '/usr/local/lib/commonapi'
[CAPI][DEBUG] Loading library for local:org.example.HelloWorld:v1_0:test proxy.
[CAPI][DEBUG] Added address mapping: local:org.example.HelloWorld:v1_0:test <--> [1234.5678(1.0)]
[CAPI][VERBOSE] Registering function for creating "org.example.HelloWorld:v1_0" proxy.
[CAPI][DEBUG] Loading interface library "libproxy_binding_shared_lib.so" succeeded.
[CAPI][VERBOSE] Creating proxy for "local:org.example.HelloWorld:v1_0:test"
2023-10-14 18:54:20.762361 [info] Using configuration file: "example/config/someip.json".
2023-10-14 18:54:20.763305 [info] Parsed vsomeip configuration in 0ms
2023-10-14 18:54:20.763497 [info] Configuration module loaded.
2023-10-14 18:54:20.763664 [info] Initializing vsomeip (3.3.8) application "ClientExample".
2023-10-14 18:54:20.764403 [info] Instantiating routing manager [Proxy].
2023-10-14 18:54:20.765730 [info] Client [2000] is connecting to [0] at /tmp/vsomeip-0
2023-10-14 18:54:20.767089 [info] Application(ClientExample, 2000) is initialized (11, 100).
2023-10-14 18:54:20.770588 [info] Starting vsomeip application "ClientExample" (2000) using 2 threads I/O nice 255
[CAPI][DEBUG] Loading library for local:org.example.HelloWorld:v1_0:test proxy.
[CAPI][VERBOSE] Creating proxy for "local:org.example.HelloWorld:v1_0:test"
2023-10-14 18:54:20.772470 [info] main dispatch thread id from application: 2000 (ClientExample) is: 7f8ef5dfd640 TID: 80922
2023-10-14 18:54:20.774035 [info] io thread id from application: 2000 (ClientExample) is: 7f8ef65fe640 TID: 80921
2023-10-14 18:54:20.776504 [info] io thread id from application: 2000 (ClientExample) is: 7f8ef4dfb640 TID: 80924
2023-10-14 18:54:20.773027 [info] shutdown thread id from application: 2000 (ClientExample) is: 7f8ef55fc640 TID: 80923
2023-10-14 18:54:20.778046 [error] Didn't receive valid clientID! Won't register application.
2023-10-14 18:54:23.778010 [warning] Client 0x2000 request client timeout! Trying again...
2023-10-14 18:54:23.779073 [info] local_uds_client_endpoint_impl::receive_cbk Error: Operation canceled
2023-10-14 18:54:23.784196 [error] Didn't receive valid clientID! Won't register application.
2023-10-14 18:54:26.783143 [warning] Client 0x2000 request client timeout! Trying again...
2023-10-14 18:54:26.785013 [info] local_uds_client_endpoint_impl::receive_cbk Error: Operation canceled
2023-10-14 18:54:26.794067 [error] Didn't receive valid clientID! Won't register application.
```

**Server Side Logging**

```bash
[CAPI][DEBUG] Adding proxy mapping: local:org.example.HelloWorld:v1_0:test --> libproxy_binding_shared_lib.so
[CAPI][DEBUG] Adding stub mapping: local:org.example.HelloWorld:v1_0:test --> libstub_binding_shared_lib.so
[CAPI][INFO] Loading configuration file 'example/config/commonapi.ini'
[CAPI][INFO] Using default binding 'someip'
[CAPI][INFO] Using default shared library folder '/usr/local/lib/commonapi'
[CAPI][DEBUG] Loading library for local:org.example.HelloWorld:v1_0:test stub.
[CAPI][DEBUG] Added address mapping: local:org.example.HelloWorld:v1_0:test <--> [1234.5678(1.0)]
[CAPI][INFO] Registering function for creating "org.example.HelloWorld:v1_0" stub adapter.
[CAPI][DEBUG] Loading interface library "libstub_binding_shared_lib.so" succeeded.
[CAPI][INFO] Registering stub for "local:org.example.HelloWorld:v1_0:test"
2023-10-14 18:50:50.124269 [info] Using configuration file: "example/config/someip.json".
2023-10-14 18:50:50.139539 [info] Parsed vsomeip configuration in 2ms
2023-10-14 18:50:50.139727 [info] Configuration module loaded.
2023-10-14 18:50:50.139936 [info] Initializing vsomeip (3.3.8) application "ServerExample".
2023-10-14 18:50:50.140548 [info] Instantiating routing manager [Host].
2023-10-14 18:50:50.142906 [info] create_routing_root: Routing root @ /tmp/vsomeip-0
2023-10-14 18:50:50.149129 [info] Service Discovery enabled. Trying to load module.
2023-10-14 18:50:50.379900 [info] Service Discovery module loaded.
2023-10-14 18:50:50.381882 [info] Application(ServerExample, 3000) is initialized (11, 100).
2023-10-14 18:50:50.383912 [info] Starting vsomeip application "ServerExample" (3000) using 2 threads I/O nice 255
2023-10-14 18:50:50.385770 [info] Client [3000] routes unicast:192.168.1.8, netmask:255.255.255.0
2023-10-14 18:50:50.385817 [info] main dispatch thread id from application: 3000 (ServerExample) is: 7f14f0bfd640 TID: 80317
2023-10-14 18:50:50.386051 [info] shutdown thread id from application: 3000 (ServerExample) is: 7f14ebfff640 TID: 80318
2023-10-14 18:50:50.389054 [info] Watchdog is disabled!
2023-10-14 18:50:50.389771 [info] OFFER(3000): [1234.5678:1.0] (true)
2023-10-14 18:50:50.390698 [info] io thread id from application: 3000 (ServerExample) is: 7f14f13fe640 TID: 80316
2023-10-14 18:50:50.390799 [info] io thread id from application: 3000 (ServerExample) is: 7f14eaffd640 TID: 80320
2023-10-14 18:50:50.391716 [info] vSomeIP 3.3.8 | (default)
2023-10-14 18:50:50.400289 [info] create_local_server: Listening @ /tmp/vsomeip-3000
Successfully Registered Service!
Waiting for calls... (Abort with CTRL+C)
2023-10-14 18:51:00.393194 [info] vSomeIP 3.3.8 | (default)
```

Can someone help out here what I am missing?

biggoldenbee, 2023-10-16:
No see multicast joins on "192.168.1.8"

biggoldenbee, 2023-10-23:
@dprogm yes, running the client/server locally it bases on local socket (Domain socket), it doesn't go into tcp/ip stack. you may reference the chapter 17.2 of "Advanced Programming in the UNIX Environment, 3rd Edition":
“UNIX domain sockets only copy data; they have no protocol processing to perform,
no network headers to add or remove, no checksums to calculate, no sequence numbers to generate,
and no acknowledgments to send.”. This means that vsomeipd only packs data stream into SOME/IP
payload and unpacks data stream from a SOME/IP payload, and data stream doesn’t go into UDP
TCP/IP stack, so there is no length limitation of UDP and there is no acknowledgments of TCP.

fcmonteiro, 2023-11-14:
@dprogm please provide both json files used, command used to launch the service/client.
Also, what OS and version is being used?

biggoldenbee, 2023-11-17:
Please check the attachments, as below:
[ipv6.zip](https://github.com/COVESA/vsomeip/files/13390026/ipv6.zip)
VM0: ubuntu 18.04, Linux ubuntu 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
VM1: ubuntu 18.04,  Linux ubuntu 4.15.0-33-generic #36-Ubuntu SMP Wed Aug 15 16:00:05 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux

duartenfonseca, 2024-09-26:
@dprogm do you still have this issue? can you provide your json configs?

duartenfonseca, 2024-11-04:
closing as there was no response for a month

closed 2024-11-04 by duartenfonseca