COVESA/vsomeip #539: vsomeip-qnx issue with running on qnx platform

12falcon, 2023-10-18:
hello,
I've build vsomeip for qnx on linux through cross compilation
I'm getting an issue while running vsomeip on qnx as follows

# env VSOMEIP_CONFIGURATION=/var/qnxlib/vsomeip-local.json VSOMEIP_APPLICATION_NAME=client-sample ./subscribe-sample
1970-00-01 00:52:18.384238 [warning] Reliability type for event [1234.5678.777] was not configured Using : RT_UNRELIABLE
1970-00-01 00:52:18.384238 [warning] Reliability type for event [1234.5678.778] was not configured Using : RT_UNRELIABLE
1970-00-01 00:52:18.384238 [warning] Reliability type for event [1234.5678.779] was not configured Using : RT_UNRELIABLE
1970-00-01 00:52:18.384238 [info] Parsed vsomeip configuration in 3ms
1970-00-01 00:52:18.384238 [info] Using configuration file: "/var/qnxlib/vsomeip-local.json".
1970-00-01 00:52:18.384238 [info] Configuration module loaded.
1970-00-01 00:52:18.384238 [info] Initializing vsomeip application "client-sample".
1970-00-01 00:52:18.384238 [info] Instantiating routing manager [Proxy].
1970-00-01 00:52:18.384238 [info] Client [1344] is connecting to [0] at /tmp/vsomeip-0
1970-00-01 00:52:18.384238 [info] Application(client-sample, 1344) is initialized (11, 100).
Client settings [protocol=UDP]
1970-00-01 00:52:18.384248 [info] Starting vsomeip application "client-sample" (1344) using 2 threads I/O nice 255
1970-00-01 00:52:18.384248 [info] main dispatch thread id from application: 1344 (client-sample) is: 2
1970-00-01 00:52:18.384248 [info] shutdown thread id from application: 1344 (client-sample) is: 3
1970-00-01 00:52:18.384248 [info] io thread id from application: 1344 (client-sample) is: 4
1970-00-01 00:52:18.384248 [info] io thread id from application: 1344 (client-sample) is: 1
1970-00-01 00:52:18.384248 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:18.385259 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:18.387269 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:19.391280 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:19.399282 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:21.415286 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:23.431290 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:24.447295 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:26.463299 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
1970-00-01 00:52:27.479323 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED

Is there any solution to resolve this issue?

12falcon, 2023-10-18:
I observed with changing the name in application and in json file that segmentation fault was occurring after below logs...

env VSOMEIP_CONFIGURATION=vsomeip-local.json VSOMEIP_APPLICATION_NAME=client-sample ./notify-sample
1970-00-01 00:43:16.968223 [warning] Reliability type for event [1234.5678.777] was not configured Using : RT_UNRELIABLE
1970-00-01 00:43:16.968223 [warning] Reliability type for event [1234.5678.778] was not configured Using : RT_UNRELIABLE
1970-00-01 00:43:16.968223 [warning] Reliability type for event [1234.5678.779] was not configured Using : RT_UNRELIABLE
1970-00-01 00:43:16.968233 [info] Parsed vsomeip configuration in 3ms
1970-00-01 00:43:16.968233 [info] Using configuration file: "vsomeip-local.json".
1970-00-01 00:43:16.968233 [info] Configuration module loaded.
1970-00-01 00:43:16.968233 [info] Initializing vsomeip application "app-name".
1970-00-01 00:43:16.968233 [info] Instantiating routing manager [Host].
1970-00-01 00:43:16.968233 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
1970-00-01 00:43:16.968233 [error] Server endpoint creation failed. Client ID: 0000: acceptor bind: Permission denied
1970-00-01 00:43:16.968233 [info] Service Discovery enabled. Trying to load module.
1970-00-01 00:43:16.968263 [info] Service Discovery module loaded.
1970-00-01 00:43:16.968263 [info] Application(app-name, 0100) is initialized (11, 100).
1970-00-01 00:43:16.968263 [info] REGISTER EVENT(0100): [1234.5678.8778:is_provider=true]
1970-00-01 00:43:16.968263 [info] Starting vsomeip application "app-name" (0100) using 2 threads I/O nice 255
1970-00-01 00:43:16.968263 [info] OFFER(0100): [1234.5678:0.0] (true)
1970-00-01 00:43:16.968263 [info] main dispatch thread id from application: 0100 (app-name) is: 4
1970-00-01 00:43:16.968263 [info] shutdown thread id from application: 0100 (app-name) is: 5
1970-00-01 00:43:16.968263 [error] create_server_endpoint Server endpoint creation failed. Reason: bind: Can't assign requested address Port: 30490 (reliable=unreliable)
1970-00-01 00:43:16.968263 [error] Service Discovery endpoint could not be created. Please check your network configuration.
1970-00-01 00:43:16.968263 [error] Couldn't start service discovery
1970-00-01 00:43:16.968263 [info] SOME/IP routing ready.
1970-00-01 00:43:16.968263 [info] create_local_server Routing endpoint at /tmp/vsomeip-0
1970-00-01 00:43:16.968263 [error] create_server_endpoint Server endpoint creation failed. Reason: bind: Can't assign requested address Port: 30509 (reliable=unreliable)
1970-00-01 00:43:16.968263 [error] Server endpoint creation failed. Client ID: 0000: acceptor bind: Permission denied
1970-00-01 00:43:16.968263 [info] Watchdog is disabled!
1970-00-01 00:43:16.968263 [error] Local server endpoint creation failed. Client ID: 0100 Path: /tmp/vsomeip-100 Reason: acceptor bind: Permission denied
Application app-name is registered.

ajinkyak20, 2023-11-05:
Hello 12falcon,
 I think this issue is because of permission for crating socket files on board. Can you check if any configuration required or any  security policy for creating socket files.
 
 Regards,
 ajinkyak20

ajinkyak20, 2023-11-07:
Hello 12falcon,
Can you check with secpol tool and check what error you are getting on /dev/secpolgenerate/errors  and send it here in comments

Regards,
ajinkyak20

12falcon, 2023-11-08:
Hello,

Yes please find output 
---
1970-01-01 00:01:19 io-pkt-v6-hc (pid:245792 type:io_pkt_v6_hc_t) failed to attach to path /var/vsomeip-0: no rule
 ---
I'm getting above logs. I've resolved the issue by updating security policy for vsomeip files

closed 2023-11-08 by 12falcon

mhappyboy, 2024-07-06:
Hello,
I am try to run rpcbind service in qnx system, but there is an error similarly to above, I don't know how to configure the security policy, the error message is :
# rpcbind
rpcbind: Cannot bind `local': Permission denied
rpcbind: Failed to open /var/run/rpcbind.sock: Permission denied
# secpolgenerate: io-pkt-v6-hc (pid:77852 type:io_pkt_v6_hc_t) failed to attach to path /var/run/rpcbind.sock: no rule
how can I configure the security policy? thanks!