COVESA/vsomeip #540: Failed to handle eventgroup subscription for multiple routing managers with single process

zhaoxin39913, 2023-10-18:
Hi vsomeip experts:

Currently I am trying to test multiple routing managers in V3.3.8 (with build option ENABLE_MULTIPLE_ROUTING_MANAGERS=1) in single process for VLAN user case. However I encountered eventgroup subscription error in my test. Here are my test setup:

**Server Device:**
Simply modify example code "notify-sample.cpp" to create 2 vsomeip instances in notify-sample process, each instance has a dedicated JSON file to bind its own IP address in different VLAN.
instance 1:
    _name: "notify_sample_1" 
    file: "notify_sample_1.json"
    unicast : "10.10.1.104"
    netmask : "255.255.255.0"
    multicast: 225.0.0.1_

instance 2:
    _name: "notify_sample_2"
    file: notify_sample_2.json
    unicast: "10.10.2.104"
    netmask: "255.255.255.0"
    multicast: 226.0.0.1_

**Client Device 1:**
subscribe-sample process -> notify_sample_1
   _unicast: "10.10.1.101"
   netmask: "255.255.255.0"
   multicast: 225.0.0.1_

**Client Device 2:**
subscribe-sample process -> notify_sample_2
   _unicast: "10.10.2.102"
   netmask: "255.255.255.0"
   multicast: 226.0.0.1_

On Server device side I run below command which starts 2 vsomeip instances in notify-sample process but always get below errors. However I didn't see such issue if I run notify_sample_1 and notify_sample_2 in separate processes. I think it should be a bug. 
`zhax@zhax-desktop:~/work/COVESA/src/vsomeip/build/examples$ VSOMEIP_CONFIGURATION_notify_sample_1=./notify_sample_1.json VSOMEIP_CONFIGURATION_notify_sample_2=./notify_sample_2.json ./notify-sample
2023-10-18 17:10:47.250235 [info] Using configuration file: "./notify_sample_1.json".
2023-10-18 17:10:47.252889 [info] Parsed vsomeip configuration in 1ms
2023-10-18 17:10:47.253443 [info] Initializing vsomeip (3.3.8) application "notify_sample_1".
2023-10-18 17:10:47.254329 [info] Instantiating routing manager [Host].
2023-10-18 17:10:47.257402 [info] create_routing_root: Routing root @ /tmp/notify_sample_1-0
2023-10-18 17:10:47.260614 [info] Service Discovery enabled. Trying to load module.
2023-10-18 17:10:47.269711 [info] Service Discovery module loaded.
2023-10-18 17:10:47.271031 [info] Application(notify_sample_1, 0111) is initialized (11, 100).
2023-10-18 17:10:47.271592 [info] offer_event: Event [1234.5678.8778] uses configured cycle time 0ms
2023-10-18 17:10:47.272285 [info] REGISTER EVENT(0111): [1234.5678.8778:is_provider=true]
2023-10-18 17:10:47.273334 [info] OFFER(0111): [1234.5678:0.0] (true)
2023-10-18 17:10:47.277271 [info] Starting vsomeip application "notify_sample_1" (0111) using 2 threads I/O nice 255
2023-10-18 17:10:47.278073 [info] create_local_server: Listening @ /tmp/notify_sample_1-111
Setting event (Length=1).
2023-10-18 17:10:47.283114 [info] main dispatch thread id from application: 0111 (notify_sample_1) is: ffffb260f100 TID: 700048
2023-10-18 17:10:47.284613 [info] Using configuration file: "./notify_sample_2.json".
2023-10-18 17:10:47.284034 [info] Client [0111] routes unicast:10.10.1.104, netmask:255.255.255.0
2023-10-18 17:10:47.284963 [info] shutdown thread id from application: 0111 (notify_sample_1) is: ffffb1dff100 TID: 700049
2023-10-18 17:10:47.290826 [info] Watchdog is disabled!
Application notify_sample_1 is registered.
2023-10-18 17:10:47.292896 [info] io thread id from application: 0111 (notify_sample_1) is: ffffb2e1f100 TID: 700047
2023-10-18 17:10:47.293133 [info] io thread id from application: 0111 (notify_sample_1) is: ffffb0ddf100 TID: 700051
2023-10-18 17:10:47.295273 [info] vSomeIP 3.3.8 | (default)
2023-10-18 17:10:47.296075 [info] Parsed vsomeip configuration in 0ms
2023-10-18 17:10:47.296569 [info] Initializing vsomeip (3.3.8) application "notify_sample_2".
2023-10-18 17:10:47.296655 [info] Network interface "eth0.1" state changed: up
2023-10-18 17:10:47.297433 [info] Instantiating routing manager [Host].
2023-10-18 17:10:47.300810 [info] create_routing_root: Routing root @ /tmp/notify_sample_2-0
2023-10-18 17:10:47.301189 [info] Route "225.0.0.1/32 if: eth0.1 gw: n/a" state changed: up
2023-10-18 17:10:47.302513 [info] udp_server_endpoint_impl: SO_RCVBUF is: 212992 (1703936) local port:30490
2023-10-18 17:10:47.302670 [info] Service Discovery enabled. Trying to load module.
2023-10-18 17:10:47.303517 [info] Service Discovery module loaded.
2023-10-18 17:10:47.304173 [info] Application(notify_sample_2, 0222) is initialized (11, 100).
2023-10-18 17:10:47.304552 [info] offer_event: Event [1234.5678.8778] uses configured cycle time 0ms
2023-10-18 17:10:47.305105 [info] REGISTER EVENT(0222): [1234.5678.8778:is_provider=true]
2023-10-18 17:10:47.306189 [info] udp_server_endpoint_impl<multicast>: SO_RCVBUF is: 212992 (1703936) local port:30490
[... 419 lines cut ...]
                {
		    "eventgroup" : "0x4466",
                    "multicast" : { "address" : "238.238.0.1", "port" : "45678" },
                    "threshold" : "1"
                }
            ]
        }
    ],
    "routing" : "notify_sample_2",
    "network" : "notify_sample_2",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "226.0.0.1",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}`

zhaoxin39913, 2023-11-03:
Can this bug be fixed in upcoming release?

zhaoxin39913, 2024-01-17:
Hi COVESA team,

This issue is getting fixed by commenting out below code (between "#ifdef 0" and "endif") in below function in service_discovery_impl.cpp. Could you please check if the change is reasonable or not?

```
bool service_discovery_impl::check_ipv4_address(
        const boost::asio::ip::address& its_address) const {
    //Check unallowed ipv4 address
    bool is_valid = true;

    static const boost::asio::ip::address_v4::bytes_type its_unicast_address =
            unicast_.to_v4().to_bytes();
    const boost::asio::ip::address_v4::bytes_type endpoint_address =
            its_address.to_v4().to_bytes();
    static const boost::asio::ip::address_v4::bytes_type its_netmask =
            configuration_->get_netmask().to_v4().to_bytes();

    //same address as unicast address of DUT not allowed
    if (its_unicast_address == endpoint_address) {
        VSOMEIP_ERROR << "Subscriber's IP address is same as host's address! : "
                << its_address;
        is_valid = false;
    } else {
#if 0
        const std::uint32_t self = VSOMEIP_BYTES_TO_LONG(its_unicast_address[0],
                its_unicast_address[1], its_unicast_address[2], its_unicast_address[3]);
        const std::uint32_t remote = VSOMEIP_BYTES_TO_LONG(endpoint_address[0],
                endpoint_address[1], endpoint_address[2], endpoint_address[3]);
        const std::uint32_t netmask = VSOMEIP_BYTES_TO_LONG(its_netmask[0],
                its_netmask[1], its_netmask[2], its_netmask[3]);
        if ((self & netmask) != (remote & netmask)) {
            VSOMEIP_ERROR<< "Subscriber's IP isn't in the same subnet as host's IP: "
                    << its_address;
            is_valid = false;
#endif
        }
    }
    return is_valid;
}
```

vramasubbu, 2024-02-16:
@zhaoxin39913 , I also faced the same issue , Is that issue get resolved on with the mentioned solution 
Do we have any updated version on this?

zhaoxin39913, 2024-04-09:
> @zhaoxin39913 , I also faced the same issue , Is that issue get resolved on with the mentioned solution Do we have any updated version on this?

I see this is already fixed in latest branch but not tagged yet.

omitheaussie, 2024-10-30:
I am also facing this issue. I have two machines running on two different subnets but connected to a multicast network. I am able to communicate between the two machines via the multicast IP address - I tested using iperf). I am getting following error when I start a sender application on one machine and receiver app on another machine. The receiver app tried to subscribe to the sender but I get an error as shown below in the sender machine log:
[image]

duartenfonseca, 2024-12-03:
@zhaoxin39913 can you retry with the most recent version?

omitheaussie, 2024-12-03:
I cannot test with latest as there are other dependencies that depend on this version. But I tried suggestion as per "zhaoxin39913 on Jan 17" and it worked for me.

duartenfonseca, 2025-05-27:
@zhaoxin39913 
some changes that i would like you to try:
- see ifconfig. 
- no need to use ENABLE_MULTIPLE_ROUTING_MANAGERS when building flag. 
- try in a more recent version