COVESA/vsomeip #543: Issue in building the vsomeip for QNX SDP 7.1

akashnagpalofficial, 2023-10-20:
Hi I followed the process of QNX build from the commit comment #34d3173 so can you help me in the installing required libraries and building it for QNX SDP 7.1

ERROR

CMake Error in CMakeLists. txt:
Found relative path while evaluating include directories of
"vsomeip3":
"Boost_INCLUDE_DIR-NOTFOUND"
CMake Error in CMakeLists.txt:
Found relative path while evaluating include directories of
"vsomeip3-cfg":
"Boost_INCLUDE_DIR-NOTFOUND"
... many more related same issue.

And after setting the include and library path in the cmakelist.txt the process gone upto 6 percent and then error related to boost library came..

I wonder which library version if boost is required for building vsomeip as mentioned by @chachoi in the commit comment #34d3173 is not able to build.
So please clarify the exact process.

chachoi, 2023-10-20:
Hello,

I should've put a README with build instructions in the build_qnx folder.
I will probably create a pull request for that.

For now, you can use the instruction below:
```bash
# source the QNX environment script
source <path-to-qnxsdp-env.sh>/qnxsdp-env.sh

# Create a workspace
mkdir ~/vsomeip_workspace
WORKSPACE=~/vsomeip_workspace

# Build and install boost
cd $WORKSPACE
git clone https://gitlab.com/qnx/libs/boost.git && cd boost
git checkout QNX_7.1_v1.63.0
git submodule update --init --recursive
cd tools/build
git apply ../../boost_tools_build_qnx.patch
cd -
mkdir -p ./boost/utility
cp -r ./libs/utility/include/boost/utility/detail ./boost/utility
JLEVEL=8 make install

# Get googletest
cd $WORKSPACE
git clone https://gitlab.com/qnx/libs/googletest.git
export GTEST_ROOT=$WORKSPACE/googletest

# Build and install vsomeip
git clone https://github.com/chachoi/vsomeip.git && cd vsomeip/build_qnx

# Before building, look at $WORKSPACE/vsomeip/build_qnx/common.mk
# set TEST_IP_MASTER to the target ip address
# set TEST_IP_SLAVE to the host PC ip address for tests
JLEVEL=8 make install
```

akashnagpalofficial, 2023-10-30:
Hi @chachoi ,
Thanks for your help!
I tried to install the by the above Steps-

1. While Installing the Boost Library 
     I found that Make Says : INSTALL is up to date.
      -> So I tried to do make clean and then make so able to generate binaries with 
           "qcc" compiler.
  2. While Installing the  vsomeip after running the Make install command I faced 
       "vsomeip3":
        "Boost_INCLUDE_DIR-NOTFOUND"
         CMake Error in CMakeLists.txt:
         Found relative path while evaluating include directories of
         "vsomeip3-cfg":
         "Boost_INCLUDE_DIR-NOTFOUND"
        -> So I set the BOOST include library path manually and tried to build it again so 
             I am able to configure but faced the Issue while building.
3.  While building it shows fatel Error : boost/asio/io_context.hpp not found.
      -> when I analysed about the Issue so I found that the branch QNX_7.1_v1.63.0 which you mentioned above this file is not exists,

So could you please confirm again about the exact version and steps to Follow for building it for QNX.

Thanks

akashnagpalofficial, 2023-11-13:
Hello

Can you help in build the application and libraries for qnx 7.1.0

MinhBQ2602, 2023-11-14:
hi @chachoi ,
I face issue when following your guide:

1. No inlcude dir in libs/utility
_cp -r ./libs/utility/include/boost/utility/detail ./boost/utility_

2. Then make command error
_cp -r /someip/boost/libs/*/include/boost /qnx-sdp/target/qnx7/usr/include
cp: cannot stat '/someip/boost/libs/*/include/boost': No such file or directory
../../../common.mk:71: recipe for target 'install' failed_

Could you double check source code in your side?

akashbwj, 2024-02-02:
Hi @chachoi 
I followed all the steps you mentioned. On the last step - JLEVEL=8 make install
I get this error:

fatal error: systemd/sd-daemon.h: No such file or directory
 #include <systemd/sd-daemon.h>
          ^~~~~~~~~~~~~~~~~~~~~
compilation terminated.

I am using ubuntu, and installing libsystemd-dev package also didn't help. Any clues about this error?

dappchandu, 2024-02-21:
Hi , can you help me for this error

 JLEVEL=8 make install
Makefile:8: recurse.mk: No such file or directory
make: *** No rule to make target 'recurse.mk'.  Stop.

akashnagpalofficial, 2024-02-22:
The below steps work perfectly fine for me:

Make workspace
mkdir -p ~/vsomeip_workspace && cd ~/vsomeip_workspace
WORKSPACE=${PWD}

# Don't forget to source the qnxsdp-env.sh script

# boost build steps
git clone git@gitlab.com:qnx/libs/boost -b QNX_7.1_v1.63.0 && cd boost
git submodule update --init --recursive

# Remove iosock variants if you do not need them
rm -rf nto/aarch64/le.iosock nto/x86_64/o.iosock

# Apply patch to tools/build
cd tools/build
git apply ../../boost_tools_build_qnx.patch
cd -

# Copy utility folder to avoid a build error
mkdir -p ./boost/utility
cp -r ./libs/utility/include/boost/utility/detail ./boost/utility

# Build and install
JLEVEL=4 make install

# vsomeip build steps
cd $WORKSPACE
# Optional: get googletest if you want to run vsomeip tests
git clone git@gitlab.com:qnx/libs/googletest.git -b QNX_7.1_v1.13.0
export GTEST_ROOT=$WORKSPACE/googletest

# Clone vsomeip
git clone https://github.com/COVESA/vsomeip.git && cd vsomeip
git checkout 3.4.10

# Edit TEST_IP_MASTER and TEST_IP_SLAVE in build_qnx/common.mk for tests!
# I personally put the IP address of my HOST LINUX PC for TEST_IP_SLAVE
# and the IP address of my target for TEST_IP_MASTER.

JLEVEL=4 make -C build_qnx/ install

closed 2024-02-22 by akashnagpalofficial