COVESA/vsomeip #544: Communication between two devices without service discovery with static routing

jennyj1, 2023-10-20:
Hello Community,

I am trying to establish the communication between two devices (device1 10.10.1.91/32 and device2 10.10.1.92/32).

I am using the static routing concept and have changed the configuration (json) file by adding unicast address under Services for both sender and receiver. 

But the data is not sent. 
1. Can someone help with if any other changes needs to be made in the configuration for it to run. 
2. What does the sender and the receiver code look like without service discovery with static routing?

jennyj1, 2023-10-25:
Hi @nachtfuchs,

Thank you for your quick and detailed response.

I tried running the code by replacing iPv6 addresses with iPv4 addresses since that is my application. But communication is not established. Do we need to make any other changes in order to run the code with iPv4 addressing. 

Please Note: My aim is to have a client server communication without service discovery between two devices.

nachtfuchs, 2023-10-25:
Did you setup your network connections correctly? Can you ping between them?
What do you see in Wireshark? Do you see outgoing messages?
And when you start your application, do you see that your application is loading the correct ethernet device? Usually it tells which IPs it is using and which device it uses. This must be aligned with your network settings.

jennyj1, 2023-11-03:
Dear @nachtfuchs ,

Thanks a lot for the code and the support. I am able to run the code sent by you. But since there is no part configured as client that is why we are not able to receive any data and hence check whether our network is communicating or not and if data is being sent or not. In my case I am not able to see any data packets in wireshark also.

It would have been great if you could help with modifying the code to configure some client/receiver to capture the payload and print it.  I am not able to do it.

nachtfuchs, 2023-11-03:
If you check my code, there are two specific files. One is "request-response.cpp". The other is "request-response-mirrored.cpp". They do exactly what you are trying to achieve. One sends data, the other receives data. And vice versa, because there are two services. So the solution is - in my eyes - right in front of you.

Gokul123-dev, 2023-11-08:
Hi @nachtfuchs , I have checked the above code , in "request-response.cpp" and "request-response-mirrored.cpp" both are sending data and receiving data on both ends . It would have been great if you provide the code for one client and one server where "request-response-mirrored.cpp" acts as server sending data to client "request-response.cpp" with out service discovery . IMP : Without Service Discovery! from one server to one client.

nachtfuchs, 2023-11-08:
@Gokul123-dev The code already deactivated service discovery. If you want further freelance work, I can do it for you. It is USD 500 per hour ;-). Let me know if you are interested, I will send you my payment details.

maramharsha99, 2023-12-11:
hello @jennyj1 Can you help me with a similar problem ? I was not able to figure out the issue. I am using hello_world codes to build communication between two devices.