COVESA/vsomeip #548: Helloworld.json doesn't work

coderjingluo, 2023-10-28:
[image]
Although I copied the configuration file to the helloworld directory and ran the sh script, the json file doesn't seem to work

coderjingluo, 2023-10-28:
Also, why can't I catch any traffic about vsomeip using wireshark

fcmonteiro, 2023-10-31:
Hi @coderjingluo,
As you are starting the example using a script with multiple lines, you need to use a backslash at the end of each line.
Something like this,
[image]
[Some documentation.](https://www.tutorialspoint.com/running-multi-line-shell-code-at-once-from-single-prompt)

Also, if you are running client and the server on the same machine, this use internal communication using UDS and you will not see the traffic in wireshark.
Thanks

closed 2023-11-10 by goncaloalmeida