COVESA/vsomeip #552: package not found via tcpdump

123130, 2023-10-30:
I have run the example(hello_world, subscribe/notify and request/response) correctly, but I did not catch any packets via the command(sudo tcpdump -i lo -w someip.pcap).
Do you know where the problem is?
request:
[image]
response:
[image]
[image]

dprogm, 2023-11-12:
If you are running the client and server locally on your machine you won't see any network traffic because SomeIP is using Unix Domain Sockets for IPC which utilizes kernel buffer memory for data exchange.

123130, 2023-11-13:
@dprogm Dear you, thanks a lot for replying my issue.
So, can I use Peach fuzzer to test notify-sample locally on my machine?

dprogm, 2023-11-13:
The question is what you are trying to achieve? There are methods for sniffing on Unix Domain Sockets, e.g. by using tracepoints or kernel probes and dumping the data. Some months ago I stumbled over [sockdump](https://github.com/mechpen/sockdump), but this requires some extra work. I have asked about it here: https://github.com/mechpen/sockdump/issues/21 

I guess Peach Fuzzer is for Fuzz-Testing. This might be another  topic.

123130, 2023-11-14:
@dprogm Thanks a lot!
I also want to capture the traffic of someip, and have seen https://github.com/mechpen/sockdump/issues/21.
Would you like to share the dummy.lua?
Thanks.

Avinash19999, 2024-03-09:
@123130 Hi, If you are running the server and client applications in a single machine then you will only find the Someip-SD
message when you try to sniff it. When you run the server and client application between two machines or VM's then you can capture the Someip frames.

closed 2024-09-26 by duartenfonseca