COVESA/vsomeip #556: vsomeip: Didn't receive a multicast SD message for 200ms. Leaving the multicast group and join

guanghuinie, 2023-11-03:
1. I am conducting VSOMEIP sleep-wake stress testing in the car, and initially, VSOMEIP is working fine
2. Occasional issues: Didn't receive a multicast SD message for 200ms. Leaving the multicast group and join  
3. he issue persists after it occurs. 
4. tcpdump and another test tool receive multicast messages normally, and after restarting the VSOMEIP process, it resumes normal
5.
[image]

any  advices ? Thanks

guanghuinie, 2023-11-03:
vSomeip version :3.3.0

guanghuinie, 2023-11-03:
My initial suspicion is that the port vsomeip listens on has changed during the pressure test.

guanghuinie, 2023-11-06:
Log of Reproduction：
  leave address: error ec Cnanot assign requested address
  adresss: ********   joined error: No buffer space available .

chen53085308, 2024-03-15:
> 1. I am conducting VSOMEIP stress testing， During stress testing, there may be disconnection and reconnection of Ethernet and router.   and initially, VSOMEIP is working fine
> 2. Occasional issues: Didn't receive a multicast SD message for 200ms. Leaving the multicast group and join
> 3. the issue persists after it occurs.
> 4. tcpdump and another test tool receive multicast messages normally, and after restarting the VSOMEIP process, it resumes normal
> 
> ![wdgQNUzsZB](https://private-user-images.githubusercontent.com/11964095/280182120-3fa18d06-5d05-46d7-acc7-d8fcfe25de20.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA0NzQxMTgsIm5iZiI6MTcxMDQ3MzgxOCwicGF0aCI6Ii8xMTk2NDA5NS8yODAxODIxMjAtM2ZhMThkMDYtNWQwNS00NmQ3LWFjYzctZDhmY2ZlMjVkZTIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDAzMzY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4MTJkMzFiZjVmMWFhMWNkMmM1OTIxMjgwZDM4NjNkODJjZTYyMWE0YmIwZTg5ZWVlOGYzZTJiODUxODdjM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Tq_gPk3tiRkAz5pSjkIw9KudWz9LA91ADu45Ja1H4V0)
> 
> any advices ? Thanks

我也遇到了这个问题，我发现是因为客户端起来后发送find service的SD报文导致的，会触发服务端的多播检测定时器，如果组网中没有其他的offer service，而只有client的find service，服务端就会有这个问题。

guanghuinie, 2024-03-20:
> > 1. I am conducting VSOMEIP stress testing， During stress testing, there may be disconnection and reconnection of Ethernet and router.   and initially, VSOMEIP is working fine
> > 2. Occasional issues: Didn't receive a multicast SD message for 200ms. Leaving the multicast group and join
> > 3. the issue persists after it occurs.
> > 4. tcpdump and another test tool receive multicast messages normally, and after restarting the VSOMEIP process, it resumes normal
> > 
> > ![wdgQNUzsZB](https://private-user-images.githubusercontent.com/11964095/280182120-3fa18d06-5d05-46d7-acc7-d8fcfe25de20.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA0NzQxMTgsIm5iZiI6MTcxMDQ3MzgxOCwicGF0aCI6Ii8xMTk2NDA5NS8yODAxODIxMjAtM2ZhMThkMDYtNWQwNS00NmQ3LWFjYzctZDhmY2ZlMjVkZTIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDAzMzY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4MTJkMzFiZjVmMWFhMWNkMmM1OTIxMjgwZDM4NjNkODJjZTYyMWE0YmIwZTg5ZWVlOGYzZTJiODUxODdjM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Tq_gPk3tiRkAz5pSjkIw9KudWz9LA91ADu45Ja1H4V0)
> > any advices ? Thanks
> 
> I also encountered this problem. I found that it was caused by the client sending the SD message of find service after it got up, which would trigger the multicast detection timer of the server. If there are no other offer services in the network, but only the client's find service, the server will have this problem. ![image](https://private-user-images.githubusercontent.com/50629395/313060308-c584acab-452a-4fa1-8c6c-0ed99282bba1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA5MjY4NzMsIm5iZiI6MTcxMDkyNjU3MywicGF0aCI6Ii81MDYyOTM5NS8zMTMwNjAzMDgtYzU4NGFjYWItNDUyYS00ZmExLThjNmMtMGVkOTkyODJiYmExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIwVDA5MjI1M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTE3MWU1MzgyMDczY2NlMmMwZjk1MmNkYjNjZmZjYzQ4NWFjMTY2MTEwMThjZDFmNDY0MjA2ZDU4ZWEyMGE3NzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.2DRa6TZ1k-Q8TB6UwtiPRM-2m2zEO9_4GuKV1VjjePw)

Not the same issue,  The situation you encountered is normal one.   The situation I encountered was that the offer packet was normal, but the join failed in the protocol stack, which may be because of the underlying reason of my machine or Or maybe a bug in the protocol stack is causing the port join to fail.
   "log of Reproduction：
            leave address: error ec Can not assign requested address
           adresss: ******** joined error: No buffer space available ."

My temporary solution was to close the socket if the join failed

pomforandroid, 2024-04-28:
> > 1. 我正在进行VSOMEIP压力测试，在压力测试过程中，可能会出现以太网和路由器的断开和重新连接的情况。最初，VSOMEIP 工作正常
> > 2. 偶尔出现的问题：200 毫秒内未收到组播 SD 消息。离开多播组并加入
> > 3. 发生后问题依然存在。
> > 4. tcpdump和另一个测试工具正常接收组播报文，重启VSOMEIP进程后恢复正常
> > 
> > ![wdgQNUzsZB](https://private-user-images.githubusercontent.com/11964095/280182120-3fa18d06-5d05-46d7-acc7-d8fcfe25de20.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA0NzQxMTgsIm5iZiI6MTcxMDQ3MzgxOCwicGF0aCI6Ii8xMTk2NDA5NS8yODAxODIxMjAtM2ZhMThkMDYtNWQwNS00NmQ3LWFjYzctZDhmY2ZlMjVkZTIwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTUlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE1VDAzMzY1OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWE4MTJkMzFiZjVmMWFhMWNkMmM1OTIxMjgwZDM4NjNkODJjZTYyMWE0YmIwZTg5ZWVlOGYzZTJiODUxODdjM2QmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Tq_gPk3tiRkAz5pSjkIw9KudWz9LA91ADu45Ja1H4V0)
> > 有什么建议吗？谢谢
> 
> 我也遇到了这个问题。我发现是客户端起床后发送find service的SD消息，会触发服务器的组播检测定时器造成的。如果网络中没有其他offer服务，只有client的find服务，服务器就会出现这个问题。 ![图像](https://private-user-images.githubusercontent.com/50629395/313060308-c584acab-452a-4fa1-8c6c-0ed99282bba1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQyNzA3MDMsIm5iZiI6MTcxNDI3MDQwMywicGF0aCI6Ii81MDYyOTM5NS8zMTMwNjAzMDgtYzU4NGFjYWItNDUyYS00ZmExLThjNmMtMGVkOTkyODJiYmExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDA0MjglMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwNDI4VDAyMTMyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTFmMWQ5OWU1MTUzMjFjNmQ3N2U3MTY2ZGYyMDUwNjAzMTZmYjgyNmZhY2VkMzBkMGIxZDBhNGM2OTJiYmFjNjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.dWI6LKeDuOTOPii3FbZn_6n_TOwHs1ew32yhJJei-88)
Is this behavior normal? How can this issue be resolved? Would offering any arbitrary service help in this situation?

kshang, 2024-10-15:
> I also encountered this problem. I found that it was caused by the client sending the SD message of find service after it got up, which would trigger the multicast detection timer of the server. If [...]

这问题你最后是怎么解决的呢