COVESA/vsomeip #567: Vsomeip can't be found.

wangjingxiaomidou, 2023-11-09:
I build and install vsomeip based on the guldlie, below is my method
cmake -DCMAKE_INSTALL_PREFIX:PATH=/etc/vsomeip ..
make
make install

But whem I use the vsomeip, it still have issues.
  By not providing "Findvsomeip.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "vsomeip", but
  CMake did not find one.

  Could not find a package configuration file provided by "vsomeip"
  (requested version 2.6.0) with any of the following names:

    vsomeipConfig.cmake
    vsomeip-config.cmake

  Add the installation prefix of "vsomeip" to CMAKE_PREFIX_PATH or set
  "vsomeip_DIR" to a directory containing one of the above files.  If
  "vsomeip" provides a separate development package or SDK, be sure it has
  been installed

goncaloalmeida, 2023-11-09:
please in your cmake try to find vsomeip3, not vsomeip.

wangjingxiaomidou, 2023-11-10:
Oh, yeah, Thanks.

closed 2023-11-10 by goncaloalmeida