COVESA/vsomeip #568: How to fix the issue: "1 Configuration module could not be loaded!"?

wangjingxiaomidou, 2023-11-10:
I use the tag 3.3.8 of the code base. My cmake file is as below:
PROJECT(vSomeIPExample LANGUAGES CXX C)
CMAKE_MINIMUM_REQUIRED(VERSION 3.22.1)

set (CMAKE_CXX_FLAGS "-g -std=c++0x")

#find_package (vsomeip 2.6.0 REQUIRED)
find_package(vsomeip3)
find_package( Boost 1.55 COMPONENTS system thread log REQUIRED )
MESSAGE( STATUS "aaaaa = #${VSOMEIP_INCLUDE_DIRS}.")
include_directories (
    ${Boost_INCLUDE_DIR}
    ${VSOMEIP_INCLUDE_DIRS}
)

add_executable(service-example service-example.cpp)
add_executable(client-example client-example.cpp)
add_executable(client-example-test client-example-test.cpp)
target_link_libraries(service-example vsomeip3 ${Boost_LIBRARIES})
target_link_libraries(client-example vsomeip3 ${Boost_LIBRARIES})
target_link_libraries(client-example-test vsomeip3 ${Boost_LIBRARIES})

When I build vsomeip3lib, and go to build and run the hello world, I meet the issue, how to fix it?
[image]

goncaloalmeida, 2023-11-10:
https://github.com/COVESA/vsomeip/issues/481#issuecomment-1607168854 :)

closed 2023-11-10 by goncaloalmeida

wangjingxiaomidou, 2023-11-11:
@goncaloalmeida ：Hi, I use "ldd /usr/local/lib/libvsomeip3-cfg.so" shows as bwlow, but it still not works.
root@uu1:/media/psf/share/vsomeip-3.3.8/examples/hello_world/build# ldd /usr/local/lib/libvsomeip3-cfg.so
	linux-vdso.so.1 (0x0000ffffb0ec3000)
	libvsomeip3.so.3 => /usr/local/lib/libvsomeip3.so.3 (0x0000ffffb0b80000)
	libboost_filesystem.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_filesystem.so.1.74.0 (0x0000ffffb0b30000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffffb0900000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffffb08d0000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffffb0720000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffffb0e8a000)
	libboost_thread.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_thread.so.1.74.0 (0x0000ffffb06e0000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffffb0640000)
[image]