COVESA/vsomeip #569: ~/vsomeip/examples/hello_world$ VSOMEIP_CONFIGURATION=./helloworld-local.json VSOMEIP_APPLICATION_NAME=hello_world_service ./hello_world_service  1 Configuration module could not be loaded!

Hithesh0899, 2023-11-10:
I tried all of my ways to run this but not able to run, kindly help me asap possible because it is very important for me. 
I have give chmod+x even though it is the same , also i have did all the resolutions previously mentioned on this issues portal, but none of them didnt resolve the issue, kindly please resolve.

Thanks and regards,
Hithesh

wangjingxiaomidou, 2023-11-11:
I also meet the issue.

Hithesh0899, 2023-11-11:
Seems so strange even if all dependencies are present, it is not getting executed..

wangjingxiaomidou, 2023-11-11:
Yes, what version of vsomeip do you use?

Hithesh0899, 2023-11-11:
I am using 3.4.9 the latest version

wangjingxiaomidou, 2023-11-11:
@Hithesh0899 Do you try the solutions mentioned in https://github.com/COVESA/vsomeip/issues/568?

Hithesh0899, 2023-11-11:
Yess i have did it ane also Few more resolutions I have tried

goncaloalmeida, 2023-11-11:
where are declared the VSOMEIP_CONFIGURATION, VSOMEIP_APPLICATION_NAME and LD_LIBRARY_PATH?

wangjingxiaomidou, 2023-11-11:
@goncaloalmeida How to check? Actually, I use the default parameters. I follow the step:
mkdir build
cd build
cmake ..
make
make install

After install, the installation path is /usr/local/lib
[image]
wj@uu1:/usr/local/lib$ echo $VSOMEIP_CONFIGURATION
wj@uu1:/usr/local/lib$ echo $VSOMEIP_APPLICATION_NAME
wj@uu1:/usr/local/lib$ echo $LD_LIBRARY_PATH
:/usr/local/lib

Hithesh0899, 2023-11-11:
> where are declared the VSOMEIP_CONFIGURATION, VSOMEIP_APPLICATION_NAME and LD_LIBRARY_PATH?

yeah how to check, and why we have to use VSOMEIP_CONFIGURATION, VSOMEIP_APPLICATION_NAME , why cant we run the application ./hello_world_service ? this issue is so frequent.. kindly give a proper resolution step on the documentation, it will be so helpful

wangjingxiaomidou, 2023-11-11:
Sorry, I don't understand. I do not use the VSOMEIP_CONFIGURATION, VSOMEIP_APPLICATION_NAME. When I run /hello_world_service , it show as below:
/media/psf/share/vsomeip-3.3.8/examples/hello_world/build$ ./hello_world_service 
1 Configuration module could not be loaded!

I follow the doc here https://github.com/COVESA/vsomeip, Is it right?

goncaloalmeida, 2023-11-11:
@wangjingxiaomidou you need to read the readme files, and all the steps indicated in the comments.
https://github.com/COVESA/vsomeip/blob/70d59edc01c9978ada66615c033e941e40561bb6/examples/hello_world/readme

wangjingxiaomidou, 2023-11-11:
@goncaloalmeida understand, Actually, I have run with the readme resolution, the issue is still exit. Please see the result below:
wj@uu1:/media/psf/share/vsomeip-3.3.8/examples/hello_world/build$ VSOMEIP_CONFIGURATION=../helloworld-local.json \
VSOMEIP_APPLICATION_NAME=hello_world_service \
./hello_world_service
1 Configuration module could not be loaded!
[image]

wangjingxiaomidou, 2023-11-11:
My infomation
wj@uu1:/media/psf/share/vsomeip-3.3.8/examples/hello_world/build$ ldd /usr/local/lib/libvsomeip3-cfg.so
	linux-vdso.so.1 (0x0000ffff80661000)
	libvsomeip3.so.3 => /usr/local/lib/libvsomeip3.so.3 (0x0000ffff80310000)
	libboost_filesystem.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_filesystem.so.1.74.0 (0x0000ffff802c0000)
	libstdc++.so.6 => /lib/aarch64-linux-gnu/libstdc++.so.6 (0x0000ffff80090000)
	libgcc_s.so.1 => /lib/aarch64-linux-gnu/libgcc_s.so.1 (0x0000ffff80060000)
	libc.so.6 => /lib/aarch64-linux-gnu/libc.so.6 (0x0000ffff7feb0000)
	/lib/ld-linux-aarch64.so.1 (0x0000ffff80628000)
	libboost_thread.so.1.74.0 => /lib/aarch64-linux-gnu/libboost_thread.so.1.74.0 (0x0000ffff7fe70000)
	libm.so.6 => /lib/aarch64-linux-gnu/libm.so.6 (0x0000ffff7fdd0000)

Hithesh0899, 2023-11-11:
> @wangjingxiaomidou you need to read the readme files, and all the steps indicated in the comments. https://github.com/COVESA/vsomeip/blob/70d59edc01c9978ada66615c033e941e40561bb6/examples/hello_world/readme VSOMEIP_CONFIGURATION and VSOMEIP_APPLICATION_NAME are necessary

Yes i have did the same as per the steps but issue is not getting resolved, please suggest the next steps, thanks  - Hithesh

wangjingxiaomidou, 2023-11-11:
@Hithesh0899 what version of linux do you use? Mine is ubuntu 22.04 arm processor.

Hithesh0899, 2023-11-11:
Hi @wangjingxiaomidou I am using Ubuntu 20.04, Intel x86_64 processor

wangjingxiaomidou, 2023-11-14:
Hi, @Hithesh0899 @goncaloalmeida : I have found the solution.
Actually, I add log in the vsomeip code and find dlopen is failed. I use the “dlerror” to print the error. And the result is that  the error “undefined symbol _zn5boost10filesystem6detail11dir_itr_impd1ev” happened.
[image]
[image]

 So my solution is reinstall the boost based on https://www.boost.org/doc/libs/1_83_0/more/getting_started/unix-variants.html
cd path/to/boost_1_83_0
$ ./bootstrap.sh --help
./bootstrap.sh --prefix=/usr/local
./b2 install

And then, I recompile vsomeip. It works well.
[image]
Hope it is useful for you.

closed 2023-11-14 by goncaloalmeida

Hithesh0899, 2023-11-15:
hi @wangjingxiaomidou, so just i have to follow this steps to reinstall ?
cd path/to/boost_1_83_0
$ ./bootstrap.sh --help
./bootstrap.sh --prefix=/usr/local
./b2 install
2. I have to rebuild the vsomeip from git ? is it the resolution, please confirm thank you

wangjingxiaomidou, 2023-11-15:
Yes, Which country do you come from? Maybe we can talk offline.

506480019
***@***.***

                       
Original Email
                       
                     

Sender:"Hithesh0899"< ***@***.*** &gt;;

Sent Time:2023/11/15 15:07

To:"COVESA/vsomeip"< ***@***.*** &gt;;

Cc recipient:"wangjingxiaomidou"< ***@***.*** &gt;;"Mention"< ***@***.*** &gt;;

Subject:Re: [COVESA/vsomeip] ~/vsomeip/examples/hello_world$VSOMEIP_CONFIGURATION=./helloworld-local.jsonVSOMEIP_APPLICATION_NAME=hello_world_service ./hello_world_service  1Configuration module could not be loaded! (Issue #569)

 
hi @wangjingxiaomidou, so just i have to follow this steps to reinstall ?
 cd path/to/boost_1_83_0
 $ ./bootstrap.sh --help
 ./bootstrap.sh --prefix=/usr/local
 ./b2 install
 2. I have to rebuild the vsomeip from git ? is it the resolution, please confirm thank you

wangjingxiaomidou, 2023-11-15:
@Hithesh0899 Yes

Hithesh0899, 2023-11-15:
Thank you, i will check and come back to you..

wangjingxiaomidou, 2024-01-24:
> ./bootstrap.sh --help

The downloadable file